zephyrproject-rtos/zephyr #84453: xtensa: Do not use special register EPS1, because it does not exist

williamtcdns, 2025-01-23:
In arch/xtensa/core/gen_zsr.py , ZSR_EPS is set to EPS1 when the highest level is 1.

Causing compiler error such as:
zephyrproject/zephyr/arch/xtensa/core/xtensa_asm2_util.S:157: Error: invalid register 'EPS1' for 'wsr' instruction
zephyrproject/zephyr/arch/xtensa/core/xtensa_asm2_util.S:320: Error: unknown opcode or format name 'wsr.eps1'
zephyrproject/zephyr/arch/xtensa/core/xtensa_asm2_util.S:320: Error: unknown opcode or format name 'rsr.eps1'

ZSR_EPS must not be used when the highest level is 1.

dcpleung, 2025-01-23:
The context switching code requires using `RFI` to restore both `PC` and especially `PS`. Zephyr simply will never work without multiple interrupt levels due to this. The main reason to use `EPC*` and `EPS*` for context switching is to avoid being interrupted inside context switching once we set `PS` to the one for the incoming "next" task because this "new" PS will have PS.INTLEVEL being 0. So this is not a bug, and simply the hardware cannot support running Zephyr.

closed 2025-01-23 by dcpleung

williamtcdns, 2025-01-23:
Perhaps it should be documented somewhere as part of the minimum configuration requirement for Xtensa that, multiple interrupt levels are needed ? because we have other RTOS ports working with a single interrupt level.