zephyrproject-rtos/zephyr #84474: tests: tests/lib/devicetree/api_ext/libraries: devicetree:  test Device issue (Flash error?)

hakehuang, 2025-01-24:
**Describe the bug**
 test is Device issue (Flash error?) on v4.0.0-3929-g2e711aaf8159 on frdm_mcxa156/mcxa156
testcase path is tests/lib/devicetree/api_ext

see logs for details

**To Reproduce**
1. 
```
scripts/twister --device-testing --device-serial /dev/ttyACM0 -p frdm_mcxa156/mcxa156  -T tests/lib/devicetree/api_ext --sub-test frdm_mcxa156/mcxa156:tests/lib/devicetree/api_ext/libraries.devicetree.api_ext
```
or
```
# cd tests/tests/lib/devicetree/api_ext
# west build -b frdm_mcxa156/mcxa156
# west flash
```
2. See error

```
drivers/adc/adc_mcux_lpadc.c: In function 'mcux_lpadc_init': 
/home/jenkins/agent/workspace/zephyr_test_pipe/zephyr/drivers/adc/adc_mcux_lpadc.c:521:9: error: too many arguments to 
function 'LPADC_SetOffsetValue' 521 | LPADC_SetOffsetValue(base, | ^~~~~~~~~~~~~~~~~~~~ In file included from 
/home/jenkins/agent/workspace/zephyr_test_pipe/zephyr/drivers/adc/adc_mcux_lpadc.c:24: 
/home/jenkins/agent/workspace/zephyr_test_pipe/modules/hal/nxp/mcux/mcux-sdk/drivers/lpadc/fsl_lpadc.h:1207:20: note: 
declared here 1207 | static inline void LPADC_SetOffsetValue(ADC_Type *base, int16_t value) | ^~~~~~~~~~~~~~~~~~~~ 
[87/148] Building C object zephyr/drivers/pinctrl/CMakeFiles/drivers__pinctrl.dir/pinctrl_nxp_port.c.obj [88/148] Building C 
object zephyr/drivers/timer/CMakeFiles/drivers__timer.dir/sys_clock_init.c.obj [89/148] Building C object 
zephyr/drivers/regulator/CMakeFiles/drivers__regulator.dir/regulator_common.c.obj [90/148] Building C object 
modules/hal_nxp/hal_nxp/CMakeFiles/..__modules__hal__nxp.dir/mcux/mcux-sdk/drivers/common/fsl_common.c.obj [91/148] 
Building C object modules/hal_nxp/hal_nxp/CMakeFiles/..__modules__hal__nxp.dir/mcux/mcux-
sdk/drivers/common/fsl_common_arm.c.obj [92/148] Building C object 
zephyr/drivers/gpio/CMakeFiles/drivers__gpio.dir/gpio_mcux.c.obj [93/148] Building C object 
zephyr/drivers/timer/CMakeFiles/drivers__timer.dir/cortex_m_systick.c.obj [94/148] Building C object 
zephyr/drivers/serial/CMakeFiles/drivers__serial.dir/uart_mcux_lpuart.c.obj [95/148] Building C object 
modules/hal_nxp/hal_nxp/CMakeFiles/..__modules__hal__nxp.dir/mcux/mcux-sdk/devices/MCXA156/drivers/fsl_clock.c.obj 
ninja: build stopped: subcommand failed. 

```

**Expected behavior**
test pass

**Impact**

**Logs and console output**

```
None
    
```
**Environment (please complete the following information):**
 - OS: (e.g. Linux )
 - Toolchain (e.g Zephyr SDK)
 - Commit SHA or Version used: v4.0.0-3929-g2e711aaf8159

mmahadevan108, 2025-02-17:
@hakehuang , unable to reproduce this with `v4.1.0-rc1` tag. Can you please check again.

hakehuang, 2025-02-18:
this has been fixed by fff407d2f608af27c97267722fbb157259ed07c1

closed 2025-02-18 by hakehuang