zephyrproject-rtos/zephyr #84475: net: socket: Incorrect `msg_iovlen` value with TCP Sockets in `zsock_recvmsg()`

skryjapetr, 2025-01-24:
**Describe the bug**
The function `ssize_t zsock_recvmsg(int sock, struct msghdr *msg, int flags)` is expected to receive data and store it into buffers specified by the `msg->msg_iov` array. The `msg->msg_iovlen` field indicates the number of buffers in the array. However, when using a TCP socket, the `msg->msg_iovlen` value is incorrectly updated after a successful receive operation, reducing its value unexpectedly. This issue does not occur with UDP sockets.

**To Reproduce**

1. Set up a UDP socket.
2. Prepare a struct msghdr with two buffers specified in `msg_iov`:
    - Buffer 1: 5 bytes
    - Buffer 2: 5 bytes
    - Set `msg->msg_iovlen` = 2.
3. Send a 10-byte message from peer socket.
4. Call `zsock_recvmsg()` on the client to receive the message.
5. Check `msg->msg_iovlen` value
6. Set up a TCP socket and connect to a server
7. Repeat steps 1-5 with the TCP socket

**Expected behavior**
- `msg->msg_iovlen` remains unchanged at 2 as two buffers were used for storing the data

**Impact**
This bug causes confusion and potential misinterpretation of the `msg->msg_iovlen` value, particularly in cases where applications rely on it to verify the number of buffers used for data storage.

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain Zephyr SDK 4.0

**Additional context**
The `msg_iovlen` is modified in function `zsock_recv_stream_timed()` in module `subsys/net/lib/sockets/sockets_inet.c`.

The code:
```
if (iovec == max_iovlen) {
	break;
}
```
should be probably moved after `msg->msg_iovlen = iovec;` in order to fix this issue.

jukkar, 2025-01-24:
> Actually, I wonder if `zsock_recvmsg()` should modify `msg->msg_iovlen` or `msg_iov[n].iov_len` at all?

That is a good point. It is better not to modify the msg fields so that user can call the `recvmsg()` again with same msg without needing to "reset" the lengths.

skryjapetr, 2025-01-24:
@rlubos  @jukkar that makes sense to me. If so decided, the function `zsock_recv_dgram()`, which is called when doing `recvmsg` with a UDP socket, should also be fixed, because this function also modifies `msg->msg_iovlen` and `msg_iov[n].iov_len`.

PR #84659 (2025-01-30): net: sockets: Make sure recvmsg() doesn't modify buffer configuration 
The recvmsg() function should not modify the buffer configuration provided in struct msghdr, such behavior is not specified in the POSIX function description, nor is it consistent with well-known behavior of this function in Linux.

Align respective tests to verify this.

Fixes #84475
closed 2025-01-30 by kartben