zephyrproject-rtos/zephyr #84541: MCUBoot and Zephyr Link Layer Not Compiling

Jesse-AH-Carpenter, 2025-01-24:
I have an existing project that currently builds and runs fine using the NCS 2.7.0 SDK with nRF-Zephyr 3.6.99. It's a bluetooth application, and I have it configured to use the Zephyr Bluetooth link layer by enabling the CONFIG_BT_LL_SW_SPLIT option within my project config file. We also use MCUBoot as our bootloader which enables us to use DFU. When we migrated to 2.7.0, I moved our project over to a Sysbuild setup as well, and all of these configurations were building and operating just fine. 

I want to update our SDK to the latest NCS v2.9.0 with nRF-Zephyr 3.7.99-ncs2 to mitigate security risks. However when I attempt to do this, I run into the following compilation error:

> warning: Experimental symbol BT_LL_SW_SPLIT is enabled.
Parsing C:/nordicsemi/v2.9.0/zephyr/Kconfig
Loaded configuration 'C:/nordicsemi/myapps/blinky_1/build_9/blinky_1/zephyr/.config'
Merged configuration 'C:/nordicsemi/myapps/blinky_1/build_9/blinky_1/zephyr/.config.sysbuild'
No change to configuration in 'C:/nordicsemi/myapps/blinky_1/build_9/blinky_1/zephyr/.config'
No change to Kconfig header in 'C:/nordicsemi/myapps/blinky_1/build_9/blinky_1/zephyr/include/generated/zephyr/autoconf.h'
-- Including signing script: C:/nordicsemi/v2.9.0/nrf/cmake/sysbuild/image_signing.cmake
-- Configuring done
-- Generating done
-- Build files have been written to: C:/nordicsemi/myapps/blinky_1/build_9/blinky_1
[4/193] Generating include/generated/zephyr/version.h
-- Zephyr version: 3.7.99 (C:/nordicsemi/v2.9.0/zephyr), build: v3.7.99-ncs2
[188/193] Linking C executable zephyr\zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map C:/nordicsemi/myapps/blinky_1/build_9/blinky_1/zephyr/zephyr_pre0.map 
cmd.exe /C "cd . && C:\ncs\toolchains\b620d30767\opt\zephyr-sdk\arm-zephyr-eabi\bin\arm-zephyr-eabi-gcc.exe -Os -DNDEBUG -gdwarf-4 -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr\zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=C:/nordicsemi/myapps/blinky_1/build_9/blinky_1/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/libarch__arm__core.a  zephyr/arch/arch/arm/core/cortex_m/libarch__arm__core__cortex_m.a  zephyr/arch/arch/arm/core/mpu/libarch__arm__core__mpu.a  zephyr/lib/libc/picolibc/liblib__libc__picolibc.a  zephyr/lib/libc/common/liblib__libc__common.a  zephyr/lib/net_buf/liblib__net_buf.a  zephyr/soc/soc/nrf52840/libsoc__nordic.a  zephyr/subsys/random/libsubsys__random.a  zephyr/subsys/bluetooth/common/libsubsys__bluetooth__common.a  zephyr/subsys/bluetooth/host/libsubsys__bluetooth__host.a  zephyr/subsys/bluetooth/controller/libsubsys__bluetooth__controller.a  zephyr/subsys/bluetooth/controller/ll_sw/nordic/libsubsys__bluetooth__controller__ll_sw__nordic.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/entropy/libdrivers__entropy.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  modules/nrf/drivers/hw_cc3xx/lib..__nrf__drivers__hw_cc3xx.a  modules/hal_nordic/nrfx/libmodules__hal_nordic__nrfx.a  modules/segger/libmodules__segger.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -LC:/nordicsemi/myapps/blinky_1/build_9/blinky_1/zephyr  zephyr/arch/common/libisr_tables.a  -mcpu=cortex-m4  -mthumb  -mabi=aapcs  -mfp16-format=ieee  -mtp=soft  -fuse-ld=bfd  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -Wl,-no-pie  -specs=picolibc.specs  -DPICOLIBC_LONG_LONG_PRINTF_SCANF  C:/nordicsemi/v2.9.0/nrfxlib/crypto/nrf_cc310_platform/lib/cortex-m4/soft-float/no-interrupts/libnrf_cc310_platform_0.9.19.a -L"c:/ncs/toolchains/b620d30767/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/thumb/v7e-m/nofp" -lc -lgcc && cmd.exe /C "cd /D C:\nordicsemi\myapps\blinky_1\build_9\blinky_1\zephyr && C:\ncs\toolchains\b620d30767\opt\bin\cmake.exe -E true""
c:/ncs/toolchains/b620d30767/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd.exe: zephyr/subsys/bluetooth/host/libsubsys__bluetooth__host.a(hci_core.c.obj):(.data.bt_dev+0x134): undefined reference to `__device_dts_ord_134'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
[10/20] No configure step for 'mcuboot'
FAILED: _sysbuild/sysbuild/images/blinky_1-prefix/src/blinky_1-stamp/blinky_1-build C:/nordicsemi/myapps/blinky_1/build_9/_sysbuild/sysbuild/images/blinky_1-prefix/src/blinky_1-stamp/blinky_1-build 
cmd.exe /C "cd /D C:\nordicsemi\myapps\blinky_1\build_9\blinky_1 && C:\ncs\toolchains\b620d30767\opt\bin\cmake.exe --build ."
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'C:\ncs\toolchains\b620d30767\opt\bin\cmake.EXE' --build C:/nordicsemi/myapps/blinky_1/build_9

I was able to move past this error by adding the bt-ll-sw-split snippet in the build configuration. With this added, the main image compiles without error. However when it goes to build the mcuboot image, it fails with the following error:

> Parsing C:/nordicsemi/v2.9.0/zephyr/Kconfig
Loaded configuration 'C:/nordicsemi/myapps/blinky_1/build_8/blinky_1/zephyr/.config'
Merged configuration 'C:/nordicsemi/myapps/blinky_1/build_8/blinky_1/zephyr/.config.sysbuild'
No change to configuration in 'C:/nordicsemi/myapps/blinky_1/build_8/blinky_1/zephyr/.config'
No change to Kconfig header in 'C:/nordicsemi/myapps/blinky_1/build_8/blinky_1/zephyr/include/generated/zephyr/autoconf.h'
-- Including signing script: C:/nordicsemi/v2.9.0/nrf/cmake/sysbuild/image_signing.cmake
-- Configuring done
-- Generating done
-- Build files have been written to: C:/nordicsemi/myapps/blinky_1/build_8/blinky_1
[4/195] Generating include/generated/zephyr/version.h
-- Zephyr version: 3.7.99 (C:/nordicsemi/v2.9.0/zephyr), build: v3.7.99-ncs2
[195/195] Linking C executable zephyr\zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       64588 B     499200 B     12.94%
             RAM:       18960 B       256 KB      7.23%
        IDT_LIST:          0 GB        32 KB      0.00%
Generating files from C:/nordicsemi/myapps/blinky_1/build_8/blinky_1/zephyr/zephyr.elf for board: nrf52840dk
image.py: sign the payload
image.py: sign the payload
[6/245] Generating include/generated/zephyr/version.h
-- Zephyr version: 3.7.99 (C:/nordicsemi/v2.9.0/zephyr), build: v3.7.99-ncs2
[240/245] Linking C executable zephyr\zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map C:/nordicsemi/myapps/blinky_1/build_8/mcuboot/zephyr/zephyr_pre0.map 
cmd.exe /C "cd . && C:\ncs\toolchains\b620d30767\opt\zephyr-sdk\arm-zephyr-eabi\bin\arm-zephyr-eabi-gcc.exe -Os -DNDEBUG -gdwarf-4 -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr\zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=C:/nordicsemi/myapps/blinky_1/build_8/mcuboot/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/libarch__arm__core.a  zephyr/arch/arch/arm/core/cortex_m/libarch__arm__core__cortex_m.a  zephyr/arch/arch/arm/core/mpu/libarch__arm__core__mpu.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/libc/common/liblib__libc__common.a  zephyr/lib/net_buf/liblib__net_buf.a  zephyr/soc/soc/nrf52840/libsoc__nordic.a  zephyr/subsys/random/libsubsys__random.a  zephyr/subsys/bluetooth/common/libsubsys__bluetooth__common.a  zephyr/subsys/bluetooth/host/libsubsys__bluetooth__host.a  zephyr/subsys/bluetooth/controller/libsubsys__bluetooth__controller.a  zephyr/subsys/bluetooth/controller/ll_sw/nordic/libsubsys__bluetooth__controller__ll_sw__nordic.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/entropy/libdrivers__entropy.a  zephyr/drivers/flash/libdrivers__flash.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a [... cut ...] yer and just use the nRF Softdevice controller, it builds just fine. So the issue seems to be narrowed down to the interaction between MCUBoot and the Zephyr Link Layer, but only in NCS 2.9.0, not 2.7.0.

I was unable to find any known issues between MCUBoot and the Zephyr Link Layer, but I am hoping someone here can point me in the right direction. I have also made a post on the Nordic DevZone since I am aware that this might just be an issue on their end and not Zephyr's. Thank you in advance.

henrikbrixandersen, 2025-01-25:
Thank you for reporting this. However - unless you are able to reproduce this issue with upstream Zephyr `main` - please report issues with the nRF Connect SDK (NCS) on the [Nordic Semiconductor DevZone](https://devzone.nordicsemi.com/).

fabiobaltieri, 2025-01-28:
Please open a new issue in the NCS project.

closed 2025-01-28 by fabiobaltieri