zephyrproject-rtos/zephyr #84588: ztest: error_hook: testing.ztest.error_hook.no_userspace reports timeout

katgiadla, 2025-01-27:
**Describe the bug**
Test` tests/ztest/error_hook/testing.ztest.error_hook.no_userspace` fails.

Observed for `nrf54h20dk@0.9.0/nrf54h20/cpuppr`.

**To Reproduce**
Steps to reproduce the behavior:
1. have `nrf54h20dk/nrf54h20` connected
2. go to your zephyr dir
3. call `./scripts/twister -s testing.ztest.error_hook.no_userspace -p nrf54h20dk/nrf54h20/cpuppr --device-testing --device-serial /dev/ttyACM1 -v --inline-logs --west-flash="--erase"`
4. See console output with error (using e.g. minicom)

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
*** Booting Zephyr OS build v4.0.0-3992-gf0d0264c0572 ***
Running TESTSUITE error_hook_tests
===================================================================
START - test_catch_assert_fail
1
2
3
4
 FAIL - test_catch_assert_fail in 0.002 seconds
===================================================================
START - test_catch_assert_in_isr
 PASS - test_catch_assert_in_isr in 0.001 seconds
===================================================================
START - test_catch_fatal_error
case type is 3
E: 
E:  mcause: 11, Environment call from M-mode
E:   mtval: 0
E:      a0: 00000004    t0: 00000000
E:      a1: 2fc058a0    t1: 00000000
E:      a2: 2fc07290    t2: 00000001
E:      a3: 00000008
E:      a4: 2fc06508
E:      a5: 2fc06000
E:      sp: 2fc072a0
E:      ra: 2fc0155e
E:    mepc: 2fc01566
E: mstatus: 00001880
E: 
E:      s0: 2fc014fc
E:      s1: 2fc05a94
E: 
E: call trace:
E:       0: sp: 2fc072a0 ra: 2fc01566
E: 
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x2fc06508 (unknown)
Caught system error -- reason 4 1
Fatal error expected as part of test case.
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
- Toolchain Zephyr SDK 0.17.0
- Commit SHA or Version used: ` v4.0.0-3190-g99a63a`.

nika-nordic, 2025-01-27:
@masz-nordic can you take a look whether this test can be executed at VPR at all?

masz-nordic, 2025-01-30:
`test_catch_assert_fail` needs explicit `CONFIG_ASSERT=y` since [VPRs disable assertions by default](https://github.com/zephyrproject-rtos/zephyr/blob/main/soc/nordic/nrf54h/Kconfig.defconfig.nrf54h20_cpuppr#L13).

`test_catch_fatal_error` is using nested traps, so this is a HW limitation.