zephyrproject-rtos/zephyr #84644: MQTT-SN doesn't not wait for SUBACK or REGACK befor sending the next SUBSCRIBE or REGISTER packet

canatella, 2025-01-28:
**Describe the bug**
The MQTT-SN spec says:

> 6.5 Topic Name Registration Procedur
> ...
> At any point in time a client may have only one REGISTER message outstanding, i.e. it has to wait for a
REGACK message before it can register another topic name.
> ...
> 6.9 Client’s Topic Subscribe/Un-subscribe Procedure
> ...
> As for the REGISTER procedure, a client may have only one SUBSCRIBE or one UNSUBCRIBE transaction
open at a time.
> ...

The provided MQTT-SN library does not wait for  either SUBACK or REGACK before sending the next subscribe or register packet nor does it provide an event to let the library user handle that. 

**To Reproduce**

Just add a second call to `mqtt_sn_subscribe` in the https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/net/mqtt_sn_publisher/src/udp.c

**Expected behavior**

The MQTT-SN client should send the first packet, wait for the SUBACK or REGACK then continue processing the topic list.

**Impact**

I'm not sure how MQTT-SN gateway will behave regarding this. At the moment I'm prototyping and there's no actual MQTT GW or server, so I'm handling that. I'm not sure how an actual SN GW or MQTT broker implementation` wouldl handle that.

**Logs and console output**

N/A

**Environment (please complete the following information):**

I'm on 6c5454b22ee under linux developing on a nrf5340 dev board.

**Additional context**

This could probably be easily implemented in the by just having two flags in the client so that it knows it's waiting for a SUBACK/REGACK.

BeckmaR, 2025-01-28:
I will try to take a look in the next few days, but I haven't actually worked on the library since initially publishing it.

jukkar, 2025-01-28:
> I will try to take a look in the next few days, but I haven't actually worked on the library since initially publishing it.

I added you to assigned list because you were mentioned as a collaborator for this code. It would be great if you can take a look to this.

rlubos, 2025-01-28:
I started to wonder how could this be addressed, not sure if it could be as simple as adding extra flags. Perhaps we need to introduce some new topic states (`MQTT_SN_TOPIC_STATE_REGISTERED_PENDING` etc.) and upon reception of the REGACK packet change the state of the oldest pending entry to `MQTT_SN_TOPIC_STATE_REGISTERING` and reschedule the process work. Similarly for subscriptions.

BeckmaR, 2025-01-28:
@rlubos Either there should be a new topic state, or it could possibly be done by looking at the retry time, _or_ one could simply return `-EAGAIN` when a call would require to subscribe to a topic when a subscription is already in progress. Might be the simplest solution, but of course slightly less convenient for the user - the usual question of "when should I retry?".

rlubos, 2025-01-28:
> or one could simply return -EAGAIN when a call would require to subscribe to a topic when a subscription is already in progress. Might be the simplest solution, but of course slightly less convenient for the user - the usual question of "when should I retry?"

That's the point, w/o extending the notifications with some REGACK representation to notify the user it's ok to proceed I wouldn't really go that way. And still, the whole topic registration seems kind of hidden from the user so it'd be a bit counter-intuitive they need to monitor some registration-related events.

BeckmaR, 2025-01-31:
Looking at the code, I'm going to do the following things:
1. Rename a few internal things that aren't intuitive right now
1. Add documentation, not everything is as easy to understand as it seemed two years ago
1. Add a test that demonstrates the non-conforming behavior
1. Find a way to get the test to pass without changing the public API (ideally, as discussed with @rlubos)

BeckmaR, 2025-02-01:
PR is open, @canatella could you also take a look

canatella, 2025-02-03:
> PR is open, [@canatella](https://github.com/canatella) could you also take a look
Looks good to me, thanks!

closed 2025-02-11 by fabiobaltieri

canatella, 2025-02-13:
Thanks for the work and for acting so fast on this!

PR #85001 (2025-02-11): net: mqtt-sn: Wait for SUBACK and REGACK
Closes #84644
commit cc334912 (2025-02-11): net: mqtt-sn: Wait for register & subscribe to complete

From the MQTT-SN spec:

6.5 Topic Name Registration Procedure

...
At any point in time a client may have only one
REGISTER message outstanding, i.e. it has to wait
for a REGACK message before it can register another
topic name.
...

6.9 Client’s Topic Subscribe/Un-subscribe Procedure

...
As for the REGISTER procedure, a client may have only
one SUBSCRIBE or one UNSUBCRIBE transaction open at a time.
...

Until now, the library did not comply with these requirements.
An additional "waiting" state for topics was introduced:
REGISTER, SUB [... cut]