zephyrproject-rtos/zephyr #84662: USB: device_next: assert in udc_event_xfer_in when doing nrjfprof --reset on nRF5340 Audio DK

Thalley, 2025-01-28:
**Describe the bug**
There is an assert in the `udc_event_xfer_in` function that may be triggered when resetting the board via `nrjfprof --reset`

https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/usb/udc/udc_nrf.c#L173-L179

**To Reproduce**
1) Build and flash the sample from https://github.com/zephyrproject-rtos/zephyr/pull/84359 for `nrf5340_audio_dk/nrf5340/cpuapp` with `CONFIG_ASSERT=y`
2) After the sample has been running a while, do a `nrjfprof --reset` on the board. It happens about 15% of the time

**Expected behavior**
Expect no asserts on a reboot. 

**Impact**
Only affects builds with asserts enabled. When asserts are disabled no issues were seen. 

**Logs and console output**
```*** Booting Zephyr OS build v4.0.0-3988-g94b3ac667bfb ***
[00:00:00.281,372] <inf> bt_hci_core: hci_vs_init: HW Platform: Nordic Semiconductor (0x0002)
[00:00:00.281,402] <inf> bt_hci_core: hci_vs_init: HW Variant: nRF53x (0x0003)
[00:00:00.281,433] <inf> bt_hci_core: hci_vs_init: Firmware: Standard Bluetooth controller (0x00) Version 4.0 Build 99
[00:00:00.283,752] <dbg> bt_smp: bt_smp_init: LE SC enabled
[00:00:00.283,905] <inf> bt_hci_core: bt_dev_show_info: Identity: E6:92:AD:99:8C:A9 (random)
[00:00:00.283,935] <inf> bt_hci_core:[00:00:00.283,966] <inf> bt_hci_core: bt_dev_show_info: LMP: version 5.4 (0x0d) subver 0xffff
Bluetooth initialized
[00:00:00.284,576] <inf> lc3: lc3_init: LC3 initialized
[00:00:00.284,667] <inf> udc_nrf: udc_nrf_init: Initialized
[00:00:00.284,698] <inf> usbd_init: init_configuration_inst: interface 0 alternate 0
[00:00:00.284,729] <inf> usbd_init: init_configuration_inst: interface 1 alternate 0
[00:00:00.284,729] <inf> usbd_init: init_configuration_inst: interface 1 alternate 1
[00:00:00.284,790] <inf> usbd_init: init_configuration_inst:    ep 0x88 mps 0x00c4 interface ep-bm 0x01000000
[00:00:00.284,820] <inf> usbd_init: init_configuration_inst: Instance iface-bm 0x00000003 ep-bm 0x01000000
[00:00:00.284,820] <inf> usbd_init: init_configuration: Init class node 0x20000484, descriptor length 118
[00:00:00.284,851] <inf> usbd_init: init_configuration: bNumInterfaces 2 wTotalLength 127
[00:00:00.284,851] <inf> usbd_init: usbd_init_configurations: FS bNumConfigurations 1
[00:00:00.285,003] <inf> usb: usb_init: USB initialized
Reset
Starting advertising
Waiting for Broadcast Assistant
[00:00:00.289,123] <inf> udc_nrf: udc_nrf_thread: SUSPEND state detected
[00:00:00.632,598] <inf> udc_nrf: usbd_event_handler: Reset
[00:00:00.632,659] <inf> udc_nrf: udc_nrf_thread: RESUMING from suspend
[00:00:00.632,690] <inf> udc: udc_ep_dequeue: ep 0x80 is not halted|disabled
[00:00:00.632,720] <inf> usbd_core: event_handler_bus_reset: Actual device speed 1
[00:00:00.664,398] <dbg> bt_smp: bt_smp_pkey_ready: 
[00:00:00.722,656] <inf> usbd_ch9: usbd_handle_ctrl_xfer: Handle control 0x2001579c ep 0x00, len 8, s:1 d:0 s:0
[00:00:00.722,930] <inf> udc_nrf: usbd_event_handler: Reset
[00:00:00.722,991] <inf> udc: udc_ep_dequeue: ep 0x80 is not halted|disabled
[00:00:00.722,991] <inf> usbd_core: event_handler_bus_reset: Actual device speed 1
[00:00:00.812,683] <inf> usbd_ch9: usbd_handle_ctrl_xfer: Handle control 0x200157c0 ep 0x00, len 8, s:1 d:0 s:0
[00:00:00.812,805] <inf> usbd_ch9: usbd_handle_ctrl_xfer: Handle control 0x2001579c ep 0x80, len 0, s:0 d:0 s:1
[00:00:00.812,805] <inf> usbd_ch9: usbd_handle_ctrl_xfer: s-(out)-status finished
[00:00:00.825,958] <inf> usbd_ch9: usbd_handle_ctrl_xfer: Handle control 0x2001579c ep 0x00, len 8, s:1 d:0 s:0
[00:00:00.826,416] <inf> usbd_ch9: usbd_handle_ctrl_xfer: Handle control 0x200157c0 ep 0x00, len 8, s:1 d:0 s:0
[00:00:00.826,446] <inf> usbd_ch9: handle_setup_request: protocol error:
[00:00:00.826,477] <inf> usbd_ch9: handle_setup_request: setup:
                                   80 06 00 06 00 00 0a 00                          |........         
[00:00:00.826,477] <inf> usbd_ch9: handle_setup_request: not supported
[00:00:00.826,782] <inf> udc: udc_ctrl_update_stage: Sequence 2 not completed
[00:00:00.826,843] <inf> usbd_ch9: usbd_handle_ctrl_xfer: Handle control 0x2001579c ep 0x00, len 8, s:1 d:0 s:0
[00:00:00.826,904] <inf> usbd_ch9: handle_setup_request: protocol error:
[00:00:00.826,904] <inf> usbd_ch9: handle_setup_request: setup:
                                   80 06 00 06 00 00 0a 00                          |........         
[00:00:00.826,934] <inf> usbd_ch9: handle_setup_request: not supported
[00:00:00.827,178] <inf> udc: udc_ctrl_update_stage: Sequence 2 not completed
[00:00:00.827,270] <inf> usbd_ch9: usbd_handle_ctrl_xfer: Handle control 0x200157c0 ep 0x00, len 8, s:1 d:0 s:0
[00:00:00.827,301] <inf> usbd_ch9: handle_setup_request: protocol error:
[00:00:00.827,331] <inf> usbd_ch9: handle_setup_request: setup:
                                   80 06 00 06 00 00 0a 00                          |........         
[00:00:00.827,331] <inf> usbd_ch9: handle_setup_request: not supported
[00:00:00.827,606] <inf> udc: udc_ctrl_update_stage: Sequence 2 not completed
[... 33 lines cut ...]
[00:00:00.835,784] <err> udc_nrf: udc_event_xfer_in: ep 0x88 queue is empty
ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/drivers/usb/udc/udc_nrf.c:177
[00:00:00.835,845] <err> os: esf_dump: r0/a1:  0x00000004  r1/a2:  0x000000b1  r2/a3:  0x00000050
[00:00:00.835,845] <err> os: esf_dump: r3/a4:  0x00000004 r12/ip:  0x00000014 r14/lr:  0x0001bb91
[00:00:00.835,876] <err> os: esf_dump:  xpsr:  0x01000000
[00:00:00.835,876] <err> os: esf_dump: s[ 0]:  0xaaaaaaaa  s[ 1]:  0xaaaaaaaa  s[ 2]:  0xaaaaaaaa  s[ 3]:  0x00060000
[00:00:00.835,906] <err> os: esf_dump: s[ 4]:  0xaaaa0088  s[ 5]:  0x20011f60  s[ 6]:  0x000000c0  s[ 7]:  0x00000000
[00:00:00.835,906] <err> os: esf_dump: s[ 8]:  0xaaaaaaaa  s[ 9]:  0x2000fd70  s[10]:  0x0001ba41  s[11]:  0x00033b70
[00:00:00.835,937] <err> os: esf_dump: s[12]:  0x00000000  s[13]:  0x00000000  s[14]:  0x00000008  s[15]:  0x00000000
[00:00:00.835,937] <err> os: esf_dump: fpscr:  0x00000000
[00:00:00.835,968] <err> os: esf_dump: Faulting instruction address (r15/pc): 0x0002cfb0
[00:00:00.835,998] <err> os: z_fatal_error: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
[00:00:00.836,029] <err> os: z_fatal_error: Current thread: 0x200049f8 (udc_nrfx)
[00:00:01.628,173] <err> os: k_sys_fatal_error_handler: Halting system
[00:00:00.253,143] <inf> udc_nrf: udc_nrf_driver_init: Preinit
```

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows): Arch Linux
 - Toolchain (e.g Zephyr SDK, ...): Zephyr SDK 0.17.0
 - Commit SHA or Version used: https://github.com/zephyrproject-rtos/zephyr/pull/84359 on 4455cd7351ba2990aa6cb44bc09803b154d4a3f5

**Additional context**
N/A

jfischer-no, 2025-01-28:
You need to reproduce it with the examples we have in the tree and make sure it is not an nRF5440 specific issue.

Thalley, 2025-01-28:
> You need to reproduce it with the examples we have in the tree and make sure it is not an nRF5340 specific issue.

Reopening as that is not requirement for bug reports, and this needs to be fixed before we can start to rely on USB_DEVICE_STACK_NEXT.

Thalley, 2025-01-30:
Using the same sample application with `nrf52840dk/nrf52840` triggers the same issue:
```
*** Booting Zephyr OS build v4.0.0-41[00:00:00.374,694] <inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)
[00:00:00.374,725] <inf> bt_hci_core: HW Variant: nRF52x (0x0002)
[00:00:00.374,755] <inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 4.0 Build 99
[00:00:00.376,007] <inf> bt_hci_core: Identity: C4:E7:C8:4B:98:F5 (random)
[00:00:00.376,037] <inf> bt_hci_core: HCI: version 5.4 (0x0d) revision 0x0000, manufacturer 0x05f1
[00:00:00.376,068] <inf> bt_hci_core: LMP: version 5.4 (0x0d) subver 0xffff
Bluetooth initialized
[00:00:00.376,647] <inf> lc3: LC3 initialized
[00:00:00.376,739] <inf> udc_nrf: Initialized
[00:00:00.376,770] <inf> usbd_init: interface 0 alternate 0
[00:00:00.376,800] <inf> usbd_init: interface 1 alternate 0
[00:00:00.376,800] <inf> usbd_init: interface 1 alternate 1
[00:00:00.376,892] <inf> usbd_init:     ep 0x88 mps 0x00c4 interface ep-bm 0x01000000
[00:00:00.376,922] <inf> usbd_init: Instance iface-bm 0x00000003 ep-bm 0x01000000
[00:00:00.376,922] <inf> usbd_init: Init class node 0x20000854, descriptor length 118
[00:00:00.376,953] <inf> usbd_init: bNumInterfaces 2 wTotalLength 127
[00:00:00.376,953] <inf> usbd_init: FS bNumConfigurations 1
[00:00:00.377,105] <inf> usb: USB initialized
Reset
Starting advertising
Waiting for Broadcast Assistant
[00:00:00.381,286] <inf> udc_nrf: SUSPEND state detected
[00:00:00.550,109] <inf> udc_nrf: Reset
[00:00:00.550,140] <inf> udc_nrf: RESUMING from suspend
[00:00:00.550,201] <inf> udc: ep 0x80 is not halted|disabled
[00:00:00.550,231] <inf> usbd_core: Actual device speed 1
[00:00:00.619,567] <inf> usbd_ch9: Handle control 0x20018c94 ep 0x00, len 8, s:1 d:0 s:0
[00:00:00.620,147] <inf> udc_nrf: Reset
[00:00:00.620,208] <inf> udc: ep 0x80 is not halted|disabled
[00:00:00.620,208] <inf> usbd_core: Actual device speed 1
[00:00:00.692,993] <inf> usbd_ch9: Handle control 0x20018cb8 ep 0x00, len 8, s:1 d:0 s:0
[00:00:00.693,115] <inf> usbd_ch9: Handle control 0x20018c94 ep 0x80, len 0, s:0 d:0 s:1
[00:00:00.693,145] <inf> usbd_ch9: s-(out)-status finished
[00:00:00.706,176] <inf> usbd_ch9: Handle control 0x20018c94 ep 0x00, len 8, s:1 d:0 s:0
[00:00:00.706,665] <inf> usbd_ch9: Handle control 0x20018cb8 ep 0x00, len 8, s:1 d:0 s:0
[00:00:00.706,695] <inf> usbd_ch9: protocol error:
[00:00:00.706,726] <inf> usbd_ch9: setup:
                                   80 06 00 06 00 00 0a 00                          |........         
[00:00:00.706,756] <inf> usbd_ch9: not supported
[00:00:00.707,031] <inf> udc: Sequence 2 not completed
[00:00:00.707,122] <inf> usbd_ch9: Handle control 0x20018c94 ep 0x00, len 8, s:1 d:0 s:0
[00:00:00.707,153] <inf> usbd_ch9: protocol error:
[00:00:00.707,183] <inf> usbd_ch9: setup:
                                   80 06 00 06 00 00 0a 00                          |........         
[00:00:00.707,183] <inf> usbd_ch9: not supported
[00:00:00.707,611] <inf> udc: Sequence 2 not completed
[00:00:00.707,672] <inf> usbd_ch9: Handle control 0x20018cb8 ep 0x00, len 8, s:1 d:0 s:0
[00:00:00.707,733] <inf> usbd_ch9: protocol error:
[00:00:00.707,733] <inf> usbd_ch9: setup:
                                   80 06 00 06 00 00 0a 00                          |........         
[00:00:00.707,763] <inf> usbd_ch9: not supported
[00:00:00.708,160] <inf> udc: Sequence 2 not completed
[00:00:00.708,251] <inf> usbd_ch9: Handle control 0x20018c94 ep 0x00, len 8, s:1 d:0 s:0
[00:00:00.708,770] <inf> usbd_ch9: Handle control 0x20018cb8 ep 0x00, len 8, s:1 d:0 s:0
[00:00:00.709,442] <inf> usbd_ch9: Handle control 0x20018c94 ep 0x00, len 8, s:1 d:0 s:0
[00:00:00.710,083] <inf> usbd_ch9: Handle control 0x20018cb8 ep 0x00, len 8, s:1 d:0 s:0
[00:00:00.710,540] <inf> usbd_ch9: Handle control 0x20018c94 ep 0x00, len 8, s:1 d:0 s:0
[00:00:00.711,090] <inf> usbd_ch9: Handle control 0x20018cb8 ep 0x00, len 8, s:1 d:0 s:0
[00:00:00.712,219] <inf> usbd_ch9: Handle control 0x20018c94 ep 0x00, len 8, s:1 d:0 s:0
[00:00:00.712,249] <inf> usbd_ch9: Set Configuration Request value 1
[00:00:00.712,280] <inf> usbd_iface: No endpoints, skip interface
[00:00:00.712,310] <inf> usbd_iface: No endpoints, skip interface
[00:00:00.712,463] <inf> usbd_ch9: Handle control 0x20018cb8 ep 0x80, len 0, s:0 d:0 s:1
[00:00:00.712,463] <inf> usbd_ch9: s-(out)-status finished
[00:00:00.712,677] <inf> usbd_ch9: Handle control 0x20018cb8 ep 0x00, len 8, s:1 d:0 s:0
[00:00:00.713,623] <inf> usbd_ch9: Handle control 0x20018c94 ep 0x00, len 8, s:1 d:0 s:0
[00:00:00.714,233] <inf> usbd_ch9: Handle control 0x20018cb8 ep 0x00, len 8, s:1 d:0 s:0
[00:00:00.714,691] <inf> usbd_ch9: Handle control 0x20018c94 ep 0x00, len 8, s:1 d:0 s:0
[... 6 lines cut ...]
[00:00:00.716,979] <inf> usbd_ch9: Handle control 0x20018c94 ep 0x00, len 8, s:1 d:0 s:0
[00:00:00.717,041] <inf> usbd_iface: Set Interfaces 1, alternate 0 -> 1
[00:00:00.717,041] <inf> usbd_iface: Modify interface 1 ep 0x88 by op 0 ep_bm 0
[00:00:00.717,071] <inf> usbd_iface: No endpoints, skip interface
[00:00:00.717,102] <inf> usbd_iface: Modify interface 1 ep 0x88 by op 1 ep_bm 1000000
[00:00:00.717,224] <inf> usbd_ch9: Handle control 0x20018cb8 ep 0x80, len 0, s:0 d:0 s:1
[00:00:00.717,254] <inf> usbd_ch9: s-(out)-status finished
[00:00:00.718,078] <inf> usbd_ch9: Handle control 0x20018cb8 ep 0x00, len 8, s:1 d:0 s:0
[00:00:00.718,139] <inf> usbd_iface: Set Interfaces 1, alternate 1 -> 0
[... cut ...]
[00:00:00.718,414] <err> os: s[ 8]:  0xaaaaaaaa  s[ 9]:  0x20013f70  s[10]:  0x00037495  s[11]:  0x0004dfd8
[00:00:00.718,414] <err> os: s[12]:  0x00000000  s[13]:  0x00000000  s[14]:  0x00000008  s[15]:  0x00000000
[00:00:00.718,444] <err> os: fpscr:  0x00000000
[00:00:00.718,444] <err> os: Faulting instruction address (r15/pc): 0x0004770a
[00:00:00.718,475] <err> os: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
[00:00:00.718,536] <err> os: Current thread: 0x20005500 (udc_nrfx)
[00:00:02.696,685] <err> os: Halting system
```

tmon-nordic, 2025-01-31:
The assert is not really triggered by reset. The root cause behind this is switching from Alternate configuration with isochronous IN endpoints (here: Interface 1 alternate 1) to Alternate configuration without isochronous IN endpoints (here: Interface 1 alternate 0). The problem is in nRF UDC driver.

This happens because `udc_nrf_thread()` is not synchronized against the UDC driver actions. The execution is as follows:
  * IN endpoint data is available (on SOF)
  * DMA is started to copy the iso IN data to buffer
  * DMA finishes, `NRF_USBD_COMMON_EVT_EPTRANSFER` is posted to queue
  * Handle control to set alternate 0 executes, during the handling iso IN endpoint gets disabled and queue is emptied
  * `udc_nrf_thread` picks up the `NRF_USBD_COMMON_EVT_EPTRANSFER` from queue and notices there are no buffer queued for the endpoint

closed 2025-04-09 by carlescufi