zephyrproject-rtos/zephyr #84664: USB: device_next: `buf_release_cb` may be called with `NULL` buffer

Thalley, 2025-01-28:
**Expected behavior**
Expect `uac2_ops.buf_release_cb` to never be called with `NULL`, or that is documented to happened for some reason. 

**Impact**
Easy workaround is to add checks in the application for `NULL`

**Logs and console output**
`uac2_sof_cb` logs the address of the buffer before calling `usbd_uac2_send` and `uac2_buf_release_cb` logs the address of the buffer from the callback arguments. 

It can be seen that `uac2_sof_cb` calls `usbd_uac2_send` 2 times, but `uac2_buf_release_cb` is called 3 times; 2 times with the expected buffer addresses and once with `0`/`NULL`. 
```
[00:00:13.711,608] <err> usb: uac2_sof_cb: 0x20012020
[00:00:13.711,730] <err> udc_nrf: udc_event_xfer_in_next: ep 0x88 nrfx error: bad000b
[00:00:13.711,822] <err> usb: uac2_buf_release_cb: 0x20011f60
[00:00:13.711,914] <err> usb: uac2_sof_cb: 0x20011f60
[00:00:13.712,005] <err> usbd_uac2: uac2_request: request ep 0x88, len 192 failed
[00:00:13.712,005] <err> usb: uac2_buf_release_cb: 0x20012020
[00:00:13.712,127] <err> usb: uac2_buf_release_cb: 0
ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/kernel/mem_slab.c:273
        Invalid memory pointer provided
[00:00:13.712,219] <err> os: esf_dump: r0/a1:  0x00000004  r1/a2:  0x00000111  r2/a3:  0x0000012a
[00:00:13.712,249] <err> os: esf_dump: r3/a4:  0x00000004 r12/ip:  0x00000014 r14/lr:  0x00027c91
[00:00:13.712,280] <err> os: esf_dump:  xpsr:  0x01000000
[00:00:13.712,341] <err> os: esf_dump: s[ 0]:  0x20001928  s[ 1]:  0x00000000  s[ 2]:  0x2000ecb0  s[ 3]:  0x20009ff0
[00:00:13.712,371] <err> os: esf_dump: s[ 4]:  0x2000ecb0  s[ 5]:  0x00000088  s[ 6]:  0x00033b5c  s[ 7]:  0x00002959
[00:00:13.712,402] <err> os: esf_dump: s[ 8]:  0x200136fc  s[ 9]:  0x00002929  s[10]:  0x2000ecc8  s[11]:  0x0000b80d
[00:00:13.712,432] <err> os: esf_dump: s[12]:  0x00000000  s[13]:  0xe000ed00  s[14]:  0x20004d30  s[15]:  0xffffff88
[00:00:13.712,463] <err> os: esf_dump: fpscr:  0x000000c0
[00:00:13.712,493] <err> os: esf_dump: Faulting instruction address (r15/pc): 0x0002cfb0
[00:00:13.712,524] <err> os: z_fatal_error: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
[00:00:13.712,585] <err> os: z_fatal_error: Current thread: 0x20004180 (usbd)
[00:00:15.541,870] <err> os: k_sys_fatal_error_handler: Halting system
```

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows): Arch Linux
 - Toolchain (e.g Zephyr SDK, ...): Zephyr SDK 0.17.0
 - Commit SHA or Version used: https://github.com/zephyrproject-rtos/zephyr/pull/84359 on 4455cd7351ba2990aa6cb44bc09803b154d4a3f5

**Additional context**
N/A

Thalley, 2025-01-30:
Also reproducible with the nRF52840DK. with
```c
static void uac2_buf_release_cb(const struct device *dev, uint8_t terminal, void *buf,
				void *user_data)
{
	/* buf may be NULL sometimes for some reason
	 * https://github.com/zephyrproject-rtos/zephyr/issues/84664
	 */
	if (buf != NULL) {
		k_mem_slab_free(&usb_out_buf_pool, buf);
	} else {
		LOG_ERR("BUF WAS NULL");
	}
}
```
I see

```
[00:00:15.715,393] <err> udc_nrf: ep 0x88 nrfx error: bad000b
[00:00:15.715,454] <err> usbd_uac2: request ep 0x88, len 192 failed
[00:00:15.715,515] <err> usbd_uac2: request ep 0x88, len 192 failed
[00:00:15.715,515] <err> usb: BUF WAS NULL
[00:00:15.716,400] <err> udc_nrf: ep 0x88 nrfx error: bad000b
[00:00:15.716,461] <err> usbd_uac2: request ep 0x88, len 192 failed
[00:00:15.716,522] <err> usbd_uac2: request ep 0x88, len 192 failed
[00:00:15.716,522] <err> usb: BUF WAS NULL
[00:00:15.717,468] <err> udc_nrf: ep 0x88 nrfx error: bad000b
[00:00:15.717,559] <err> usbd_uac2: request ep 0x88, len 192 failed
[00:00:15.717,620] <err> usbd_uac2: request ep 0x88, len 192 failed
[00:00:15.717,620] <err> usb: BUF WAS NULL
[00:00:15.718,383] <err> udc_nrf: ep 0x88 nrfx error: bad000b
[00:00:15.718,444] <err> usbd_uac2: request ep 0x88, len 192 failed
[00:00:15.718,505] <err> usbd_uac2: request ep 0x88, len 192 failed
[00:00:15.718,536] <err> usb: BUF WAS NULL
[00:00:15.719,390] <err> udc_nrf: ep 0x88 nrfx error: bad000b
[00:00:15.719,451] <err> usbd_uac2: request ep 0x88, len 192 failed
[00:00:15.719,512] <err> usbd_uac2: request ep 0x88, len 192 failed
[00:00:15.719,512] <err> usb: BUF WAS NULL
[00:00:15.720,397] <err> udc_nrf: ep 0x88 nrfx error: bad000b
[00:00:15.720,458] <err> usbd_uac2: request ep 0x88, len 192 failed
[00:00:15.720,520] <err> usbd_uac2: request ep 0x88, len 192 failed
[00:00:15.720,520] <err> usb: BUF WAS NULL
[00:00:15.721,405] <err> udc_nrf: ep 0x88 nrfx error: bad000b
[00:00:15.721,466] <err> usbd_uac2: request ep 0x88, len 192 failed
[00:00:15.721,496] <err> usbd_uac2: request ep 0x88, len 192 failed
[00:00:15.721,527] <err> usb: BUF WAS NULL
[00:00:15.722,381] <err> udc_nrf: ep 0x88 nrfx error: bad000b
[00:00:15.722,442] <err> usbd_uac2: request ep 0x88, len 192 failed
[00:00:15.722,503] <err> usbd_uac2: request ep 0x88, len 192 failed
[00:00:15.722,534] <err> usb: BUF WAS NULL
[00:00:15.723,388] <err> udc_nrf: ep 0x88 nrfx error: bad000b
[00:00:15.723,449] <err> usbd_uac2: request ep 0x88, len 192 failed
[00:00:15.723,510] <err> usbd_uac2: request ep 0x88, len 192 failed
[00:00:15.723,510] <err> usb: BUF WAS NULL
[00:00:15.724,395] <err> udc_nrf: ep 0x88 nrfx error: bad000b
[00:00:15.724,456] <err> usbd_uac2: request ep 0x88, len 192 failed
[00:00:15.724,517] <err> usbd_uac2: request ep 0x88, len 192 failed
[00:00:15.724,517] <err> usb: BUF WAS NULL
[00:00:15.725,402] <err> udc_nrf: ep 0x88 nrfx error: bad000b
[00:00:15.725,463] <err> usbd_uac2: request ep 0x88, len 192 failed
[00:00:15.725,494] <err> usbd_uac2: request ep 0x88, len 192 failed
[00:00:15.725,524] <err> usb: BUF WAS NULL
[00:00:15.726,379] <err> udc_nrf: ep 0x88 nrfx error: bad000b
[00:00:15.726,440] <err> usbd_uac2: request ep 0x88, len 192 failed
[00:00:15.726,501] <err> usbd_uac2: request ep 0x88, len 192 failed
[00:00:15.726,531] <err> usb: BUF WAS NULL
[00:00:15.727,478] <err> udc_nrf: ep 0x88 nrfx error: bad000b
[00:00:15.727,539] <err> usbd_uac2: request ep 0x88, len 192 failed
[00:00:15.727,600] <err> usbd_uac2: request ep 0x88, len 192 failed
[00:00:15.727,630] <err> usb: BUF WAS NULL
[00:00:15.728,393] <err> udc_nrf: ep 0x88 nrfx error: bad000b
[00:00:15.728,454] <err> usbd_uac2: request ep 0x88, len 192 failed
[00:00:15.728,485] <err> usbd_uac2: request ep 0x88, len 192 failed
[00:00:15.728,515] <err> usb: BUF WAS NULL
[00:00:15.729,370] <err> udc_nrf: ep 0x88 nrfx error: bad000b
[00:00:15.729,431] <err> usbd_uac2: request ep 0x88, len 192 failed
[00:00:15.729,492] <err> usbd_uac2: request ep 0x88, len 192 failed
[00:00:15.729,522] <err> usb: BUF WAS NULL
[00:00:15.729,980] <inf> usbd_ch9: Handle control 0x20018cb8 ep 0x00, len 8, s:1 d:0 s:0
[00:00:15.730,041] <inf> usbd_iface: Set Interfaces 1, alternate 1 -> 0
[00:00:15.730,072] <inf> usbd_iface: No endpoints, skip interface
[00:00:15.730,224] <wrn> udc_nrf: aborted IN ep 0x88
[00:00:15.730,285] <inf> usbd_iface: Modify interface 1 ep 0x88 by op 2 ep_bm 0
[00:00:15.730,316] <inf> usbd_iface: No endpoints, skip interface
[00:00:15.730,377] <wrn> usbd_uac2: request ep 0x88, len 192 cancelled
[00:00:15.730,407] <err> usb: BUF WAS NULL
[00:00:15.730,560] <inf> usbd_ch9: Handle control 0x20018c94 ep 0x80, len 0, s:0 d:0 s:1
[00:00:15.730,560] <inf> usbd_ch9: s-(out)-status finished
[00:00:15.784,301] <inf> usbd_ch9: Handle control 0x20018c94 ep 0x00, len 8, s:1 d:0 s:0
[00:00:15.785,308] <inf> usbd_ch9: Handle control 0x20018cb8 ep 0x00, len 8, s:1 d:0 s:0
[00:00:15.785,339] <inf> usbd_iface: Set Interfaces 1, alternate 0 -> 1
[00:00:15.785,400] <inf> usbd_iface: Modify interface 1 ep 0x88 by op 0 ep_bm 0
[00:00:15.785,430] <inf> usbd_iface: No endpoints, skip interface
[00:00:15.785,491] <inf> usbd_iface: Modify interface 1 ep 0x88 by op 1 ep_bm 1000000
[00:00:15.785,644] <inf> usbd_ch9: Handle control 0x20018c94 ep 0x80, len 0, s:0 d:0 s:1
[00:00:15.785,644] <inf> usbd_ch9: s-(out)-status finished
[00:00:23.287,017] <inf> stream_rx: [1000]: Incoming audio on stream 0x20005b20 len 40, flags 0x09, seq_num 1000 and ts 23019572: Valid 1000 | Error 0 | Loss 0 | Dup TS 00
[00:00:23.341,033] <inf> usb: [1000]: Adding USB audio frame
[00:00:23.341,644] <inf> usb: [1000]: Sending 1 USB audio frame
[00:00:23.374,176] <inf> usb: [10000]: Sending USB audio

```

It seems to only happen in the beginning of the stream. Unsure if the amount of logging hogs the CPU and makes it worse or not.

tmon-nordic, 2025-01-31:
This is bug in nRF UDC driver, caused because:
https://github.com/zephyrproject-rtos/zephyr/blob/77a350efa74ef6795d056392cb5001b917ff4f69/drivers/usb/udc/udc_nrf.c#L118-L119

submits the buffer to the stack but does not remove it from the queue. Then the very same buffer gets submitted again, but because the `net_buf` gets freed on its first handling, the `__buf` is NULL.

PR #84966 (2025-02-05): drivers: udc_nrf: Do not submit buffer more than once
When the UDC buffer gets submitted it should no longer reside in the endpoint queue. While this commit does not address the underlying issue of not being able to start transfer for whatever reason, it prevents the problem from cascading into buffer double completion (e.g. receive buffer double free in UAC2).

Fix: #84664
closed 2025-02-05 by kartben