zephyrproject-rtos/zephyr #84710: [Coverity CID: 487708] Operands don't affect result in subsys/bluetooth/controller/ll_sw/ull_sync_iso.c

zephyrbot, 2025-01-28:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/f0d0264c057255c60e91e422406f47a7b87d8a8a/subsys/bluetooth/controller/ll_sw/ull_sync_iso.c

Category: Integer handling issues
Function: `ull_sync_iso_setup`
Component: Bluetooth
CID: [487708](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=487708)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/f0d0264c057255c60e91e422406f47a7b87d8a8a/subsys/bluetooth/controller/ll_sw/ull_sync_iso.c#L503

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=487708

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_CONSTANT_EXPRESSION_RESULT). ([CWE-569](http://cwe.mitre.org/data/definitions/569.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

cvinayak, 2025-09-24:
> Marked as intentional in Coverity.

Re-opening, this needs fix...suggestion below:

```c
	if (lll->pto) {
		uint8_t nse;
		uint8_t ptc;

		nse = lll->irc * lll->bn; /* 4 bits * 3 bits, total 7 bits */
		if (nse >= lll->nse) {
			return;
		}

		ptc = lll->nse - nse;

		/* FIXME: Do not remember why ptc is 4 bits, it should be 5 bits as ptc is a
		 *        running buffer offset related to nse.
		 *        Fix ptc and ptc_curr definitions, until then we keep an assertion check
		 *        here.
		 */
		LL_ASSERT(ptc <= BIT_MASK(4));

		lll->ptc = ptc;
	} else {
		lll->ptc = 0U;
	}

```

cvinayak, 2025-09-25:
> Unless you're dying for it, I can take care of the PR.

No, you can take care (my response was just a natural instinct to fix my original stupidity ;-) ).

PR #96586 (2025-09-30): Bluetooth: Controller: fix assertion check for ptc value
Fix the issue reported by Coverity CID 487708. The LL_ASSERT was performed on lll->ptc, which is a 4-bit bitfield and therefore always succeeds. Instead, the computation (lll->nse - nse) should be checked to ensure it falls within the 4-bit value range before assigning it to lll->ptc.

Fixes: #84710