zephyrproject-rtos/zephyr #84737: eth_stm32_hal: Failed to obtain RX buffer - error makes device unreachable

javlands, 2025-01-28:
**Describe the bug**

From the moment the `eth_stm32_hal: Failed to obtain RX buffer` error is logged, the device is not reachable anymore on the network (i.e. not pingable anymore). A reboot is required to get the device back up and running again. The device uses an stm32h573. The error does not seem to appear at a consistent interval.

**To Reproduce**

I did not find a way yet to consistently reproduce the error. However, limiting `CONFIG_NET_PKT_RX_COUNT` to a low number (e.g., 2) makes the error pop up more easily.

**Expected behavior**

The device recovers from this error and is still reachable on the network.

**Impact**

A reboot is needed, which can only be done via a physical button or a shell command in our case, which is quite disturbing for devices in the field.

**Logs and console output**

```
uart:~$ [00:06:53.095,000] <dbg> net_pkt: pkt_alloc_with_buffer: (stm_eth): On iface 0x20001638 size 60
uart:~$ [00:06:53.095,000] <dbg> net_pkt: net_pkt_unref_debug: (rx_q[0]): RX [1] pkt 0x20035d68 ref 0 frags 0x2003585c (processing_data():174)
uart:~$ [00:06:53.095,000] <dbg> net_pkt: net_pkt_unref_debug: (rx_q[0]): RDATA (?) [0] frag 0x2003585c ref 0 frags 0 (processing_data():174)
uart:~$ [00:06:53.095,000] <dbg> net_pkt: net_pkt_print: (rx_q[0]): TX 14 RX 2 RDATA 0 TDATA 0
uart:~$ [00:06:53.098,000] <dbg> net_pkt: pkt_alloc_with_buffer: (stm_eth): On iface 0x20001638 size 60
uart:~$ [00:06:53.098,000] <dbg> net_pkt: net_pkt_unref_debug: (rx_q[0]): RX [1] pkt 0x20035d68 ref 0 frags 0x2003585c (processing_data():174)
uart:~$ [00:06:53.098,000] <dbg> net_pkt: net_pkt_unref_debug: (rx_q[0]): RDATA (?) [0] frag 0x2003585c ref 0 frags 0 (processing_data():174)
uart:~$ [00:06:53.098,000] <dbg> net_pkt: net_pkt_print: (rx_q[0]): TX 14 RX 2 RDATA 0 TDATA 0
uart:~$ [00:06:53.175,000] <dbg> net_pkt: pkt_alloc_with_buffer: (stm_eth): On iface 0x20001638 size 60
uart:~$ [00:06:53.175,000] <dbg> net_pkt: net_pkt_unref_debug: (rx_q[0]): RX [1] pkt 0x20035d68 ref 0 frags 0x2003585c (processing_data():174)
uart:~$ [00:06:53.175,000] <dbg> net_pkt: net_pkt_unref_debug: (rx_q[0]): RDATA (?) [0] frag 0x2003585c ref 0 frags 0 (processing_data():174)
uart:~$ [00:06:53.175,000] <dbg> net_pkt: net_pkt_print: (rx_q[0]): TX 14 RX 2 RDATA 0 TDATA
uart:~$ [00:06:53.210,000] <dbg> net_pkt: pkt_alloc_with_buffer: (stm_eth): On iface 0x20001638 size 108
uart:~$ [00:06:53.210,000] <dbg> net_pkt: net_pkt_unref_debug: (rx_q[0]): RX [1] pkt 0x20035d68 ref 0 frags 0x2003585c (processing_data():174)
uart:~$ [00:06:53.210,000] <dbg> net_pkt: net_pkt_unref_debug: (rx_q[0]): RDATA (?) [0] frag 0x2003585c ref 0 frags 0 (processing_data():174)
uart:~$ [00:06:53.210,000] <dbg> net_pkt: net_pkt_print: (rx_q[0]): TX 14 RX 2 RDATA 0 TDATA 0
uart:~$ [00:06:53.210,000] <dbg> net_pkt: pkt_alloc_with_buffer: (stm_eth): On iface 0x20001638 size 136
uart:~$ [00:06:53.210,000] <dbg> net_pkt: net_pkt_unref_debug: (rx_q[0]): RX [1] pkt 0x20035d68 ref 0 frags 0x2003585c (processing_data():174)
uart:~$ [00:06:53.210,000] <dbg> net_pkt: net_pkt_unref_debug: (rx_q[0]): RDATA (?) [0] frag 0x2003585c ref 0 frags 0x20035610 (processing_data():174)
uart:~$ [00:06:53.210,000] <dbg> net_pkt: net_pkt_unref_debug: (rx_q[0]): RDATA (?) [0] frag 0x20035610 ref 0 frags 0 (processing_data():174)
uart:~$ [00:06:53.210,000] <dbg> net_pkt: net_pkt_print: (rx_q[0]): TX 14 RX 2 RDATA 0 TDATA 0
uart:~$ [00:06:53.210,000] <dbg> net_pkt: pkt_alloc_with_buffer: (stm_eth): On iface 0x20001638 size 167
uart:~$ [00:06:53.210,000] <dbg> net_pkt: pkt_alloc_with_buffer: (stm_eth): On iface 0x20001638 size 60
uart:~$ [00:06:53.210,000] <dbg> net_pkt: pkt_alloc_with_buffer: (stm_eth): On iface 0x20001638 size 95
uart:~$ [00:06:53.210,000] <dbg> net_pkt: net_pkt_unref_debug: (rx_q[0]): RX [0] pkt 0x20035d68 ref 0 frags 0x20035610 (processing_data():174)
uart:~$ [00:06:53.210,000] <dbg> net_pkt: net_pkt_unref_debug: (rx_q[0]): RDATA (?) [0] frag 0x20035610 ref 0 frags 0x2003585c (processing_data():174)
uart:~$ [00:06:53.210,000] <dbg> net_pkt: net_pkt_unref_debug: (rx_q[0]): RDATA (?) [0] frag 0x2003585c ref 0 frags 0 (processing_data():174)
uart:~$ [00:06:53.210,000] <dbg> net_pkt: net_pkt_print: (rx_q[0]): TX 14 RX 0 RDATA 0 TDATA 0
uart:~$ [00:06:53.210,000] <dbg> net_pkt: net_pkt_unref_debug: (rx_q[0]): RX [0] pkt 0x20035d2c ref 0 frags 0x20035824 (processing_data():174)
uart:~$ [00:06:53.210,000] <dbg> net_pkt: net_pkt_unref_debug: (rx_q[0]): RDATA (?) [0] frag 0x20035824 ref 0 frags 0 (processing_data():174)
uart:~$ [00:06:53.210,000] <dbg> net_pkt: net_pkt_print: (rx_q[0]): TX 14 RX 1 RDATA 0 TDATA 0
uart:~$ [00:06:53.210,000] <dbg> net_pkt: pkt_alloc_with_buffer: (stm_eth): On iface 0x20001638 size 102
uart:~$ [00:06:53.210,000] <dbg> net_pkt: pkt_alloc_with_buffer: (stm_eth): On iface 0x20001638 size 60
uart:~$ [00:06:53.210,000] <dbg> net_pkt: pkt_alloc_with_buffer: (rx_q[0]): On iface 0x20001638 size 81
uart:~$ [00:06:53.310,000] <err> eth_stm32_hal: Failed to obtain RX buffer
uart:~$ [00:06:53.310,000] <dbg> net_pkt: net_pkt_unref_debug: (rx_q[0]): RX [0] pkt 0x20035d68 ref 0 frags 0x20035824 (processing_data():174)
uart:~$ [00:06:53.310,000] <dbg> net_pkt: net_pkt_unref_debug: (rx_q[0]): RDATA (?) [0] frag 0x20035824 ref 0 frags 0 (processing_data():174)
uart:~$ [00:06:53.310,000] <dbg> net_pkt: net_pkt_print: (rx_q[0]): TX 14 RX 1 RDATA 0 TDATA 0
uart:~$ [00:06:53.310,000] <dbg> net_pkt: net_pkt_unref_debug: (rx_q[0]): RX [1] pkt 0x20035d2c ref 0 frags 0x2003585c (processing_data():174)
uart:~$ [00:06:53.310,000] <dbg> net_pkt: net_pkt_unref_debug: (rx_q[0]): RDATA (?) [0] frag 0x2003585c ref 0 frags 0 (processing_data():174)
uart:~$ [00:06:53.310,000] <dbg> net_pkt: net_pkt_print: (rx_q[0]): TX 14 RX 2 RDATA 0 TDATA 0
```

**Environment (please complete the following information):**

 - OS: WSL
 - Toolchain: zephyr-sdk-0.16.8
 - Commit SHA: currently checked out at 53738cf9247

**Additional context**

javlands, 2025-01-28:
Update: to easily reproduce the error:
 - `CONFIG_NET_PKT_RX_COUNT=1`
 - In [eth_stm32_hal.c:623](https://github.com/zephyrproject-rtos/zephyr/blob/b80a058e9043087f2a114310df48eab4638598b6/drivers/ethernet/eth_stm32_hal.c#L623), replace `K_MSEC(100)` with `K_NO_WAIT`
 - Run any networking related application

javlands, 2025-01-30:
Update:

After some deeper investigation, we noticed that no ethernet rx interupts are received anymore (i.e `eth_isr` is only called for outgoing packets) after the error is logged. This makes it that the `dev_data->rx_int_sem` semaphore always times out in the `rx_thread`

```
...
[00:00:06.697,000] <inf> eth_stm32_hal: eth_isr
[00:00:06.697,000] <inf> eth_stm32_hal: res == 0
[00:00:06.740,000] <inf> eth_stm32_hal: eth_isr
[00:00:06.740,000] <inf> eth_stm32_hal: res == 0
[00:00:06.748,000] <inf> eth_stm32_hal: eth_isr
[00:00:06.748,000] <inf> eth_stm32_hal: res == 0
[00:00:06.773,000] <err> eth_stm32_hal: Failed to obtain RX buffer
[00:00:07.273,000] <inf> eth_stm32_hal: res == -EAGAIN
[00:00:07.774,000] <inf> eth_stm32_hal: res == -EAGAIN
[00:00:08.274,000] <inf> eth_stm32_hal: res == -EAGAIN
[00:00:08.774,000] <inf> eth_stm32_hal: res == -EAGAIN
...
```

 The issue is easily reproducibleon the stm32h573i_dk devkit board using the method described above.

marwaiehm-st, 2025-03-18:
I tested with https://github.com/zephyrproject-rtos/zephyr/pull/84965 on the board stm32h573i_dk,
 Now The device is reachable on the network even after the `eth_stm32_hal: Failed to obtain RX buffer error` is logged.
@javlands do you confirm this ?

marwaiehm-st, 2025-03-19:
Resolved https://github.com/zephyrproject-rtos/zephyr/pull/84965

closed 2025-03-19 by marwaiehm-st