zephyrproject-rtos/zephyr #84754: [Coverity CID: 487634] Overflowed constant in arch/arm/core/mpu/arm_mpu_v7_internal.h

zephyrbot, 2025-01-28:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/f0d0264c057255c60e91e422406f47a7b87d8a8a/arch/arm/core/mpu/arm_mpu_v7_internal.h

Category: Integer handling issues
Function: `mpu_configure_static_mpu_regions`
Component: Architectures
CID: [487634](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=487634)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/f0d0264c057255c60e91e422406f47a7b87d8a8a/arch/arm/core/mpu/arm_mpu_v7_internal.h#L236

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=487634

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_INTEGER_OVERFLOW). ([CWE-190](http://cwe.mitre.org/data/definitions/190.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

wearyzen, 2025-02-07:
This issue is a False positive. Explanation:
As the scan shows there is an overflow reported at [this line](https://github.com/zephyrproject-rtos/zephyr/blob/786cd7fdd87c45718752ef1b5e50746c553e05cf/arch/arm/core/mpu/arm_mpu_v7_internal.h#L236) when mpu_configure_regions returns -EINVAL however, [mpu_configure_static_mpu_regions](https://github.com/zephyrproject-rtos/zephyr/blob/786cd7fdd87c45718752ef1b5e50746c553e05cf/arch/arm/core/mpu/arm_mpu_v7_internal.h#L220C12-L220C44) is only called by arm_core_mpu_configure_static_mpu_regions() [here](https://github.com/zephyrproject-rtos/zephyr/blob/786cd7fdd87c45718752ef1b5e50746c553e05cf/arch/arm/core/mpu/arm_mpu.c#L365) and there is a check in place for -EINVAL.

I have a PR #84863 which can close this coverity issue by adding a check but I as @kartben suggested if we are sure this is false positive it is better to avoid that change and mark this as false positive.

So, I tried to mark this as false positive on this https://scan9.scan.coverity.com/#/project-view/62006/12996?selectedIssue=487634 but the  "Classification" field and "Apply+Next"/"Apply" buttons are greyed out. I may not have the right permissions to mark this as false positive so requesting help from others on how to mark this as false positive.

kartben, 2025-02-07:
marked as false positive in Coverity. @nashif might be able to comment on why you didn't have permissions to do it yourself

Thanks!

closed 2025-02-07 by kartben