zephyrproject-rtos/zephyr #84783: [Coverity CID: 444378] Macro compares unsigned to 0 in drivers/video/video_common.c

zephyrbot, 2025-01-28:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/f0d0264c057255c60e91e422406f47a7b87d8a8a/drivers/video/video_common.c

Category: Integer handling issues
Function: `video_closest_frmival_stepwise`
Component: Drivers
CID: [444378](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=444378)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/f0d0264c057255c60e91e422406f47a7b87d8a8a/drivers/video/video_common.c#L123

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=444378

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_NO_EFFECT). ([CWE-570](http://cwe.mitre.org/data/definitions/570.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

josuah, 2025-01-28:
I'll need to gather some time first but will do my best to get this done timely.
I am to `git blame` for multiple of these.

josuah, 2025-02-13:
I believe this one can be acknowledged, as it is unhappy about the `< 0` in this macro:

https://github.com/zephyrproject-rtos/zephyr/blob/1f25d9d9b99b788b04b8c694009c6e4e5eeab8b8/include/zephyr/sys/util.h#L369-L372

The other two are addressed through https://github.com/zephyrproject-rtos/zephyr/pull/85697

Some people do advent of code, I prefer doing Coverity CIDs. :)