zephyrproject-rtos/zephyr #84786: [Coverity CID: 444374] Logically dead code in drivers/mfd/mfd_npm2100.c

zephyrbot, 2025-01-28:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/f0d0264c057255c60e91e422406f47a7b87d8a8a/drivers/mfd/mfd_npm2100.c

Category: Control flow issues
Function: `config_shphold`
Component: Drivers
CID: [444374](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=444374)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/f0d0264c057255c60e91e422406f47a7b87d8a8a/drivers/mfd/mfd_npm2100.c#L222

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=444374

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_DEADCODE). ([CWE-561](http://cwe.mitre.org/data/definitions/561.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

gmarull, 2025-02-03:
@seov-nordic

seov-nordic, 2025-02-03:
fixed in #83639 (commit 2c9a92437634021400a70ee66d5d534128a3a2c0)

closed 2025-02-03 by gmarull