zephyrproject-rtos/zephyr #84797: [Coverity CID: 399568] Structurally dead code in subsys/shell/shell.c

zephyrbot, 2025-01-28:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/f0d0264c057255c60e91e422406f47a7b87d8a8a/subsys/shell/shell.c

Category: Control flow issues
Function: `state_collect`
Component: Shell
CID: [399568](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=399568)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/f0d0264c057255c60e91e422406f47a7b87d8a8a/subsys/shell/shell.c#L1160

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=399568

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_UNREACHABLE). ([CWE-561](http://cwe.mitre.org/data/definitions/561.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

michalpiekos, 2025-02-02:
My understanding is that `z_transport_buffer_flush(sh);` should be within the scope of while loop to flush after each handling of `sh->ctx->receive_state`, meaning between current line 1157 and 1158.
If you agree with my thinking I will create PR here.

jakub-uC, 2025-02-06:
let me digg in

jakub-uC, 2025-02-20:
@michalpiekos No, there is no need to flush after each iteration of the while loop. We buffer the data and flush it either after the loop completes or during the loop when the buffer is full.

The issue arose after I added a Kconfig option for the autoflush functionality. I will look into fixing it.

closed 2025-03-07 by carlescufi