zephyrproject-rtos/zephyr #84805: [Coverity CID: 347195] Untrusted loop bound in drivers/gpio/gpio_ite_it8xxx2_v2.c

zephyrbot, 2025-01-28:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/f0d0264c057255c60e91e422406f47a7b87d8a8a/drivers/gpio/gpio_ite_it8xxx2_v2.c

Category: Insecure data handling
Function: `gpio_ite_isr`
Component: Drivers
CID: [347195](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=347195)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/f0d0264c057255c60e91e422406f47a7b87d8a8a/drivers/gpio/gpio_ite_it8xxx2_v2.c#L383

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=347195

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_TAINTED_SCALAR). ([CWE-606](http://cwe.mitre.org/data/definitions/606.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

GTLin08, 2025-02-03:
Hi @fabiobaltieri ,
Sorry to bother you, but I tried accessing the provided link and received a "401: Unauthorized" error, which means I don’t have the necessary permissions to view the details of this issue.
Could you check if you have access and provide more information on the specific problem? Thanks!

jhedberg, 2025-02-03:
Does this help?
[image]

PR #85113 (2025-02-08): drivers/gpio: ite: Validate num_pins to prevent untrusted loop bound
Added a assert to limit num_pins to a maximum allowable value to prevent potential out-of-bounds access or infinite loops.

Fixes: #84805
closed 2025-02-08 by kartben