zephyrproject-rtos/zephyr #84811: [Coverity CID: 321152] Use of 32-bit time_t in drivers/counter/maxim_ds3231.c

zephyrbot, 2025-01-28:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/f0d0264c057255c60e91e422406f47a7b87d8a8a/drivers/counter/maxim_ds3231.c

Category: High impact quality
Function: `sync_prep_write`
Component: Drivers
CID: [321152](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=321152)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/f0d0264c057255c60e91e422406f47a7b87d8a8a/drivers/counter/maxim_ds3231.c#L854

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=321152

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_Y2K38_SAFETY). ([CWE-197](http://cwe.mitre.org/data/definitions/197.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

michalpiekos, 2025-02-02:
I would like to help with this and similar issues with 32-bit time_t. In order to do so I have a couple of questions.
- time_t definition comes from std C therefore Zephyr cannot really enforce it to be defined as 64-bit. True/False?
- We should change time_t type to int64_t or is there some other acceptable solution?

sudarsan-22, 2025-06-09:
> I would like to help with this and similar issues with 32-bit time_t. In order to do so I have a couple of questions.
>
>     * time_t definition comes from std C therefore Zephyr cannot really enforce [...]

time_t is defined by the C standard (<time.h>) as an arithmetic type (typically signed integer), but its width is implementation-defined.

On many 32-bit systems, especially those using newlib, time_t is 32-bit, which overflows in 2038
Zephyr cannot safely rely on time_t being 64-bit unless the toolchain guarantees it. So, using int64_t or uint64_t explicitly is the safe and portable approach.