zephyrproject-rtos/zephyr #84813: [Coverity CID: 321117] Use of 32-bit time_t in drivers/counter/maxim_ds3231.c

zephyrbot, 2025-01-28:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/f0d0264c057255c60e91e422406f47a7b87d8a8a/drivers/counter/maxim_ds3231.c

Category: High impact quality
Function: `ds3231_counter_get_value`
Component: Drivers
CID: [321117](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=321117)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/f0d0264c057255c60e91e422406f47a7b87d8a8a/drivers/counter/maxim_ds3231.c#L764

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=321117

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_Y2K38_SAFETY). ([CWE-197](http://cwe.mitre.org/data/definitions/197.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

aescolar, 2025-06-10:
These 4 are related / almost dups.
* https://github.com/zephyrproject-rtos/zephyr/issues/84809
* https://github.com/zephyrproject-rtos/zephyr/issues/84811
* https://github.com/zephyrproject-rtos/zephyr/issues/84812
* https://github.com/zephyrproject-rtos/zephyr/issues/84813

Note that independently of the glibc defined time_t size, the code itself is masking the time_t value to the lowest 32bits in several places.
So this is not just, like quite a few other time_t related Coverity reports, about using time_t and time_t being possible only 32bits.

aescolar, 2025-06-28:
@michalpiekos @sudarsan-22 Note that independently of the size of time_t,
in several places in the code, (i.e. https://github.com/zephyrproject-rtos/zephyr/blob/f0d0264c057255c60e91e422406f47a7b87d8a8a/drivers/counter/maxim_ds3231.c#L1233 )
The code is masking the value to 32bits:
https://github.com/zephyrproject-rtos/zephyr/blob/f0d0264c057255c60e91e422406f47a7b87d8a8a/drivers/counter/maxim_ds3231.c#L34

closed 2025-07-03 by zephyrbot