zephyrproject-rtos/zephyr #84828: Build sensor sample using uarte with nrf52dk/nrf52832 failed

afontaine-invn, 2025-01-28:
**Describe the bug**

Building sensor sample with nrf52dk/nrf52832 platform FAILED

>west build -p always -b nrf52dk/nrf52832 samples\sensor\accel_polling
[...]
C:/Data/Zephyr/zephyrproject/zephyr/drivers/serial/uart_nrfx_uarte.c:2418:49: error: 'NRF_UARTE_BASE_FREQUENCY_16MHZ' undeclared here (not in a function); did you mean 'NRF_TIMER_BASE_FREQUENCY_16MHZ'?
 2418 |         (NRF_PERIPH_GET_FREQUENCY(UARTE(idx)) / NRF_UARTE_BASE_FREQUENCY_16MHZ)
      |                                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

 - What target platform are you using?
Target platform : nrf52dk/nrf52832

 - What have you tried to diagnose or workaround this issue?
Revert latest change in zephyr/drivers/serial/uart_nrfx_uarte.c:

 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
This is a regression from the commit :
https://github.com/zephyrproject-rtos/zephyr/commit/e9f9cc605a7858a31196bc45006758186f0a27c8

**To Reproduce**

1.>west build -p always -b nrf52dk/nrf52832 samples\sensor\accel_polling
2. See error

**Expected behavior**

Build passed

**Impact**

All build faield for the platform I used

**Logs and console output**

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**

fabiobaltieri, 2025-01-28:
Hey can you make sure that the modules are up to date? I.e. run `west update` before the build

carlescufi, 2025-01-30:
Cannot reproduce locally, closing. @afontaine-invn please reopen if you think this is a real issue.

closed 2025-01-30 by carlescufi