zephyrproject-rtos/zephyr #84845: ADC in STM32L412KB (nucleo-L412KB) not working correctly

robus814, 2025-01-29:
**Describe the bug**
The adc on the controller returns always a nearly fixed value of around "80" - corresponding to ~62mV (see console output below). Pulling the pin down to 0V or 3.3V does not change the ADC value. The ADC is set up correctly (without error), switching the adc to other pins shows the same result. A second nucleo-L412KB was tested to rule out a hardware error, the result was the same. The software (except overlay) was tested successfully on STM32L476. 
It seems to be an error in the ADC driver configuration of the STM32L412 controller. Maybe the internal reference is not set correctly? 

**To Reproduce**
1. Use sample application "samples/boards/st/power_mgmt/adc"
2. Disable PM related Configs in prj.conf, as lp_tick_source ist not defined in basic boards dts
3. Add overlay to the project: ("nucleo_l412rb_p.overlay")

````
#include <zephyr/dt-bindings/adc/adc.h>

/ {
        zephyr,user {
                /* adjust channel number according to pinmux in board.dts */
                io-channels = <&adc1 1>;
        };
};

&adc1 {
        #address-cells = <1>;
        #size-cells = <0>;

        channel@1 {
                reg = <1>;
                zephyr,gain = "ADC_GAIN_1";
                zephyr,reference = "ADC_REF_INTERNAL";
                zephyr,vref-mv = <3300>;
                zephyr,acquisition-time = <ADC_ACQ_TIME_DEFAULT>;
                zephyr,resolution = <12>;
        };
};
````
3. Build with <`west build -p nucleo_l412rb_p .`>
4. Flash 

**Expected behavior**
The SW should print out the correct ADC value. 

**Logs and console output**
````
*** Booting Zephyr OS build v4.0.0-3996-g3585c3743710 ***
Device ready
ADC reading[0]:
- adc@50040000, channel 1: 400 = 322 mV
ADC reading[1]:
- adc@50040000, channel 1: 80 = 64 mV
ADC reading[2]:
- adc@50040000, channel 1: 79 = 63 mV
ADC reading[3]:
- adc@50040000, channel 1: 77 = 62 mV
ADC reading[4]:
- adc@50040000, channel 1: 79 = 63 mV
ADC reading[5]:
- adc@50040000, channel 1: 78 = 62 mV
ADC reading[6]:
- adc@50040000, channel 1: 79 = 63 mV
ADC reading[7]:
- adc@50040000, channel 1: 79 = 63 mV
ADC reading[8]:
- adc@50040000, channel 1: 78 = 62 mV
ADC reading[9]:
- adc@50040000, channel 1: 80 = 64 mV
ADC reading[10]:
- adc@50040000, channel 1: 78 = 62 mV
ADC reading[11]:
- adc@50040000, channel 1: 79 = 63 mV
ADC reading[12]:
- adc@50040000, channel 1: 78 = 62 mV
```

gautierg-st, 2025-01-29:
Hi @robus814,

Which pin are you using for the ADC on L412? The default one in the board dts is `adc1_in5_pa0`, i.e. channel 5. So the overlay should use `io-channels = <&adc1 5>;` and
```
channel@5 {
                reg = <5>;
```

Looking into it, I realize that there is also an inconsistency between the board dts and the overlay of tests/drivers/adc/adc_api...

On another note, please also be aware that L412 ADC has fast and slow channels. Channels 1 to 5 are fast, the others are slow. You may need to increase the sampling time with the slower channels if you use them.

robus814, 2025-01-29:
Hi Guillaume, 

thank you very much, your advice to change the channel to the dedicated input channel defined in the pinmux worked. ADC returns now valid values.

closed 2025-01-29 by robus814