zephyrproject-rtos/zephyr #84856: Take out modbus floating-point extension (CONFIG_MODBUS_FP_EXTENSIONS)

gatzka, 2025-01-29:
## Introduction

The current modbus implementation contains code which is non-conformant to the latest modbus specification.

### Problem description
The current modbus specification (https://modbus.org/docs/Modbus_Application_Protocol_V1_1b3.pdf)
nowhere mentions the handling of floating point values. The modbus code has specific callbacks for handling floating point values
and also has a hard-coded value for addresses where floating point holding registers should be (MODBUS_FP_EXTENSIONS_ADDR in modbus_internal.h).

This introduces unnecessary dependencies to floating point support. In addition, this vendor-specific floating-point extension increases the maintenance effort of the zephyr modbus code.

### Proposed change
Strip down the zephyr modbus code to be conformant to the modbus specification.

Vendor-specific implementations of floating point accesses can easily made in the application layer (the user-defined callbacks).

## Detailed RFC
Remove all dependencies to CONFIG_MODBUS_FP_EXTENSIONS in the zephyr code base.

### Proposed change (Detailed)

Take out the Kconfig option CONFIG_MODBUS_FP_EXTENSIONS and the
code relying on it. This change will affect subsys/modbus/modbus_client.c,
subsys/modbus/modbus_internal.h and subsys/modbus/modbus_server.c

In general, this change will the code cleaner, easier to follow, to maintain and test.

All code sections in modbus_server.c using MODBUS_FP_EXTENSIONS (mbs_fc03_hreg_read(), mbs_fc04_inreg_read(), mbs_fc16_hregs_write()) will be cleaned up.

The same will happen to modbus_client.c (modbus_read_holding_regs_fp(), modbus_write_holding_regs_fp() and mbc_validate_rd_response()).

### Dependencies
There are no dependencies to other areas of the zephyr code base.

### Concerns and Unresolved Questions
This is a breaking change. Applications relying on on this API will fail after this change.

## Alternatives
Currently, I see no alternatives.

gatzka, 2025-01-29:
> What exactly is non-conformity? Whether the extension is enabled or not, on the layer below, the implementation uses 16-bit words. No one has to enable it if they do not want to. For me, this is no reason to remove it. But yes, it can also be handled by the user in specified register command callbacks.

Well, non-conformity is probably the wrong wording.
Here are my main 3 arguments:

- This CONFIG_MODBUS_FP_EXTENSIONS is completely end user vendor-specific and therefore should not be in universally applicable code
- There are unnecessary callbacks in struct modbus_user_callbacks (input_reg_rd_fp, holding_reg_rd_fp, holding_reg_wr_fp) making the structure bigger than necessary
- It simplifies code maintenance. Actually, the PR I made (https://github.com/zephyrproject-rtos/zephyr/pull/83963/commits/7d52b975f8808b89d0c5b08775cae83d1483c615) was necessary only just because of this. The err variable would have been initialized always correctly _without_ that CONFIG_MODBUS_FP_EXTENSIONS

lindblandro, 2025-02-03:
I also thought about the necessity of the FP extension recently, I think I noticed the same uninitialized variable warning as @gatzka.
 
Maybe the current implementation could use some overall refactoring as most of the register read / write functions in `modbus_server.c` are nearly identical, which is probably the real reason the uninitialized error variable went unnoticed for so long in the first place. Multi-register value reads are probably something that isn't going away and currently they are  a bit cumbersome as the register data might change between calls to the read callbacks, requiring synchronization in the user callback, which can be a little tricky to implement reliably.

I've been toying around the idea that the modbus subsystem would contain a multi-register word order configuration (ABCD, BADC, DCBA, CDAB and so on,  sort of how [libmodbus float API](https://libmodbus.org/reference/modbus_get_float/) is designed),  and a bunch of macros that honor the configured word order, such as `MODBUS_PACK_F32(), MODBUS_PACK_U64(), MODBUS_PACK_STR()`. Instead of calling the register read / write callback for each address, the callback API would just contain something like
```c
int modbus_register_read_cb(uint16_t start_addr, uint8_t qty, enum modbus_register_type type, uint8_t *buf);
```
The distinction between input / holding / coil registers probably was more meaningful at some point in history (special hardware for selecting memories or something), but I don't see why the type couldn't be just a namespace in modern software. The user code could pack the application specific values in that address range using the configured macros.

Not very polished, but something along those lines could support floating point values as well as other, weirder types users have decided to implement.

gatzka, 2025-02-03:
Thanks @lindblandro . I also think that the modbus code needs some refactoring. In addition to the issues I already mentioned I see more problems as well.

For instance, I don't like the implementation of "Write Multiple Registers" (FC16). The code in modbus_server.c iterates itself over the quantity of registers and is calling the user callback holding_reg_wr(). This prevents the implementation of transactions and/or checking the address range upfront _before_ really changing the state of the modbus server, which is mandated by the modbus specification.

Anyhow, I suggest to do this kind of changes incrementally, by first taking out code which is not really helpful or even harmful and then start the refactoring by defining the interface etc.

lindblandro, 2025-02-03:
Yep, sounds like we're on the same page.

lindblandro, 2025-02-06:
See also https://github.com/zephyrproject-rtos/zephyr/issues/73237