zephyrproject-rtos/zephyr #84876: build system: add a linter and github action for cmake files

cfriedt, 2025-01-29:
**Is your enhancement proposal related to a problem? Please describe.**

There are several CMake "coding style" issues that exist today in Zephyr, and many new PRs also tend to introduce new issues if they are not caught by reviewers. However, only a handful of reviewers actually make change requests on PRs for CMake formatting, and when that happens, a PR that would otherwise be passing all tests needs to go through the round-trip-time of another revision by the original developer, lots of wasted time in CI, and then finally another review by the reviewer who requested CMake formatting changes.

**Describe the solution you'd like**
Like most / many other source file and metadata file types, add a linter for CMake formatting. The linter should run on-diff. We should also be able to use the same tool to reformat all `CMakeLists.txt` files or `*.cmake` files in-place.

**Describe alternatives you've considered**
Seeing the same `if (CONFIG_FOO)` -> `if(CONFIG_FOO)` change requests being made over, and over again by one or two reviewers.

**Additional context**

nordicjm, 2025-01-30:
The problem having tried them is... unfortunately they all suck, even the CMake maintainers say none of them are worth using. So each of them gets some things right, gets some things wrong, and some even get some things so bad that the file is incomprehensible compared to how it started out

cfriedt, 2025-01-31:
KitWare (or whatever the company is) should just add this type of thing to clang-format