zephyrproject-rtos/zephyr #84898: drivers: ethernet: eth_stm32_hal: support `get_config`

clamattia, 2025-01-30:
**Is your enhancement proposal related to a problem? Please describe.**
There is currently no good way to query the configuration of the Ethernet on this driver (to my knowledge).

**Describe the solution you'd like**
Please implement `int(* 	get_config )(const struct device *dev, enum ethernet_config_type type, struct ethernet_config *config)` for the `eth_stm32_hal`-driver.

**Describe alternatives you've considered**
None

**Additional context**
None

ramram885, 2025-02-07:
Hello community,

I'm just new to zephyr and open source software development. Recently started to study about network drivers development too.

In zephyr project, I could see that many of the ethernet drivers have implemented only [set_config()](https://docs.zephyrproject.org/apidoc/latest/structethernet__api.html#ae204fdf7e8c72fdea3dee67a7068afe1) function in the driver code. 
I couldn't find much drivers that has implemented [get_config()](https://docs.zephyrproject.org/apidoc/latest/structethernet__api.html#a3f71e6bf922b91289efa3ac97df70e81) function. Can any one please explain reason behind this.

And I could see that many drivers have implementation for [get_capabilities()](https://docs.zephyrproject.org/apidoc/latest/structethernet__api.html#a8731846f9bd07398b2f5c154c6ec0fe3). Is the [get_capabilities()](https://docs.zephyrproject.org/apidoc/latest/structethernet__api.html#a8731846f9bd07398b2f5c154c6ec0fe3) function itself enough to fulfill the implementation of [get_config()?](https://docs.zephyrproject.org/apidoc/latest/structethernet__api.html#a3f71e6bf922b91289efa3ac97df70e81). Because I am thinking that, If we implement the get_config(), the functionality will be more or less same as get_capabilities().
Is the understanding is correct or not?

Thanks!

erwango, 2025-02-17:
@ramram885 Please open a dedicated discussion for you questions or alternatively engage on Zephyr Discord the network channel.

PR #86315 (2025-03-19): drivers: ethernet: stm32: Add get_config() support
Implementation of get_config() api added
Fixes: #84898
commit 66b2a6ad (2025-03-19): drivers: ethernet: stm32: Add get_config() support

Implementation of get_config() api added
Fixes: #84898
closed 2025-03-19 by kartben