zephyrproject-rtos/zephyr #84914: application_development: software_bill_of_materials: sbom.spdx fails

katgiadla, 2025-01-30:
**Describe the bug**
Sample `tests/application_development/software_bill_of_materials/sbom.spdx` reports timeout.

Observed for
- `nrf9160dk@0.14.0/nrf9160/ns`
- `nrf5340dk/nrf5340/cpuapp/ns`

**To Reproduce**
Steps to reproduce the behavior:
1. have `nrf5340dk/nrf5340` connected
2. go to your zephyr dir
3. call `./scripts/twister -s sbom.spdx -p nrf5340dk/nrf5340/cpuapp/ns --device-testing --device-serial /dev/ttyACM1 -v --inline-logs --west-flash="--erase"`
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
[OUTPUT] 
Internal ctest changing into directory: zephyr/zephyr/twister-out/nrf5340dk_nrf5340_cpuapp_ns/zephyr/tests/application_development/software_bill_of_materials/sbom.spdx

[HANDLER_OUTPUT] 
Test project zephyr/zephyr/twister-out/nrf5340dk_nrf5340_cpuapp_ns/zephyr/tests/application_development/software_bill_of_materials/sbom.spdx

    Start 1: spdx_gen
1/1 Test #1: spdx_gen .........................***Failed    0.35 sec
parsing CMake Cache file
ERROR: CONFIG_BUILD_OUTPUT_META must be enabled to generate spdx files; bailing
ERROR: SPDX walker failed; bailing
FATAL ERROR: Failed to create SPDX output

0% tests passed, 1 tests failed out of 1

Total Test time (real) =   0.35 sec

The following tests FAILED:
  1 - spdx_gen (Failed)
[ERROR_MESSAGE] 
Errors while running CTest
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain: Zephyr SDK 0.17.0
 - Commit SHA or Version used:  `v4.0.0-3405-g8cf014`
 
**Additional information**
Bisection does not find out the cause of issue.

pdgendt, 2025-01-30:
The test shouldn't run on an actual device, it verifies meta data output.

carlescufi, 2025-01-30:
@nordicjm if this is not applicable to the hw platforms mentioned in the description, please close the issue.

nordicjm, 2025-02-03:
Test should not run on actual hardware, is a build-only test

closed 2025-02-03 by nordicjm