zephyrproject-rtos/zephyr #84960: net: phase out use of `net_sprint_ll_addr` in favor of save alternatives

clamattia, 2025-01-31:
**Is your enhancement proposal related to a problem? Please describe.**
The function `net_sprint_ll_addr` is a data-race, due to the single static buffer.

**Describe the solution you'd like**
- Replace all use of `net_sprint_ll_addr` with safer alternatives.
There are safer alternatives, that use a caller provided buffer. For example `net_sprint_ll_addr_buf`. We could also consider a convenience function of the sort:
```C
char *net_sprint_ll_addr(const struct net_linkaddr *lladr, char *buf, size_t buf_size) {
    return net_sprint_ll_addr_buf(lladr->addr, lladr->len, buf, size);
}
```

**Describe alternatives you've considered**
- Only provide `net_sprint_ll_addr` if `CONFIG_MULTITHREADING` is not enabled.
- Make it very clear, that it is an unsafe function. For example by renaming it to `net_sprint_ll_addr_UNSAFE`.
- Provide build and/or runtime warnings if the function is used in combination with `CONFIG_MULTITHREADING`.

**Additional context**
None

jukkar, 2025-01-31:
Couple of notes:
* the said unsafe function is private network stack internal function that is only used for debug prints
* changing all the usage of this would mean increasing the stack memory consumption when debug printing is enabled
* if the current functionality is causing problems (I have not seen evidence for it), we could increase the amount of static buffers in `net_sprint_ll_addr()`, so same way as `net_sprint_addr()` is doing

clamattia, 2025-01-31:
> * if the current functionality is causing problems (I have not seen evidence for it), we could increase the amount of static buffers in `net_sprint_ll_addr()`, so same way as `net_sprint_addr()` is doing

I was not aware of `net_sprint_addr`. Should it not also be phased out in favor safer functions with caller provided buffers?

We could also consider makros that allocate space on the stack for these strings automatically?
```C
NET_LOG_DBG_WITH_ADDR(addr, fmt, ...) \
    do {\
         char buf[X];\
         net_sprint_ll_addr_buf((a)->addr, (a)->len, buf, sizeof(buf));\
         NET_DBG("[%s]" fmt, buf, __VA_ARGS__); \
    } while(0)
```

> changing all the usage of this would mean increasing the stack memory consumption when debug printing is enabled

Not necessarily, the caller may also allocate them statically, if they can guarantee, that they are using them from a safe/single-threaded context. Then it would increase the memory consumption in the static data though, of course.

clamattia, 2025-01-31:
A makro-solution without any additional storage:
```C
#define FMT_LL_ADDR_6 "0x%x:0x%x:0x%x:0x%x:0x%x:0x%x"
#define PRINT_LL_ADDR_6(x) x[0], x[1], x[2], x[3], x[4], x[5]

#include <stdio.h>
#include <stdint.h>

#define NET_LOG(...) fprintf(stderr, __VA_ARGS__)

void main(void)
{
    unsigned char addr[6] = {1, 2, 3, 4, 5, 134};
    char *context = (char *)0xcafebabe;
    uint16_t type = 0xbeef;
    int index = 1;
    char *iface = (char *)0xfacefeed;

    NET_LOG("Context %p bind to type 0x%04x iface[%d] %p addr " FMT_LL_ADDR_6,
    	    context, type, index, iface, PRINT_LL_ADDR_6(addr));

}
```

Outputs : `Context 0xcafebabe bind to type 0xbeef iface[1] 0xfacefeed addr 0x1:0x2:0x3:0x4:0x5:0x86`