zephyrproject-rtos/zephyr #84968: RFC: Add fixed-subpartitions binding

nordicjm, 2025-01-31:
### Problem description

At present, flash partitions can be defined in zephyr with a single layer, that is you have one flash device, there can be multiple flash partitions on that device but you cannot sub-partition those partitions into smaller partitions. Whilst most scenarios do not need such a system, one area that does is TF-M whereby there is a secure and non-secure part of the image, and both combined formed the whole "slot" of the image. This presents a problem for firmware updating as you depending upon configuration, you need to be able to either upload to one of the sub-partitions or both sub-partitions (but treating it as a whole area, not as 2 distinct partitions).

### Proposed change
A new binding, `fixed-subpartitions` which can be used to split a flash partition into multiple sub-partitions
A neat feature of this is that this also builds which do not make use of these sub-partitions e.g. non-TF-M builds can use the top level partition instead of the smaller divided up partition, for nRF5340-based devices using the default flash files this means that bootloader-based applications not using TF-M can now be up to 448KiB in size instead of 256KiB with no changes needed in application/sample code.

## Detailed RFC
Sample PR showing addition and how it can be used on a board: https://github.com/zephyrproject-rtos/zephyr/pull/84967

### Dependencies
DFU libraries would need to be updated to work using this, there is already an issue for this: https://github.com/zephyrproject-rtos/zephyr/issues/77972

### Concerns and Unresolved Questions
None known at present time

## Alternatives
None

Laczen, 2025-02-01:
Why not reuse the fixed-partition compatible? To know if it is a sub partition check the compatible of the parent.

nordicjm, 2025-02-03:
> Why not reuse the fixed-partition compatible? To know if it is a sub partition check the compatible of the parent.

Because that only works for flash controller devices without a label

closed 2025-04-30 by kartben