zephyrproject-rtos/zephyr #84984: ASSERTION FAIL [0] hci_core.c when disconnect happens while sending packets

Rico-van-Dongen, 2025-01-31:
**Describe the bug**

When sending large volumes of data via Bluetooth notifications the hci_core asserts when the connection is lost during the transmission of multiple packets. 

**To Reproduce**

take any board that is using BT_HCI and send frequent messages using bt_gatt_notify_cb. While streaming data,  bring your device out of range or disable Bluetooth on the receiving end. You will see the bt_gatt_notify_cb function keeps on pushing data with the return value success and the complete callback function keeps triggering. At some point the HCI layer will fail 
**<err> bt_hci_core: packets count mismatch
ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/subsys/bluetooth/host/hci_core.c:576**

It looks like the connection is disconnected while processing data in the while loop of line 564. The check before the while loop on an active connection passed but when debugging the conn variable contained only 0 values and showed state = BT_CONN_DISCONNECTED so it makes sense the sys_slist_get(&conn->tx_pending) returns 0. 

**Expected behavior**

I was expecting the the disconnect callback to fire before the assert. 
The assert was added in commit [28535fe](https://github.com/zephyrproject-rtos/zephyr/commit/28535fe2f27a027b8f0712ce724224b250261d6b#diff-774b3de86adb275da54becf2a0958a2d4dae03d63e47840037ecc575e9d658b4)

When commenting the assert the code continues as expected and the disconnect callback is triggered just after the packets count mismatch error message. 
possible solutions: 

- remove the assert again. 
- re-instate locks around the while loop preventing the conn object to be updated while processing packets
- somehow flush the TX queue on the first sign of a disconnect?

**Impact**

the device crashes unexpected when the connection is lost. this could happen at any given moment so it is quite a showstopper at the moment. 

**Logs and console output**

With the assert:
```
[00:01:32.724,000] <dbg> bt_hci_core: bt_tx_irq_raise: kick TX
[00:01:32.725,000] <dbg> bt_hci_core: bt_tx_irq_raise: kick TX
[00:01:32.725,000] <dbg> bt_hci_core: bt_tx_irq_raise: kick TX
[00:01:32.726,000] <dbg> bt_hci_core: bt_tx_irq_raise: kick TX
[00:01:32.726,000] <dbg> bt_hci_core: tx_processor: TX process start
[00:01:32.726,000] <dbg> bt_hci_core: bt_tx_irq_raise: kick TX
[00:01:32.726,000] <dbg> bt_hci_core: tx_processor: TX process start
[00:01:32.740,000] <dbg> bt_hci_core: bt_recv_unsafe: buf 0x1001ec48 len 7[0m
[00:01:32.741,000] <dbg> bt_hci_core: hci_num_completed_packets: num_handles 1
[00:01:32.741,000] <dbg> bt_hci_core: hci_num_completed_packets: handle 0 count 1
[00:01:32.741,000] <dbg> bt_hci_core: bt_recv_unsafe: buf 0x1001d59c len 6
[00:01:32.741,000] <dbg> bt_hci_core: hci_disconn_complete_prio: status 0x00 handle 0 reason 0x13
[00:01:32.742,000] <dbg> bt_hci_core: bt_recv_unsafe: buf 0x1001ec48 len 7
[00:01:32.742,000] <dbg> bt_hci_core: hci_num_completed_packets: num_handles 1
[00:01:32.698,000] <dbg> bt_hci_core: bt_tx_irq_raise: kick TX
[00:01:32.743,000] <dbg> bt_hci_core: hci_num_completed_packets: handle 0 count 4
[00:01:32.743,000] <err> bt_hci_core: packets count mismatch
**ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/subsys/bluetooth/host/hci_core.c:576**
[00:01:32.743,000] <err> os: r0/a1:  0x00000004  r1/a2:  0x00000240  r2/a3:  0x00000037
[00:01:32.743,000] <err> os: r3/a4:  0x00000004 r12/ip:  0xa0000000 r14/lr:  0x00013777
[00:01:32.743,000] <err> os:  xpsr:  0x21000000
[00:01:32.743,000] <err> os: Faulting instruction address (r15/pc): 0x0002cca8
[00:01:32.743,000] <err> os: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
[00:01:32.743,000] <err> os: Current thread: 0x10004988 (spi_hddr)
[00:01:33.218,000] <dbg> file: store: LOG :118420
```

Without assert:
```
[00:01:04.836,000] <dbg> bt_hci_core: bt_tx_irq_raise: kick TX
[00:01:04.838,000] <dbg> bt_hci_core: bt_tx_irq_raise: kick TX
[00:01:04.838,000] <dbg> bt_hci_core: tx_processor: TX process start
[00:01:04.847,000] <dbg> bt_hci_core: bt_recv_unsafe: buf 0x1001d59c len 6
[00:01:04.847,000] <dbg> bt_hci_core: hci_disconn_complete_prio: status 0x00 handle 0 reason 0x13
[00:01:04.848,000] <dbg> bt_hci_core: bt_recv_unsafe: buf 0x1001ec48 len 7
[00:01:04.848,000] <dbg> bt_hci_core: hci_num_completed_packets: num_handles 1
[00:01:04.848,000] <dbg> bt_hci_core: hci_num_completed_packets: handle 0 count 5
[00:01:04.848,000] <err> bt_hci_core: packets count mismatch
[00:01:04.848,000] <dbg> bt_hci_core: rx_work_handler: Getting net_buf from queue
[00:01:04.848,000] <dbg> bt_hci_core: rx_work_handler: buf 0x1001d59c type 1 len 6
[00:01:04.848,000] <dbg> bt_hci_core: hci_event: event 0x05
[00:01:04.848,000] <dbg> bt_hci_core: hci_disconn_complete: status 0x00 handle 0 reason 0x13
```

**Environment (please complete the following information):**

 - Cortex M4 MCU with Renesas BLE device connected via HCI-SPI
 - Version 3.7.0

**Additional context**

Rico-van-Dongen, 2025-01-31:
It looks like simply disabling the asserts is not enough because even tough the hci_disconn_complete comes the advertisement does not seem to restart.

jhedberg, 2025-02-01:
You mention in the description that this was reproduced with version 3.7. Are you also able to reproduce it with the latest upstream tree?

cvinayak, 2025-02-02:
> Cortex M4 MCU with Renesas BLE device connected via HCI-SPI

and
> take any board that is using BT_HCI 

Please reproduce the issue with one supported board in The Zephyr Project (https://docs.zephyrproject.org/latest/boards/index.html), and describe the steps to reproduce like build commands used, flashing steps and test procedure. This will help the community in analyzing the said issue quicker.

Rico-van-Dongen, 2025-02-03:
> You mention in the description that this was reproduced with version 3.7. Are you also able to reproduce it with the latest upstream tree?

I could try, but i see the latest version has the identical loop in the hci_num_completed_packets function. I think it will be just as vulnerable for issues when the disconnect happens between the check for connection and the while loop processing the messages.

jhedberg, 2025-02-03:
@Rico-van-Dongen which HCI driver is this? The driver should ever only be in a single `recv()` call to the host, i.e. it shouldn't be possible for a disconnect event to preempt a `num_completed_packets()` call.

Rico-van-Dongen, 2025-02-05:
> [@Rico-van-Dongen](https://github.com/Rico-van-Dongen) which HCI driver is this? The driver should ever only be in a single `recv()` call to the host, i.e. it shouldn't be possible for a disconnect event to preempt a `num_completed_packets()` call.

@jhedberg, thanks for your comment. I need a bit more time to check this but maybe this could be the root cause. Judging from the logs there are some receive calls happening after the TX process start.

Rico-van-Dongen, 2025-02-07:
@jhedberg, we debugged a bit further and it turns out the actual issue is not due to preemption. Already before entering that while loop the conn state is BT_CONN_DISCONNECT_COMPLETE.   
It thus looks like the Renesas BLE device (DA14531) actually triggers the num_complete_packets() event even if the connection is already closed (or closing). 
So, the actual cause is that the num_packets_complete is called after the hci_disconn_complete_prio.   

We have found 2 potential work-arounds
1. Add a check in the num_complete_packets() to skip processing packets if the state is no longer BT_CONN_CONNECTED

```
static void hci_num_completed_packets(struct net_buf *buf)
{

...........
		
while (conn->state != BT_CONN_CONNECTED && count--) { 
			sys_snode_t *node;

.........
```

2. Skip the flushing of the TX pending when the state is clanged to BT_CONN_DISCONNECT_COMPLETE such that the complete events are properly handled. In other words, remove the function `process_unack_tx(conn)` from conn.c

I can imagine the later option is specific behavior for our BLE device and may break implementations for other devices. Is there something we can check on the behavior of the hci_disconn_complete_prio and the different conn state changes to prevent this issue

jhedberg, 2025-02-07:
@Rico-van-Dongen this seems to me like noncompliant behavior by the controller. Here's the description for the HCI_NUmber_Of_Completed_Packets event in the HCI spec (Bluetooth Core Spec 6.0 Vol 4 Part E section 7.7.19:
[image]

The key part is this: "The HCI_Number_Of_Completed_Packets event shall not specify a given Handle before the Controller has sent the event indicating that the corresponding connection or BIG has been created or after it has sent the event indicating disconnection of the corresponding connection or indicating that the BIG has been terminated."

If your controller followed the above, it would either not send the event after HCI_Disconnection_Complete, or send it with an invalid handle, which would then trigger a different codepath in `hci_core.c` (logging an error instead of triggering an assert).

Rico-van-Dongen, 2025-02-14:
@jhedberg, We are contacting the supplier of the BLE controller to see if it is indeed the case that there should be no completed_packets event after the disconnect_complete. In the mean time, could you explain what an invalid handle code in this case would be. If I look at the specification of the disconnect_complete event it looks to me that in this case the controller should have already deleted handle 0 so that the additional completed_packets would actually be on a invalid handle as handle 0 is already deleted. 

[image]

as you can see from the logs in this case the HCI_Disconnection_Complete is being send with status 0 and as we are in the peripheral role you could argue that handle 0 should already have been deleted in the disconnect_prio.

Rico-van-Dongen, 2025-03-31:
Closing this issue. By changing the priorities in our driver we managed to fix it. Now both the disconnect and disconnect_prio are handled before a new completed_packets is handled. This indeed results in just an error log that the reported handle is invalid.

closed 2025-03-31 by Rico-van-Dongen