zephyrproject-rtos/zephyr #84998: samples: usb: webusb-next: failure to enumerate on nucleo_h723zg under MS Windows

henrikbrixandersen, 2025-01-31:
**Describe the bug**
When running `samples/subsys/usb/webusb-next/` on the `nucleo_h723zg/stm32h723xx` board, it fails to enumerate on MS Windows.

Steps to reproduce the behavior:
1. `west build -b nucleo_h723zg/stm32h723xx samples/subsys/usb/webusb-next/`
2. `west flash`
3. See error below when board is plugged into a PC running MS Windows

**Expected behavior**
USB enumeration completes.

**Impact**
Unable to use USB device_next with MS Windows.

**Logs and console output**
```
*** Booting Zephyr OS build v4.0.0-4398-gf19e5bc577e2 ***
[00:00:11.898,000] <inf> main: USBD message: Device suspended
[00:00:12.044,000] <inf> main: USBD message: Bus reset
[00:00:17.681,000] <inf> main: USBD message: Bus reset
[00:00:23.383,000] <inf> main: USBD message: Bus reset
[00:00:29.083,000] <inf> main: USBD message: Bus reset
[00:00:34.231,000] <wrn> udc: Spurious suspend/resume event
[00:00:34.231,000] <inf> main: USBD message: Device suspended
```

**Environment (please complete the following information):**

 - OS: MS Windows USB host
 - Toolchain: Zephyr SDK 0.17.0
 - Commit SHA: f19e5bc577e2

**Additional context**
Same sample fails on NXP LPC55xxx: https://github.com/zephyrproject-rtos/zephyr/issues/83306

JayToltTech, 2025-02-13:
I see from https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/st/nucleo_h723zg/nucleo_h723zg.dts that the Nucleo h723 dts doesn't define a USB device.  @henrikbrixandersen did you modify the dts or use an overlay to enable the usb device on this board?

JayToltTech, 2025-02-13:
From what I can tell, this is working on MacOS 15.3 on a Nucleo-H533 @ commit 03fa6a0c33c7e918d4157eb5404835c164b2c3f1

``` bash
ioreg -p IOUSB
+-o Root  <class IORegistryEntry, id 0x100000100, retain 42>
  +-o AppleT6000USBXHCI@00000000  <class AppleT6000USBXHCI, id 0x100000573, registered, matched, active, busy 0 (433 ms$
  | +-o billboard@00100000  <class IOUSBHostDevice, id 0x100001126, registered, matched, active, busy 0 (9219 ms), reta$
  +-o AppleT6000USBXHCI@05000000  <class AppleT6000USBXHCI, id 0x1000004ab, registered, matched, active, busy 0 (250 ms$
  | +-o USBD sample@05100000  <class IOUSBHostDevice, id 0x10049b868, registered, matched, active, busy 0 (11 ms), reta$
```

I was not able to connect the sample web page via Safari, but I believe Safari doesn't support webusb...

```
00:00:00.000,000] <inf> usbd_init: bNumInterfaces 1 wTotalLength 32
[00:00:00.000,000] <inf> usbd_init: FS bNumConfigurations 1
[00:00:00.101,000] <inf> main: USBD message: Device resumed
[00:00:00.151,000] <inf> udc: ep 0x80 is not halted|disabled
[00:00:00.151,000] <inf> usbd_core: Actual device speed 1
[00:00:00.151,000] <inf> main: USBD message: Bus reset
[00:00:00.162,000] <inf> usbd_ch9: Handle control 0x2000391c ep 0x00, len 8, s:1 d:0 s:0
--- 1 messages dropped ---
[00:00:00.162,000] <inf> usbd_ch9: s-(out)-status finished
--- 8 messages dropped ---
[00:00:00.176,000] <inf> usbd_ch9: Handle control 0x200039ac ep 0x00, len 8, s:1 d:0 s:0
[00:00:00.179,000] <inf> udc: Sequence 4 not completed
[00:00:00.179,000] <inf> usbd_ch9: Handle control 0x200039d0 ep 0x00, len 8, s:1 d:0 s:0
[00:00:00.182,000] <inf> udc: Sequence 4 not completed
[00:00:00.182,000] <inf> usbd_ch9: Handle control 0x200039f4 ep 0x00, len 8, s:1 d:0 s:0
[00:00:00.185,000] <inf> udc: Sequence 4 not completed
[00:00:00.185,000] <inf> usbd_ch9: Handle control 0x20003a18 ep 0x00, len 8, s:1 d:0 s:0
[00:00:00.188,000] <inf> udc: Sequence 4 not completed
[00:00:00.188,000] <inf> usbd_ch9: Handle control 0x20003a3c ep 0x00, len 8, s:1 d:0 s:0
[00:00:00.191,000] <inf> udc: Sequence 4 not completed
[00:00:00.191,000] <inf> usbd_ch9: Handle control 0x20003a60 ep 0x00, len 8, s:1 d:0 s:0
[00:00:00.194,000] <inf> udc: Sequence 4 not completed
[00:00:00.194,000] <inf> usbd_ch9: Handle control 0x20003a84 ep 0x00, len 8, s:1 d:0 s:0
[00:00:00.197,000] <inf> udc: Sequence 4 not completed
[00:00:00.197,000] <inf> usbd_ch9: Handle control 0x20003aa8 ep 0x00, len 8, s:1 d:0 s:0
[00:00:00.205,000] <inf> udc: Sequence 4 not completed
[00:00:00.205,000] <inf> usbd_ch9: Handle control 0x20003acc ep 0x00, len 8, s:1 d:0 s:0
[00:00:00.205,000] <inf> usbd_ch9: Set Configuration Request value 1
[00:00:00.205,000] <inf> usbd_iface: Modify interface 0 ep 0x81 by op 1 ep_bm 20000
[00:00:00.205,000] <inf> usbd_iface: Modify interface 0 ep 0x01 by op 1 ep_bm 20002
[00:00:00.205,000] <inf> usbd_ch9: Handle control 0x20003af0 ep 0x80, len 0, s:0 d:0 s:1
[00:00:00.205,000] <inf> usbd_ch9: s-(out)-status finished
```

ddavidebor, 2025-02-14:
We had some issues with high-speed on the STM32H7 with zephyr and usb-next, you might want to try bumping down to full speed see if that helps

PR #86158 (2025-02-25): drivers: udc_stm32: handle ZLP flag for control transfers as well
I mistakenly assumed in the commit 6aaad0a5cd23
("drivers: udc_stm32: handle ZLP flag") that the HAL driver would handle ZLP flag in control transfers itself, but that does not seem to be the case.

Fixes: #84998

Tested on stm32f723e-disco (FS only, pity this board does not configure HS), nucleo_g431rb, nucleo_f413zh.
closed 2025-02-25 by kartben