zephyrproject-rtos/zephyr #84999: samples: usb: webusb-next: failure to enumerate on mimxrt1024_evk/mimxrt1024 under MS Windows

henrikbrixandersen, 2025-01-31:
**Describe the bug**
When running `samples/subsys/usb/webusb-next/` on the `mimxrt1024_evk/mimxrt1024` board, it fails to enumerate on MS Windows.

Steps to reproduce the behavior:
1. `west build -b mimxrt1024_evk/mimxrt1024 samples/subsys/usb/webusb-next/`
2. `west flash`
3. See error below when board is plugged into a PC running MS Windows

**Expected behavior**
USB enumeration completes.

**Impact**
Unable to use USB device_next with MS Windows.

**Logs and console output**
```
*** Booting Zephyr OS build v4.0.0-4398-gf19e5bc577e2 ***
[00:00:06.169,000] <inf> main: USBD message: Bus reset
[00:00:06.274,000] <inf> main: USBD message: Bus reset
[00:00:11.890,000] <err> usbd_core: UDC error event
[00:00:11.890,000] <err> usbd_core: UDC error event
[00:00:11.890,000] <inf> main: USBD message: Bus reset
[00:00:11.891,000] <inf> main: USBD message: Controller error
[00:00:11.891,000] <inf> main: USBD message: Controller error
[00:00:17.569,000] <err> usbd_core: UDC error event
[00:00:17.569,000] <err> usbd_core: UDC error event
[00:00:17.570,000] <inf> main: USBD message: Bus reset
[00:00:17.570,000] <inf> main: USBD message: Controller error
[00:00:17.570,000] <inf> main: USBD message: Controller error
[00:00:23.249,000] <err> usbd_core: UDC error event
[00:00:23.249,000] <err> usbd_core: UDC error event
[00:00:23.249,000] <inf> main: USBD message: Bus reset
[00:00:23.249,000] <inf> main: USBD message: Controller error
[00:00:23.249,000] <inf> main: USBD message: Controller error
```

**Environment (please complete the following information):**

 - OS: MS Windows USB host
 - Toolchain: Zephyr SDK 0.17.0
 - Commit SHA: f19e5bc577e2

**Additional context**
- Same sample fails on NXP LPC55xxx: https://github.com/zephyrproject-rtos/zephyr/issues/83306 and STM32H723ZG: https://github.com/zephyrproject-rtos/zephyr/issues/84998
- The same board/firmware build enumerates correctly on Ubuntu 24.04.
- The same firmware enumerates correctly on MS Windows when running on the `frdm_k64f` board.

jfischer-no, 2025-01-31:
**What** MS Windows version?

henrikbrixandersen, 2025-01-31:
> **What** MS Windows version?

Windows 10 Enterprise, Version 22H2.

PR #85004 (2025-02-06): drivers: udc_mcux_ehci: fix typo in transfer handler
If the to-host data stage length is less than that requested by the
host, but equal to or a multiple of MPS, the device should send a ZLP,
not receive it.

Fixes: #84999
Tested on mimxrt1020_evk

udc_mcux_ip3511.c has similar type, but according to the reporter it would not fix similar issue there, so there should be something else.
closed 2025-02-06 by kartben