zephyrproject-rtos/zephyr #85011: ESP32_BLE advertising issue

ilanchezh1yan, 2025-02-01:
Hello All,
I am a young fellowman started to work with BLE communication establishment in esp32_devikitc_wroom with zephyr. I built the "peripheral_gatt_write" in sample directory. As per the code logic the terminal logs for "Advertising started" but nrf_Connect application can't scan for the device.

There is no hardware issue I have verified with ArduinoIDE. The code works fine with nrf52832. I am blindfolded without any leads. Requesting for clarification.

It would be better if you discuss about the HCI interface between the RADIO and the controller. In Zephyr devicetree "bt_hci_bus=ipm". Is that inter-processor mailbox is the bus between the RADIO and the controller.

Infer with 
west build -b esp32_devkitc_wroom/esp32/procpu samples\bluetooth\peripheral_gatt_write
west flash 
west espressif monitor

Serial port COM10
Connecting.....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting.....
Detecting chip type... ESP32
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \.\COM10 instead...
--- idf_monitor on \.\COM10 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 12:21:46
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3ffbdb60,len:10712
load:0x40080000,len:91040
0x40080000: _WindowOverflow4 at D:/zephyrproject/zephyr/arch/xtensa/core/window_vectors.S:56

1150 mmu set 00010000, pos 00010000
entry 0x40090048
0x40090048: __start at D:/zephyrproject/zephyr/soc/espressif/common/loader.c:253

I (58) soc_init: ESP Simple boot
I (58) soc_init: compile time Feb 1 2025 11:11:01
W (59) soc_init: Unicore bootloader
I (59) spi_flash: detected chip: generic
I (62) spi_flash: flash io: dio
I (65) soc_init: chip revision: v3.1
I (69) flash_init: SPI Speed : 40MHz
I (72) flash_init: SPI Mode : DIO
I (76) flash_init: SPI Flash Size : 4MB
I (80) boot: DRAM: lma 0x00001020 vma 0x3ffbdb60 len 0x29d8 (10712)
I (86) boot: IRAM: lma 0x00003a00 vma 0x40080000 len 0x163a0 (91040)
0x40080000: _WindowOverflow4 at D:/zephyrproject/zephyr/arch/xtensa/core/window_vectors.S:56

I (92) boot: IRAM: lma 0x00019db8 vma 0x00000000 len 0x6240 (25152)
I (98) boot: IMAP: lma 0x00020000 vma 0x400d0000 len 0x33428 (209960)
0x400d0000: _stext at ??:?

I (104) boot: IRAM: lma 0x00053430 vma 0x00000000 len 0xcbc8 (52168)
I (110) boot: DMAP: lma 0x00060000 vma 0x3f400000 len 0x6b2c (27436)
I (117) boot: Image with 6 segments
I (120) boot: DROM segment: paddr=00060000h, vaddr=3f400000h, size=06B2Ch ( 27436) map
I (128) boot: IROM segment: paddr=00020000h, vaddr=400d0000h, size=33428h (209960) map
I (146) boot: Jumping to the main image...
I (174) heap_runtime: ESP heap runtime init at 0x3ffc9e34 size 88 kB.
*** Booting Zephyr OS build v4.0.0-3819-gddff91fef6ac ***
[00:00:00.175,000] esp32_bt_adapter: BT controller compile version [241c96c]
[00:00:00.397,000] net_buf: Timeout discarded. No blocking in syswq
[00:00:00.397,000] bt_hci_core: Identity: 88:13:BF:70:8D:90 (public)
[00:00:00.397,000] bt_hci_core: HCI: version 4.2 (0x08) revision 0x030e, manufacturer 0x0060
[00:00:00.397,000] bt_hci_core: LMP: version 4.2 (0x08) subver 0x030e
Bluetooth initialized
Advertising successfully started

Thanking you for your valuable time in advance.

rftafas, 2025-02-03:
> It would be better if you discuss about the HCI interface between the RADIO and the controller. In Zephyr devicetree "bt_hci_bus=ipm". Is that inter-processor mailbox is the bus between the RADIO and the controller.

I assume this is a question. BT is a peripheral and you should use normal Zephyr APIs. You don't need that on device tree.

You could try the shell: https://docs.zephyrproject.org/latest/connectivity/bluetooth/bluetooth-shell.html (lots of good info there).

ilanchezh1yan, 2025-02-04:
Thank you for considering my writing. Definitely I refer to the document
and update about the issue.

fabiobaltieri, 2025-02-04:
Hi, could you change the issue description so that it uses the [bug report template](https://github.com/zephyrproject-rtos/zephyr/issues/new?template=001_bug_report.md)? It asks for some important information that are very useful for us to understand the issue. Thanks.

rftafas, 2025-02-06:
I also noticed this is using ESP32 rev 3.1. Support for it is WIP.

ilanchezh1yan, 2025-02-07:
Thank you. Does any other revision of the zephyr toolchain have it done.

sylvioalves, 2025-02-14:
@ilanchezh1yan This should be fixed in last main. I'll close this issue. Re-open if needed.

closed 2025-02-14 by sylvioalves

ilanchezh1yan, 2025-02-15:
It's fixed!

rt-2pm2, 2025-03-12:
I am experiencing the same problem: the build and flashing is fine, the execution reports successful advertising but I don't see the device with Bluetooth scanners.
In my case, I also tried different samples and I noticed that the device correctly advertise when configured as "non-connectable": as in the _iBeacon_ sample.

Here the debug output of the build process:
```
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /<...>/samples/bluetooth/peripheral_gatt_write
-- CMake version: 3.30.3
-- Found Python3: /<...>/.venv/bin/python3 (found suitable version "3.10.12", minimum required is "3.10") found components: Interpreter
-- Cache files will be written to: <...>/.cache/zephyr
-- Zephyr version: 4.1.99 (/<...>/zephyrproject/zephyr)
-- Found west (found suitable version "1.3.0", minimum required is "0.14.0")
-- Board: esp32_devkitc_wroom, qualifiers: esp32/procpu
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.17.0 (/<...>/zephyr-sdk-0.17.0)
-- Found toolchain: zephyr 0.17.0 (/<...>/zephyr-sdk-0.17.0)
-- Found Dtc: /<...>/zephyr-sdk-0.17.0/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found BOARD.dts: /<...>/zephyrproject/zephyr/boards/espressif/esp32_devkitc_wroom/esp32_devkitc_wroom_procpu.dts
-- Generated zephyr.dts: /<...>/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated pickled edt: /<...>/zephyrproject/zephyr/build/zephyr/edt.pickle
-- Generated devicetree_generated.h: /<...>/zephyrproject/zephyr/build/zephyr/include/generated/zephyr/devicetree_generated.h
-- Including generated dts.cmake file: /<...>/zephyrproject/zephyr/build/zephyr/dts.cmake
Parsing /<...>/zephyrproject/zephyr/Kconfig
Loaded configuration '/<...>/zephyrproject/zephyr/boards/espressif/esp32_devkitc_wroom/esp32_devkitc_wroom_procpu_defconfig'
Merged configuration '/<...>/zephyrproject/zephyr/samples/bluetooth/peripheral_gatt_write/prj.conf'
Configuration saved to '/<...>/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/<...>/zephyrproject/zephyr/build/zephyr/include/generated/zephyr/autoconf.h'
-- Found GnuLd: /<...>/zephyr-sdk-0.17.0/xtensa-espressif_esp32_zephyr-elf/xtensa-espressif_esp32_zephyr-elf/bin/ld.bfd (found version "2.38")
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /<...>/zephyr-sdk-0.17.0/xtensa-espressif_esp32_zephyr-elf/bin/xtensa-espressif_esp32_zephyr-elf-gcc
-- Espressif HAL path: /<...>/zephyrproject/modules/hal/espressif
-- Use the esptool.py: /<...>/zephyrproject/modules/hal/espressif/tools/esptool_py/esptool.py
-- Image partition /soc/flash-controller@3ff42000/flash@0/partitions/partition@1000
-- Using ccache: /usr/bin/ccache
-- Configuring done (9.2s)
-- Generating done (0.2s)
-- Build files have been written to: /<...>/zephyrproject/zephyr/build
-- west build: building application
[3/370] Generating include/generated/zephyr/version.h
-- Zephyr version: 4.1.99 (/<...>/zephyrproject/zephyr), build: v4.1.0-363-g895d91169e8b
[370/370] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:      405992 B    4194048 B      9.68%
     iram0_0_seg:       92672 B       224 KB     40.40%
     dram0_0_seg:       76016 B     140452 B     54.12%
     irom0_0_seg:      209384 B     11456 KB      1.78%
     drom0_0_seg:         64 KB         4 MB      1.56%
    rtc_iram_seg:          0 GB         8 KB      0.00%
    rtc_slow_seg:          0 GB         8 KB      0.00%
    rtc_data_seg:          0 GB         8 KB      0.00%
        IDT_LIST:          0 GB         8 KB      0.00%
Generating files from /<...>/zephyrproject/zephyr/build/zephyr/zephyr.elf for board: esp32_devkitc_wroom
esptool.py v4.8.1
Creating esp32 image...
Image has only RAM segments visible. ROM segments are hidden and SHA256 digest is not appended.
Merged 25 ELF sections
Successfully created esp32 image.
```

I attach also the debug prints of the sample peripheral_gatt_write application:
```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
[... 10 lines cut ...]
I (70) flash_init: SPI Flash Size : 4MB
I (74) boot: DRAM: lma 0x00001020 vma 0x3ffbdb60 len 0x29d0   (10704)
I (80) boot: IRAM: lma 0x000039f8 vma 0x40080000 len 0x169f0  (92656)
I (86) boot: IRAM: lma 0x0001a3f8 vma 0x00000000 len 0x5c00   (23552)
I (92) boot: IMAP: lma 0x00020000 vma 0x400d0000 len 0x331e8  (209384)
I (98) boot: IRAM: lma 0x000531f0 vma 0x00000000 len 0xce08   (52744)
I (105) boot: DMAP: lma 0x00060000 vma 0x3f400000 len 0x6a04   (27140)
I (111) boot: Image with 6 segments
I (114) boot: IROM segment: paddr=00020000h, vaddr=400d0000h, size=331E8h (209384) map
I (122) boot: DROM segment: paddr=00060000h, vaddr=3f400000h, size=06A04h ( 27140) map
I (141) boot: libc heap size 62 kB.
I (141) spi_flash: detected chip: generic
I (141) spi_flash: flash io: dio
*** Booting Zephyr OS build v4.1.0-363-g895d91169e8b ***
[00:00:00.175,000] <inf> esp32_bt_adapter: BT controller compile version [f7360be]
[00:00:00.399,000] <inf> bt_hci_core: Identity: A8:42:E3:48:13:A4 (public)
[00:00:00.399,000] <inf> bt_hci_core: HCI: version 4.2 (0x08) revision 0x030e, manufacturer 0x0060
[00:00:00.399,000] <inf> bt_hci_core: LMP: version 4.2 (0x08) subver 0x030e
Bluetooth initialized
Advertising successfully started
```

Do you have some hints on how to debug this issue? For example, @sylvioalves could you please provide a reference to the specific commit that fixed the problem a month ago?

Thank you very much in advance for your time

ilanchezh1yan, 2025-03-12:
Hello, try it by changing the minimum connection interval any value more than 50ms. That is the thing fixed the issue in my case.

rt-2pm2, 2025-03-12:
> Hello, try it by changing the minimum connection interval any value more than 50ms. That is the thing fixed the issue in my case.

Thank you for the information. Did you change also the MAX value?

In the attempts I made, I tried the ***BT_LE_ADV_CONN_FAST_2*** which has a minimum connection interval of 100ms, but unfortunatelly, it did not work.

The BT_LE_ADV_CONN_FAST_1 is equivalent to
`BT_LE_ADV_PARAM(BT_LE_ADV_OPT_CONN, BT_GAP_ADV_FAST_INT_MIN_1, BT_GAP_ADV_FAST_INT_MAX_1, NULL)`, where
BT_GAP_ADV_FAST_INT_MIN_1 = 30ms
BT_GAP_ADV_FAST_INT_MAX_1 = 60ms

The BT_LE_ADV_CONN_FAST_2 is equivalent to
`BT_LE_ADV_PARAM(BT_LE_ADV_OPT_CONN, BT_GAP_ADV_FAST_INT_MIN_2, BT_GAP_ADV_FAST_INT_MAX_2, NULL)`, where
BT_GAP_ADV_FAST_INT_MIN_2 = 100ms
BT_GAP_ADV_FAST_INT_MAX_2 = 150ms

sylvioalves, 2025-03-12:
Hi Folks, I confirm the issue exists in ESP32 with revisions >= 3.0
We will provide the fix.

rt-2pm2, 2025-03-12:
> Hi Folks, I confirm the issue exists in ESP32 with revisions >= 3.0 We will provide the fix.

Thank you very much for the information and for the commitment.

Just for curiosity, did you confirm only the presence of the issue, or do you also have some hints about what is causing it? In the latter case, could you provide a reference? Thanks.