zephyrproject-rtos/zephyr #85013: Conflict between picolibc and POSIX_DEVICE_IO

lromeraj, 2025-02-01:
**Describe the bug**
There is a conflict between the `picolibc` implementation of `fdopen()` and Zephyr's implementation in `zephyr/lib/posix/options/device_io.c`. The `fopen()` function from `picolibc` uses its own `fdopen()` implementation, but instead of using `picolibc`'s `fdopen()`, Zephyr’s `fdopen()` from `device_io.c` is being used. This causes issues with file stream flags and the behavior of functions like `fgetc()`.

This issue arises from the changes introduced in PR #76352.

**Steps to Reproduce:**
1. Enable the `POSIX_API` in Zephyr by setting the appropriate configuration.
2. Build the project with `picolibc` (precompiled in toolchain or using `CONFIG_PICOLIBC_USE_MODULE=y` it doesn't matter).
3. Attempt to use `fgetc()` or other file I/O operations.
4. Observe that `__SRD` flag is not set, causing functions like `fgetc()` to fail.

You can add a breakpoint for `fdopen` and see that it is using the implementation from `zephyr/lib/posix/options/device_io.c`.

**Expected Behavior:**
The `fdopen()` function from `picolibc` should be used, ensuring proper handling of file stream flags and correct behavior of file-related functions like `fgetc()` and `fileno()`.

**Actual Behavior:**
The `fdopen()` function from `zephyr/lib/posix/options/device_io.c` is used instead, causing stream flag issues that prevent proper file I/O behavior, affecting functions such as `fgetc()` and equivalent.

**Affected Versions:**  
Zephyr 4.0.0 and later (introduced by PR #76352).

**Impact**
This issue prevents proper file I/O behavior when using picolib and POSIX API.

**Environment:**
 - OS: Windows (WSL)
 - Toolchain: Zephyr SDK 0.17
 - Version used: Zephyr >= 4.0.0

**Additional context**
I'm not sure exactly where this issue should be reported—whether in the main Zephyr repository or in the picolibc repository. However, it seems like it might be something that needs to be addressed on the picolibc side (or in my project configuration).

cfriedt, 2025-02-01:
There has been a PR in review for a while already to fix this. Thanks for filing an issue though

lromeraj, 2025-02-03:
> There has been a PR in review for a while already to fix this. Thanks for filing an issue though

Thanks for the reference! I wasn’t aware there was already work in progress to address this issue. Will this modification also be patched in the Zephyr 4.0.0 release?

cfriedt, 2025-02-03:
> Thanks for the reference! I wasn’t aware there was already work in progress to address this issue. Will this modification also be patched in the Zephyr 4.0.0 release?

We should be able to backport the fix, yes, but it does need to go through the review process first and we'll also need to see how e.g. the armclang toolchain is updated.

lromeraj, 2025-12-10:
@cfriedt - I've identified a memory leak affecting both the v2 and v3 implementations. Initially, I assumed the system was using the `fclose()` implementation from picolibc, but that wasn’t the case. Instead, Zephyr provides its own `fclose()` at `zephyr/lib/libc/common/source/stdio/fclose.c`, and this is the one actually being used.

I discovered this after repeatedly opening and closing hundreds of files. Eventually, the system crashed with a fatal error inside `__retarget_lock_init()`, where `malloc()` returned `NULL` while allocating a semaphore. The root cause is that `__bufio_close()` is never invoked because the picolibc version of `fclose()` is not the one in use. As a result, the buffer I/O resources are never released.

This means we must also account for the `zvfs_libc_file_free_cb()` function, which performs the libc-specific cleanup that is otherwise skipped. Additionally, the `__BFALL` flag doesn’t make sense in v2 or v3, it is never used, and keeping it leads to problematic behavior. If the picolibc `fclose()` is ever used, it would incorrectly call `free()` on a static memory-slab region (this already occurs in v2 #90916).

In general, implementing static memory for stdio complicates things unnecessarily. The C standard library is designed with dynamic allocation in mind. If dynamic memory usage is a concern, a cleaner approach would be to allow developers to partition memory into multiple isolated heaps. This way, critical components can operate in dedicated regions, making leaks contained and diagnosable. Fragmentation becomes a trade-off the developer can consciously choose.

In my opinion, v1 should be the way to go.