zephyrproject-rtos/zephyr #85020: doc: Documentation issue in 'develop/getting_started/index'

Gerriko, 2025-02-01:
**Describe the bug**

In section 2 (Use apt to install the required dependencies) it lists gcc-multilib g++-multilib as part of the sudo apt install command, but as there are multiple versions of gcc and g++ (9, 10,11,12, etc) , it returns an error (Package 'gcc-multilib' has no installation candidate). 

**Environment**
Linux Mint (Ubuntu 22.04)

* Page: `develop/getting_started/index`
* Version: main
* SHA-1: 2f583a8369ef5b67c4438c40d17721d31db952a0

kartben, 2025-02-02:
On what architecture architecture are you? In other words, see https://github.com/zephyrproject-rtos/zephyr/pull/84646 in case you're on an Arm machine

kartben, 2025-02-06:
@Gerriko ping?

Gerriko, 2025-02-06:
@kartben sorry, didn't have notifications enabled. I'm using my laptop (64-bit x86 architecture).