zephyrproject-rtos/zephyr #85021: CONFIG_PM=y does not allow esp32s3 to boot into main function

acf879, 2025-02-01:
**Describe the bug**
Target platform: esp32s3_devkitc
What has been done currently to diagnose: Tested on a esp32_devkitc_wroom board to see if it would behave the same but it behaved as expected with it being able to boot into main with CONFIG_PM=y while the esp32s3_devkitc did not boot into main. 
Problem: When the Kconfig CONFIG_PM=y is selected on the esp32s3_devkitc device, it does not boot into the main function.

**To Reproduce**
1. cd zephyr/samples/hello_world
2. edit prj.conf to include `CONFIG_PM=y`
3. build with `west build -p -b esp32s3_devkitc/esp32s3/procpu` (or for the esp32_devkitc_wroom  `west build -p -b esp32_devkitc_wroom`)
4. flash with `west flash --esp-device /dev/ttyACM0` (or for the esp32_devkitc_wroom  `west flash --esp-device /dev/ttyUSB0`)
5. look at log and hit reset and see that there is no hello world on the esp32s3_devkitc but there is on the esp32_devkitc_wroom

**Expected behavior**
Hello world is still printed with CONFIG_PM=y selected in prj.conf

**Impact**
Does not allow the use of  esp light sleep or other power management options for the esp32s3_devkitc so battery usage will be higher when powered on battery power

**Logs and console output**
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fc8f028,len:0x16e4
load:0x40374000,len:0xb008
load:0x600fe000,len:0x10
SHA-256 comparison failed:
Calculated: a7e7387d1000d319c861b5eda2b0538d85ace7a1d32b18abe856b2c6bdb26d6c
Expected: 00000000c0380000000000000000000000000000000000000000000000000000
Attempting to boot anyway...
entry 0x40378890
I (75) boot: ESP Simple boot
I (75) boot: compile time Feb  1 2025 14:59:02
W (75) boot: Unicore bootloader
I (75) spi_flash: detected chip: boya
I (76) spi_flash: flash io: dio
W (79) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (92) boot: chip revision: v0.2
I (94) boot.esp32s3: Boot SPI Speed : 40MHz
I (98) boot.esp32s3: SPI Mode       : DIO
I (102) boot.esp32s3: SPI Flash Size : 16MB
I (106) boot: Enabling RNG early entropy source...
I (111) boot: DRAM: lma 0x00000020 vma 0x3fc8f028 len 0x16e4   (5860)
I (117) boot: IRAM: lma 0x0000170c vma 0x40374000 len 0xb008   (45064)
I (123) boot: Image with 2 segments
I (126) boot: DROM segment: paddr=00040000h, vaddr=3c010000h, size=01240h (  4672) map
I (134) boot: IROM segment: paddr=00030000h, vaddr=42000000h, size=0456Ah ( 17770) map
```
`CONFIG_PM=y` for esp32s3_devkitc device

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fc8d7b8,len:0x162c
load:0x40374000,len:0x9798
SHA-256 comparison failed:
Calculated: 0799b91866349842d876fde966b5782bb022cf4691929c460fffcf1aa85458c6
Expected: 0000000000520000000000000000000000000000000000000000000000000000
Attempting to boot anyway...
entry 0x40377c30
I (67) boot: ESP Simple boot
I (67) boot: compile time Feb  1 2025 14:56:31
W (67) boot: Unicore bootloader
I (67) spi_flash: detected chip: boya
I (69) spi_flash: flash io: dio
W (72) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (84) boot: chip revision: v0.2
I (87) boot.esp32s3: Boot SPI Speed : 40MHz
I (91) boot.esp32s3: SPI Mode       : DIO
I (94) boot.esp32s3: SPI Flash Size : 16MB
I (98) boot: Enabling RNG early entropy source...
I (103) boot: DRAM: lma 0x00000020 vma 0x3fc8d7b8 len 0x162c   (5676)
I (109) boot: IRAM: lma 0x00001654 vma 0x40374000 len 0x9798   (38808)
I (115) boot: padd: lma 0x0000adf8 vma 0x00000000 len 0x5200   (20992)
I (121) boot: IMAP: lma 0x00010000 vma 0x42000000 len 0x38ec   (14572)
I (128) boot: padd: lma 0x000138f4 vma 0x00000000 len 0xc704   (50948)
I (134) boot: DMAP: lma 0x00020000 vma 0x3c010000 len 0x116c   (4460)
I (140) boot: Image with 6 segments
I (143) boot: DROM segment: paddr=00020000h, vaddr=3c010000h, size=01170h (  4464) map
I (151) boot: IROM segment: paddr=00010000h, vaddr=42000000h, size=038EAh ( 14570) map
I (196) heap_runtime: ESP heap runtime init at 0x3fc91680 size 352 kB.

*** Booting Zephyr OS build v4.0.0 ***
Hello World! esp32s3_devkitc/esp32s3/procpu
```
`CONFIG_PM=n` for esp32s3_devkitc device

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3ffb0000,len:6464
load:0x40080000,len:38864
entry 0x40084168
I (32) boot: ESP Simple boot
I (32) boot: compile time Feb  1 2025 15:00:53
W (32) boot: Unicore bootloader
I (33) spi_flash: detected chip: generic
I (35) spi_flash: flash io: dio
I (38) boot: chip revision: v3.1
I (41) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (52) boot: Enabling RNG early entropy source...
I (56) boot: DRAM: lma 0x00001020 vma 0x3ffb0000 len 0x1940   (6464)
I (62) boot: IRAM: lma 0x00002968 vma 0x40080000 len 0x97d0   (38864)
I (69) boot: padd: lma 0x0000c148 vma 0x00000000 len 0x3eb0   (16048)
I (75) boot: IMAP: lma 0x00010000 vma 0x400d0000 len 0x3644   (13892)
I (81) boot: padd: lma 0x0001364c vma 0x00000000 len 0xc9ac   (51628)
I (87) boot: DMAP: lma 0x00020000 vma 0x3f400000 len 0xcd8    (3288)
I (93) boot: Image with 6 segments
I (96) boot: DROM segment: paddr=00020000h, vaddr=3f400000h, size=00CD8h (  3288) map
I (104) boot: IROM segment: paddr=00010000h, vaddr=400d0000h, size=03644h ( 13892) map
I (144) heap_runtime: ESP heap runtime init at 0x3ffb4170 size 175 kB.

*** Booting Zephyr OS build v4.0.0 ***
Hello World! esp32_devkitc_wroom/esp32/procpu
```
`CONFIG_PM=y` for esp32_devkitc_wroom device

**Environment (please complete the following information):**
- OS: Fedora 40
- Zephyr SDK: 0.16.5
- Zephyr 4.0.0
- CMake: 3.31.1
- Zephyr revision: 8469084dfae85f854555f0607f2c838dad097235

**Additional context**
```
CONFIG_PM=y
```
`prj.conf`

```
/*
 * Copyright (c) 2012-2014 Wind River Systems, Inc.
 *
 * SPDX-License-Identifier: Apache-2.0
 */

#include <zephyr/kernel.h>

int main(void)
{
	printk("Hello World! %s\n", CONFIG_BOARD_TARGET);

	return 0;
}
```
`main.c`

```
# SPDX-License-Identifier: Apache-2.0

cmake_minimum_required(VERSION 3.20.0)

find_package(Zephyr REQUIRED HINTS $ENV{ZEPHYR_BASE})
project(hello_world)

target_sources(app PRIVATE src/main.c)
```
`CMakeLists.txt`

LeoBriandFiveO, 2025-02-03:
Hi, could you try to compile with the --sysbuild option

acf879, 2025-02-03:
> Hi, could you try to compile with the --sysbuild option : `west build -p -b esp32s3_devkitc/esp32s3/procpu --sysbuild`

Hello Leo, 

This seemed to work thank you, not sure why the previous sysbuild that I had on it would have made `CONFIG_PM=y` to not show up (I had done the sysbuild prior to adding the `CONFIG_PM=y` to the prj.conf)

LeoBriandFiveO, 2025-02-03:
In this case, you are using "MCUboot 2nd stage bootloader" instead of "ESP Simple boot" as you can see in the first information log : `I (61) soc_init: MCUboot 2nd stage bootloader`

acf879, 2025-02-03:
> In this case, you are using "MCUboot 2nd stage bootloader" instead of "ESP Simple boot" as you can see in the first information log :
> 
> `I (61) soc_init: MCUboot 2nd stage bootloader`

That makes sense since I forgot to reflash with simple boot. Thank you!

LeoBriandFiveO, 2025-02-03:
The implementation of ESP simple boot differs between esp32s3 and esp32 (referring to https://github.com/zephyrproject-rtos/zephyr/blob/main/soc/espressif/esp32s3/soc.c and https://github.com/zephyrproject-rtos/zephyr/blob/main/soc/espressif/esp32/soc.c) that should explain the behavior difference between these two devices.