zephyrproject-rtos/zephyr #85035: RFC: Sensor: Enforcing build error on defining SENSOR_TRIGGER AND not defining sensor trigger int pin in the sensor node

narangmayank, 2025-02-02:
## Introduction

Enforcing build error on defining SENSOR_TRIGGER AND not defining sensor trigger int pin in the sensor node

### Problem description

To define sensor trigger mode we need to enable SENSOR_TRIGGER macro and provide the interrupt gpio pin in the dts overlay file and both go hand in hand.  So just defining SENSOR_TRIGGER and not providing interrupt gpio pin in the dts overlay doesn't make any sense. 

### Proposed change

Enforcing the build error on not defining the int pin in the kconfig itself

## Detailed RFC

### Proposed change (Detailed)

It's actually pretty simple, we can change the definition of sensor trigger in [kconfig sensor template](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/sensor/Kconfig.trigger_template#L25)

From 
```
config $(module)_TRIGGER
	bool
```

To 

```
config $(module)_TRIGGER
	bool
        depends on $(module)_INT_GPIO_ENABLED
```

$(module)_INT_GPIO_ENABLED should be defined in the sensor driver kconfig

This approach enforce the build error and let user define the interrupt pin before running the application

### Dependencies

Not sure about it

### Concerns and Unresolved Questions

This change will apply to all the sensor drivers which supports interrupt mode

## Alternatives

Use one of current implementation which doesn't marks any dependency on the pin and then giving run time error instead.

henrikbrixandersen, 2025-02-03:
Internal sensors can support triggers without the need for a trigger GPIO.

narangmayank, 2025-02-04:
Yes, This proposal is meant for external sensors only

henrikbrixandersen, 2025-02-04:
> Yes, This proposal is meant for external sensors only

We have external triggers downstream, where triggers are not based on an interrupt GPIO. I don't think you can generalize it this way.

narangmayank, 2025-02-04:
> We have external triggers downstream, where triggers are not based on an interrupt GPIO. I don't think you can generalize it this way.

Can you link me to one such instance where external triggers are not based on GPIO? I would be able to relate and put up my thoughts.

narangmayank, 2025-02-06:
@henrikbrixandersen @teburd @MaureenHelm Do you feel we can take this forward? I saw there a lot of sensor drivers in zephyr which has dependency on GPIO, we can also put this dependency in a separate kconfig file. I'm looking forward to your thoughts on it.

teburd, 2025-02-06:
I'd honestly love to see the entire premise of Kconfig for sensor triggers go away myself, its kind of a weird thing that doesn't make much sense anymore given devicetree.

henrikbrixandersen, 2025-02-06:
> I'd honestly love to see the entire premise of Kconfig for sensor triggers go away myself, its kind of a weird thing that doesn't make much sense anymore given devicetree.

I agree.