zephyrproject-rtos/zephyr #85043: stm32h7rs serie has no FLASH_OPTCR_OPTSTART for the flash stm32h7 driver

FRASTM, 2025-02-03:
After the merge of the https://github.com/zephyrproject-rtos/zephyr/pull/84430/

Test ebuilding of the 
`west build -p auto -b stm32h7s78_dk tests/subsys/secure_storage/psa/its/ -- -DEXTRA_CONF_FILE="overlay-secure_storage.conf;overlay-transform_default.conf;overlay-store_custom.conf"`

```
fails for the stm32h7rs serie:
flash_stm32h7x.c:72:24: error: 'FLASH_OPTCR_OPTSTART' undeclared (first use in this function); did you mean 'FLASH_OPTICR_OPTERRF'?

```
Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - What have you tried to diagnose or workaround this issue?
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
 - ...
-->

**To Reproduce**

**Expected behavior**

**Impact**

**Logs and console output**

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**

FRASTM, 2025-02-03:
The stm32h7rs serie has a **PG_OPT** bit in its FLASH_OPTCR register which in place of the FLASH_OPTCR_OPTSTART

FRASTM, 2025-02-03:
proposed patch : 
```patch
diff --git a/drivers/flash/flash_stm32h7x.c b/drivers/flash/flash_stm32h7x.c
index 7b842e44e2c..de9d5cd02c9 100644
--- a/drivers/flash/flash_stm32h7x.c
+++ b/drivers/flash/flash_stm32h7x.c
@@ -69,6 +69,16 @@ static __unused int commit_optb(const struct device *dev)
 	FLASH_TypeDef *regs = FLASH_STM32_REGS(dev);
 	int64_t timeout_time = k_uptime_get() + STM32H7_FLASH_OPT_TIMEOUT_MS;
 
+#ifdef FLASH_OPTCR_PG_OPT
+	regs->OPTCR |= FLASH_OPTCR_PG_OPT;
+	barrier_dsync_fence_full();
+	while (regs->SR & FLASH_SR_QW) {
+		if (k_uptime_get() > timeout_time) {
+			LOG_ERR("Timeout writing option bytes.");
+			return -ETIMEDOUT;
+		}
+	}
+#else
 	regs->OPTCR |= FLASH_OPTCR_OPTSTART;
 	barrier_dsync_fence_full();
 	while (regs->OPTSR_CUR & FLASH_OPTSR_OPT_BUSY) {
@@ -77,6 +87,7 @@ static __unused int commit_optb(const struct device *dev)
 			return -ETIMEDOUT;
 		}
 	}
+#endif
 
 	return 0;
 }
@@ -118,7 +129,12 @@ static __unused int write_opt(const struct device *dev, uint32_t mask, uint32_t
 static __unused int write_optsr(const struct device *dev, uint32_t mask, uint32_t value)
 {
 	FLASH_TypeDef *regs = FLASH_STM32_REGS(dev);
+#ifdef FLASH_OPTCR_PG_OPT
+	/* Assuming the OBW1SR is addressed, not OBW2SR */
+	uintptr_t cur = (uintptr_t)regs + offsetof(FLASH_TypeDef, OBW1SR);
+#else
 	uintptr_t cur = (uintptr_t)regs + offsetof(FLASH_TypeDef, OPTSR_CUR);
+#endif /* FLASH_OPTCR_PG_OPT */
 
 	return write_opt(dev, mask, value, cur, true);
 }
```

niedzwiecki-dawid, 2025-02-03:
`FLASH_STM32_READOUT_PROTECTION` is not supported for `SOC_SERIES_STM32H7RSX` chips. I assumed `commit_optb` and `write_opt` will be compiled out for `SOC_SERIES_STM32H7RSX`, but it depends on compiler options. I'll and a proper #ifdef

erwango, 2025-02-03:
>  I'll and a proper #ifdef. 

Please go ahead.

niedzwiecki-dawid, 2025-02-04:
I'll send PR once #83317 is merged.

duda-patryk, 2025-02-04:
> proposed patch :
> 
> diff --git a/drivers/flash/flash_stm32h7x.c b/drivers/flash/flash_stm32h7x.c
> index 7b842e44e2c..de9d5cd02c9 100644
> --- a/drivers/flash/flash_stm32h7x.c
> +++ b/drivers/flash/flash_stm32h7x.c
> @@ -69,6 +69,16 @@ static __unused int commit_optb(const struct device *dev)
>  	FLASH_TypeDef *regs = FLASH_STM32_REGS(dev);
>  	int64_t timeout_time = k_uptime_get() + STM32H7_FLASH_OPT_TIMEOUT_MS;
>  
> +#ifdef FLASH_OPTCR_PG_OPT
> +	regs->OPTCR |= FLASH_OPTCR_PG_OPT;
> +	barrier_dsync_fence_full();
> +	while (regs->SR & FLASH_SR_QW) {
> +		if (k_uptime_get() > timeout_time) {
> +			LOG_ERR("Timeout writing option bytes.");
> +			return -ETIMEDOUT;
> +		}
> +	}
> +#else
>  	regs->OPTCR |= FLASH_OPTCR_OPTSTART;
>  	barrier_dsync_fence_full();
>  	while (regs->OPTSR_CUR & FLASH_OPTSR_OPT_BUSY) {
> @@ -77,6 +87,7 @@ static __unused int commit_optb(const struct device *dev)
>  			return -ETIMEDOUT;
>  		}
>  	}
> +#endif
>  
>  	return 0;
>  }
> @@ -118,7 +129,12 @@ static __unused int write_opt(const struct device *dev, uint32_t mask, uint32_t
>  static __unused int write_optsr(const struct device *dev, uint32_t mask, uint32_t value)
>  {
>  	FLASH_TypeDef *regs = FLASH_STM32_REGS(dev);
> +#ifdef FLASH_OPTCR_PG_OPT
> +	/* Assuming the OBW1SR is addressed, not OBW2SR */
> +	uintptr_t cur = (uintptr_t)regs + offsetof(FLASH_TypeDef, OBW1SR);
> +#else
>  	uintptr_t cur = (uintptr_t)regs + offsetof(FLASH_TypeDef, OPTSR_CUR);
> +#endif /* FLASH_OPTCR_PG_OPT */
>  
>  	return write_opt(dev, mask, value, cur, true);
>  }

FLASH_OPTCR_PG_OPT bit must be set *before* writing new value to the option byte. It's more like enabling write access rather than commiting the changes.

erwango, 2025-02-04:
> FLASH_OPTCR_PG_OPT bit must be set before writing new value to the option byte. It's more like enabling write access rather than commiting the changes.

Initial request is first to prevent the compilation issue in case of H7RS. I think this is the main blocking issue today.
Functionality can come later.

duda-patryk, 2025-02-05:
Yes, I agree that the compilation issue should be fixed ASAP.  My previous comment was referring to the patch proposed by @FRASTM.

niedzwiecki-dawid, 2025-04-07:
Fixed with [#85550](https://github.com/zephyrproject-rtos/zephyr/pull/85550)

closed 2025-04-07 by niedzwiecki-dawid