zephyrproject-rtos/zephyr #85050: drivers: ethernet: eth_stm32_hal.c Configure Link According to configuration

clamattia, 2025-02-03:
**Describe the bug**
If auto-negotiation is disabled, the driver should take speed from the build configuration (kconfig). But if the speed is configured to be 10M, the system still advertises 100M. This causes the peer to establish a 100M link, that does not work.
For example the following configuration:
```
CONFIG_ETH_STM32_SPEED_10M=y
CONFIG_ETH_STM32_AUTO_NEGOTIATION_ENABLE=n
```

The reason for this is, that `phy_mii.c` configures all speed at init.
```C
		/* Advertise all speeds */
		phy_mii_cfg_link(dev, LINK_HALF_10BASE_T |
				      LINK_FULL_10BASE_T |
				      LINK_HALF_100BASE_T |
				      LINK_FULL_100BASE_T |
				      LINK_HALF_1000BASE_T |
				      LINK_FULL_1000BASE_T
				      );
```
And the `stm32_hal`-driver never rectifies that.

**To Reproduce**
Build with above configuration. A 100Mbit Link will be "established", even though this should not be possible.

**Workaround**
This fixes:
```C
		/* Advertise all speeds */
		phy_mii_cfg_link(dev, LINK_HALF_10BASE_T |
				      LINK_FULL_10BASE_T //|
				      //LINK_HALF_100BASE_T |
				      //LINK_FULL_100BASE_T |
				      //LINK_HALF_1000BASE_T |
				      //LINK_FULL_1000BASE_T
				      );
```
For a proper solution, `eth_stm32_hal.c` should probably call `cfg_link` on the `ethphy_driver_api` with the correct `enum phy_link_speed` after/during initialization.

**Expected behavior**
Only 10Mbit Links are possible under above configuration.

**Impact**
As there is currently a bug with auto-negotiation in this driver (fix in progress, see for example https://github.com/zephyrproject-rtos/zephyr/issues/81847), it is very likely, that user want to disable it in favor of fixed configuration. It would be great if fixed configuration would work.

**Logs and console output**
TBD

**Environment (please complete the following information):**
TBD

**Additional context**
TBD

clamattia, 2025-03-12:
Any news on this?

surendersampath, 2025-03-17:
I tried the above fix on NUCLEO-h723zg board. Before the fix it was receiving zero Bytes but I ended up with.
Can anyone help?

[image]

clamattia, 2025-03-17:
> [@clamattia](https://github.com/clamattia) Increasing the stack size and the phy_mii workaround fixed my issue. Thank you 🦸🏽 .

Thank you. Hopefully, there will be a proper fix instead of the workaround soon.

clamattia, 2025-04-09:
@marwaiehm-st @maass-hamburg 
With latest changes, auto-negotiation supposedly works (I have some potentially unrelated problems in my tests). Nonetheless it is not possible to configure a fixed speed.
As a workaround, I had to come between the `phy` and the `ethernet_driver` with a `SYS_INIT` function and configure the link according to my needs. The basic problem:
```
        /* Advertise all speeds */
        phy_mii_cfg_link(dev, LINK_HALF_10BASE_T |
                      LINK_FULL_10BASE_T |
                      LINK_HALF_100BASE_T |
                      LINK_FULL_100BASE_T |
                      LINK_HALF_1000BASE_T |
                      LINK_FULL_1000BASE_T);
```
Ideally this could be configured instead at build time, for example via devicetree?

Note, disabling the auto-negotiation is not a solution, as then it will have the speed it has and the `phy_mii` driver does not configure the speed at all (independent of which speed is set in the `dts`-overlay.

maass-hamburg, 2025-04-09:
> [@marwaiehm-st](https://github.com/marwaiehm-st) [@maass-hamburg](https://github.com/maass-hamburg) With latest changes, auto-negotiation supposedly works (I have some potentially unrelated problems [...]

Before the latest changes without auto-negotiation enabled (at least with the v2 eth hal api), would not configure the phy at all, it would only check the link up bit in the phys register. With the latest changes, when the ``ethernet-phy`` is used with a fixed link, that driver would not do anything.

Are you using a phy that supports gigabit on a ethernet mac that don't support gigabit or phy connection type is not gigabit?

The ethernet-phy driver can detect if the phy supports ethernet and will then also enable gigabit auto-negotiation.

maass-hamburg, 2025-04-09:
> > Are you using a phy that supports gigabit on a ethernet mac that don't support gigabit or phy connection type is not gigabit?
> 
> I do not understand the question. Currently I am using a dev-kit.
> 
> I am basically asking, if it would make sense to configure a fixed speed, even if the phy would allow for auto-negotiation?

You probably don't want to disable auto-negotiation completely, just restrict it, because when you disable auto-negotiation on the phy, you also have to disable it on the other side to get a link according to the specs. But being able to restrict it to multiple or even one option via the DT seems reasonable.