zephyrproject-rtos/zephyr #85070: drivers: usb: udc: udc_mcux_ehci: UDC_MCUX_EHCI tries to acquire mutex within the ISR

RadekPolyend, 2025-02-03:
#0  arch_system_halt (reason=reason@entry=4) at fatal.c:30
#1  0x086421c0 in k_sys_fatal_error_handler (reason=reason@entry=4, esf=esf@entry=0x1ffec890 <z_interrupt_stacks+1936>) at fatal.c:44
#2  0x086422e8 in z_fatal_error (reason=reason@entry=4, esf=esf@entry=0x1ffec890 <z_interrupt_stacks+1936>) at fatal.c:119
#3  0x0862fd6a in z_arm_fatal_error (reason=4, esf=0x1ffec890 <z_interrupt_stacks+1936>) at fatal.c:86
#4  0x086509c8 in z_do_kernel_oops (esf=<optimized out>, callee_regs=<optimized out>) at fatal.c:131
#5  0x08630762 in _oops () at /home/radekp/projects/zephyrproject/zephyr/arch/arm/core/cortex_m/swap_helper.S:450
#6  <signal handler called>
#7  assert_post_action (file=file@entry=0x30667990 "WEST_TOPDIR/zephyr/kernel/mutex.c", line=line@entry=111) at zephyr/lib/os/assert.c:43
#8  0x08643078 in z_impl_k_mutex_lock (mutex=<optimized out>, timeout=...) at zephyr/kernel/mutex.c:111
#9  0x086543d6 in k_mutex_lock (timeout=..., mutex=<optimized out>) at /zephyr/syscalls/kernel.h:986
#10 udc_lock_internal (dev=<optimized out>, timeout=...) at zephyr/drivers/usb/udc/udc_common.h:257
#11 0x086543e6 in udc_mcux_lock (dev=<optimized out>) at zephyr/drivers/usb/udc/udc_mcux_ehci.c:733
#12 0x08639598 in udc_ep_buf_alloc (dev=dev@entry=0x3065a19c <__device_dts_ord_524>, ep=<optimized out>, size=size@entry=8)
    at zephyr/drivers/usb/udc/udc_common.c:619
#13 0x08653b70 in udc_ctrl_alloc (dev=dev@entry=0x3065a19c <__device_dts_ord_524>, ep=ep@entry=0 '\000', size=size@entry=8)
    at zephyr/drivers/usb/udc/udc_common.c:643
#14 0x086397fa in udc_mcux_handler_setup (dev=dev@entry=0x3065a19c <__device_dts_ord_524>, setup=0x1ffe2030 <qh_buffer+48>)
    at zephyr/drivers/usb/udc/udc_mcux_ehci.c:212
#15 0x08639b8e in USB_DeviceNotificationTrigger (handle=<optimized out>, msg=msg@entry=0x1ffec92c <z_interrupt_stacks+2092>)
    at zephyr/drivers/usb/udc/udc_mcux_ehci.c:519
#16 0x086580b4 in USB_DeviceEhciInterruptTokenDone (ehciState=ehciState@entry=0x1ffe4f74 <g_UsbDeviceEhciState>)
    at modules/hal/nxp/mcux/middleware/mcux-sdk-middleware-usb/device/usb_device_ehci.c:643
#17 0x0865899a in USB_DeviceEhciIsrFunction (deviceHandle=<optimized out>)
    at modules/hal/nxp/mcux/middleware/mcux-sdk-middleware-usb/device/usb_device_ehci.c:2252
#18 0x086544cc in udc_mcux_isr (dev=<optimized out>) at zephyr/drivers/usb/udc/udc_mcux_ehci.c:540
#19 0x08630ad2 in _isr_wrapper () at zephyr/arch/arm/core/cortex_m/isr_wrapper.c:77

kartben, 2025-02-03:
@RadekPolyend  please use our issue reporting template -- we need a lot more than just a stacktrace to be able to properly understand and analyze the issue. Thanks

RadekPolyend, 2025-02-03:
> [@RadekPolyend](https://github.com/RadekPolyend) please use our issue reporting template -- we need a lot more than just a stacktrace to be able to properly understand and analyze the issue. Thanks

What more do you need ? The logic in the driver causes locking the mutex in ISR

jfischer-no, 2025-02-04:
> > [@RadekPolyend](https://github.com/RadekPolyend) please use our issue reporting template -- we need a lot more than just a stacktrace to be able to properly understand and analyze the issue. Thanks
> 
> What more do you need ? The logic in the driver causes locking the mutex in ISR

Please provide Zephyr commit SHA or version used and which sample was used to reproduce the the issue (Steps to reproduce the behavior).

RadekPolyend, 2025-02-04:
> > > [@RadekPolyend](https://github.com/RadekPolyend) please use our issue reporting template -- we need a lot more than just a stacktrace to be able to properly understand and analyze the issue. Thanks
> > 
> > 
> > What more do you need ? The logic in the driver causes locking the mutex in ISR
> 
> Please provide Zephyr commit SHA or version used and which sample was used to reproduce the issue (Steps to reproduce the behavior).

It is v3.7-branch, the reproduction is just to use this driver on imxrt1170 soc but the platform does not matter.

jfischer-no, 2025-02-04:
Should be fixed in commit f378de7e464f94b68d0a233bff8b1251c1235faf. New device support is experimental, there are many other changes in between, I am not sure yet if it is possible to backport.

closed 2025-02-04 by fabiobaltieri