zephyrproject-rtos/zephyr #85081: Using the LPSPI driver on MCXA156 with the generic "jedec,spi-nor" driver and a external flash memory only works with optimizations disabled

ehughes, 2025-02-03:
I am tracking an obscure bug with the shim based driver for LPSPI on the MCXA156. Even though the LPSPI driver passes the spi loopback test, I was able to get it to fail when using it with real devices.ways. 

**Setup:**

I have a FRDM-MCXA156 with a W25q64 SPI Flash memory wired to LPSPI1.

I have a FRDM-MCXA156 with a W25q64  SPI Flash memory wired to LPSPI1.

I am using the "spi_flash" sample as a basis for the my test.

Note:   As a control,  I also tested the same Flash chip wired to a FRDM-MCX947 (which also uses LPSPI) and a LPC55S69Xpresso (which has flexcomm).  The spi_flash runs OK on both of them.  I narrowed it down to some interaction w/ the MCXA156 / LPSPI 

I created an overlay : 

[image]

The nor flash driver driver works through boot and most of the test.  It can read the the JEDEC flash ID,  issue erase commands, etc.  However, I noticed that writes would fail.  After some debugging I observed this on my logic  analyzer:

[image]

The specific transaction for the flash write had the chip select active >200mS,  most of that idle.     The transaction after that was a command to shut down the flash as the SPI transaction for the write returned a timeout from the driver.

[image]

[image]

[image]

The flash write is successfull if all optimizations are disabled.

[image]

Looking through the LPSPI Flash driver, I found this:

[image]

If you follow the spi_transcieve_dt function:

[image]

 I noticed when optimizations are on,   This if statement never evaluates to true.  (Using Segger Ozone to single step and evaluate)

[image]

With optimizations off,  it will evaluate to true for the write transcations.

This does appear to be timing related.   If I turn on the SPI log dbg messages,  the sample will work.

I noticed that MCXN driver instance had a fifo size of 8 while MCXA had a size of 4.    If I reduce this size, the MCXA156 version works

[image]

Values of 1,2,3 yield a successful result with optimizations enabled.    A value of 4 (the dts default) will result in the timeout error.

Let me know if there is a test I can run, etc.    

 - OS: Windows
 - Toolchain : zephyr-sdk-0.17.0
 - Commit : [491bbcf](https://github.com/zephyrproject-rtos/zephyr/commit/491bbcfcbd35b5012ee9a1929eabeaa06ae556e4)

decsny, 2025-02-04:
@ehughes thanks for the tip about MCXW71, I changed the `transfer-delay` to just be the SCK period and that fixed the issue I was having. PR is here https://github.com/zephyrproject-rtos/zephyr/pull/85189 for that

decsny, 2025-03-12:
@ehughes I tried to look into this issue today , but realized I don't have the hardware to do it.  I will have to find some board with this flash (or any spi flash) to connect to the frdm mcxa156 board. 

With that said though, can you please try again on latest main (or 4.1.0 release), because I think since you made this report I have added some code in the interrupt handler to resolve (or at least mitigate) some race conditions and make it more robust.

Also, in the future, please for the items in the bug report that help with reproduce, such as overlays and config files, please put them as text instead of screenshot.

ehughes, 2025-03-30:
@decsny 

I tried testing this today.  Unfortunately there is busfault occuring early in the startup process:

[image]

I think this is the same issue as https://github.com/zephyrproject-rtos/zephyr/issues/87768

decsny, 2025-03-31:
> [@decsny](https://github.com/decsny)
> 
> I tried testing this today. Unfortunately there is busfault occuring early in the startup process:
> 
> [image]
> 
> I think this is the same issue as [#87768](https://github.com/zephyrproject-rtos/zephyr/issues/87768)

Can you try testing on 4.1 release, it's much more stable and has the fixes I mentioned, current main branch is a bit of a mess

decsny, 2025-04-24:
@ehughes main branch is much more stable now, even better than 4.1, I would try it again

ehughes, 2025-04-25:
@decsny   Thanks for the bump.  Will check.    I have some new hardware coming next week for MCXA that will be another good test case for LPSPI

ehughes, 2025-05-03:
@decsny 

Still no joy on this.   

Moved to commit Zephyr Commit 021d90b641bb5fd7e156c7be4f7a1e8a1646e000   (Main branch on 2025-05-03)

Summary:

Running SPI_FLASH sample on MCXA156

Modifications:

Added this to prj.conf

`
CONFIG_SPI_LOG_LEVEL_ERR=y
CONFIG_LOG=y
CONFIG_SPI_NOR_SFDP_RUNTIME=y
`

Added the overlay FRDM-MCXA156.overlay

`

&lpspi1 {
	status = "okay";
	pinctrl-0 = <&pinmux_lpspi1>;
	pinctrl-names = "default";

	transfer-delay=<500>;

	tx-fifo-size = <4>;
	rx-fifo-size = <4>;

	w25q64@0 {
		compatible = "jedec,spi-nor";
		size = <0x4000000>;	// 64 Mbits 
		reg = <0>;
		spi-max-frequency = <1000000>;
		jedec-id = [ef 40 18];
		status = "okay";
	};	
};

&pinctrl
{
	pinmux_lpspi1: pinmux_lpspi1 {
		group0 {
			pinmux = <LPSPI1_SDO_P3_8>,
				<LPSPI1_SCK_P3_10>,	
				<LPSPI1_SDI_P3_9>,	
				<LPSPI1_PCS0_P3_11>;
			slew-rate = "fast";
			drive-strength = "low";
			input-enable;
		};
	};

};
`

Will Yield:

`[00:00:00.000,000] <inf> spi_nor: w25q64@0: SFDP v 1.5 AP ff with 1 PH
[00:00:00.000,000] <inf> spi_nor: PH0: ff00 rev 1.5: 16 DW @ 80
[00:00:00.001,000] <inf> spi_nor: w25q64@0: 16 MiBy flash
*** Booting Zephyr OS build v4.1.0-3452-g021d90b641bb ***

w25q64@0 SPI flash testing
==========================

Perform test on single sector
Test 1: Flash erase
Flash erase succeeded!

Test 2: Flash write
Attempting to write 4 bytes
Flash write failed! -116
[00:00:00.000,000] <inf> spi_nor: w25q64@0: SFDP v 1.5 AP ff with 1 PH
[00:00:00.000,000] <inf> spi_nor: PH0: ff00 rev 1.5: 16 DW @ 80
[00:00:00.001,000] <inf> spi_nor: w25q64@0: 16 MiBy flash
*** Booting Zephyr OS build v4.1.0-3452-g021d90b641bb ***
[00:00:00.267,000] <err> spi_lpspi: Timeout waiting for transfer complete
`

2.)

Added this to prj.conf

`# extra configurationss
CONFIG_SPI_LOG_LEVEL_ERR=y
CONFIG_LOG=y
CONFIG_NO_OPTIMIZATIONS=n

CONFIG_MAIN_STACK_SIZE=8192
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=8192
CONFIG_ISR_STACK_SIZE=2048
`

Test passes

`
w25q64@0 SPI flash testing
==========================

Perform test on single sector
Test 1: Flash erase
[00:00:00.001,000] <inf> spi_nor: w25q64@0: SFDP v 1.5 AP ff with 1 PH
[00:00:00.001,000] <inf> spi_nor: PH0: ff00 rev 1.5: 16 DW @ 80
[00:00:00.002,000] <inf> spi_nor: w25q64@0: 16 MiBy flash
*** Booting Zephyr OS build v4.1.0-3452-g021d90b641bb ***
Flash erase succeeded!

Test 2: Flash write
Attempting to write 4 bytes
Data read matches data written. Good!!`

3.)

I can make #1 work (default, no optimizations) case work by modifying the tx/rx fifo sizes:

`
	tx-fifo-size = <1>;
	rx-fifo-size = <1>;
`

Any other value (2,3,4)  will not work

I could send a SPI flash breakout to wire into your FRDM board if it helps.