zephyrproject-rtos/zephyr #85106: Nominate contributor role for Renesas RZ Platform

binhnguyen2434, 2025-02-04:
# Background

The [TSC Project Roles] defines the main roles for the Zephyr Project, including
Maintainer, Collaborator, and Contributor.

By default anyone that contributes code or documentation is a Contributor, but
with the lowest [GitHub Permission Level] of Read. For example, Contributors
with Read permission do not have the permission to add reviewers to a pull
request.

Use this template to nominate a GitHub user for the Contributor role with
Triage permission level, which allows the user to add reviewers to a pull
request and be added as a reviewer by other users.

# Nomination

## GitHub User

Provide the following information about the GitHub user:

1. Binh Nguyen
1. binhnguyen2434
1. Renesas

## Supporting Documents

Add links to 3-5 GitHub pull requests, in the Zephyr project, authored or
reviewed by the GitHub user that demonstrate the user's dedication to the
Zephyr project.

https://github.com/zephyrproject-rtos/zephyr/pull/82016
https://github.com/zephyrproject-rtos/zephyr/pull/83773
https://github.com/zephyrproject-rtos/zephyr/pull/83419
https://github.com/zephyrproject-rtos/zephyr/pull/83211
https://github.com/zephyrproject-rtos/zephyr/pull/83251

[TSC Project Roles]: <https://docs.zephyrproject.org/latest/project/project_roles.html>
[GitHub Permission Level]: <https://docs.github.com/en/organizations/managing-access-to-your-organizations-repositories/repository-roles-for-an-organization>

KhiemNguyenT, 2025-02-04:
I support this nomination. They are Renesas member and will contribute to Zephyr support for RZ devices. Please consider it @kartben  @nashif .

binhnguyen2434, 2025-02-06:
duplicated https://github.com/zephyrproject-rtos/zephyr/issues/85261

closed 2025-02-06 by binhnguyen2434