zephyrproject-rtos/zephyr #85119: Intel®︎ Cyclone®︎ V SoC: No network interface is recognized

ruien612, 2025-02-04:
Hello everyone,

I have zepyhr running on the Cyclone V according to the following instructions: https://docs.nordicsemi.com/bundle/ncs-2.5.2/page/zephyr/boards/arm/cyclonev_socdk/doc/index.html 
Next I wanted to establish a simple network communication. I use the already written Example of a DHCP client under the directory ...**_/zephyrproject/zephyr/samples/net/dhcpv4_client_** but the following error message appears:
```
[00:00:00.207,000] <err> net_if: There is no network interface to work with!
*** Booting Zephyr OS build v4.0.0-3095-gd9eed167d06b ***
Run dhcpv4 client
[00:00:00.380,000] <wrn> net_if: No default interface found!
uart:~$ device list
devices:
- intc@fffed000 (READY)
  DT node labels: intc
- gpio@ff70a000 (READY)
  DT node labels: gpio2
- gpio@ff709000 (READY)
  DT node labels: gpio1
- gpio@ff708000 (READY)
  DT node labels: gpio0
- serial1@ffc03000 (READY)
  DT node labels: uart1
- serial0@ffc02000 (READY)
  DT node labels: uart0
- sysmgr@ffd08000 (READY)
  DT node labels: sysmgr
```

Normally, the Ethernet device should also be recognized (gmac0/gmac1) ...
I know that this example was not written explicitly for my hardware. I found the folder “Boards” in the project directory, which had a .conf file and an .overlay file, so I created two files cyclonev.conf & cyclonev.overlay. In the following I show my prj.conf and cyclonev.conf & cyclonev.overlay files:
prj.conf:
```
#No changes have been made
CONFIG_NETWORKING=y
CONFIG_NET_IPV6=n
CONFIG_NET_IPV4=y
CONFIG_NET_ARP=y
CONFIG_NET_UDP=y
CONFIG_NET_DHCPV4=y
CONFIG_NET_DHCPV4_OPTION_CALLBACKS=y
CONFIG_DNS_RESOLVER=y
CONFIG_TEST_RANDOM_GENERATOR=y
CONFIG_INIT_STACKS=y
CONFIG_NET_MGMT=y
CONFIG_NET_MGMT_EVENT=y
CONFIG_NET_LOG=y
CONFIG_LOG=y
```
cyclonev.conf:
```
CONFIG_ETH_CYCLONEV=y
CONFIG_NET_L2_ETHERNET=y
CONFIG_NET_MGMT=y
CONFIG_NET_DHCPV4=y
CONFIG_NET_TCP=y
CONFIG_NET_LOG=y
CONFIG_NET_IPV4=y
```
cyclonev.overlay:
```
&gmac1 {
    status = "okay";
    local-mac-address = [e6 88 2d 82 1f d4];
};
&gmac0 {
    status = "disabled"; 
};
```
I'll show you an excerpt from my zepyhr.dts file:
```
gmac0: ethernet@ff700000 {
		compatible = "snps,ethernet-cyclonev";
		reg = < 0xff700000 0x2000 >;
		interrupts = < 0x0 0x73 0x4 0xa0 >;
		emac-index = < 0x0 >;
		status = "disabled";
	};
gmac1: ethernet@ff702000 {
	compatible = "snps,ethernet-cyclonev";
	reg = < 0xff702000 0x2000 >;
	interrupts = < 0x0 0x78 0x4 0xa0 >;
	emac-index = < 0x1 >;
	status = "okay";
	local-mac-address = [ E6 88 2D 82 1F D4 ];
};
```

I have tried a lot of other things. For example I tried to enable gmac1 and disable gmac0 or disable both but without success. I had seen the approach with the own .conf and the .overlay file several times (e.g. for a STM32). So I think my approach is correct, but no network interface is recognized (Ethernent). I hope that someone sees my mistake and can help me. I would be very grateful for that. Finally, I'll insert a small screenshot from the documentation showing which Ethernet interface I'm using:

[image]

jukkar, 2025-02-04:
@ruien612 just to verify this. You are referring to instructions in NCS, are you running NCS too or are you using upstream Zephyr with your cyclone board?

ruien612, 2025-02-04:
Thanks for your answers @jukkar 
--> I installed Zepyhr on my Windows 10 machine according to the following instructions: https://docs.zephyrproject.org/latest/develop/getting_started/index.html
So I am using upstream Zephyr. I then continued with the following instructions to flash a Hello-World example to the Cyclone: https://docs.nordicsemi.com/bundle/ncs-2.5.2/page/zephyr/boards/arm/cyclonev_socdk/doc/index.html

--> Okay thanks for the hint, I will take _CONFIG_ETH_CYCLONEV=y_ out of my cyclone.conf file. In my build/zephyr/include/generated/zephyr/autoconf.h file the option _CONFIG_DT_HAS_SNPS_ETHERNET_CYCLONEV_ENABLED_ is found and this is set. Are the remaining configurations from my prj.conf, cyclonev.conf and cyclonev.overlay file then correct?
Even if I remove from my cyclone.conf file, I still get the same error that no network interface is recognized: `<err> net_if: There is no network interface to work with!`
Even when I run device list neither gmac0 nor gmac1 appears... 
Do you have any ideas what the problem could be?

fabiobaltieri, 2025-02-05:
```
		gmac0: ethernet@ff700000 {
			compatible = "snps,ethernet-cyclonev";
			reg = < 0xff700000 0x2000 >;
			interrupts = < 0x0 0x73 0x4 0xa0 >;
			emac-index = < 0x0 >;
			status = "disabled";
		};
		gmac1: ethernet@ff702000 {
			compatible = "snps,ethernet-cyclonev";
			reg = < 0xff702000 0x2000 >;
			interrupts = < 0x0 0x78 0x4 0xa0 >;
			emac-index = < 0x1 >;
			status = "okay";
			local-mac-address = [ E6 88 2D 82 1F D4 ];
			phy-mode = "rgmii-id";
		};
```

your override is not working, anyway converting this to discussion, you can also ask for help on http://chat.zephyrproject.org/ if it works better for you