zephyrproject-rtos/zephyr #85123: display_stm32_ltdc: Display dont draw GUI when powered on

hejpadigminvan, 2025-02-04:
**Describe the bug**
After reset_api replaced `__HAL_RCC_LTDC_FORCE_RESET()` and `__HAL_RCC_LTDC_RELEASE_RESET()` in `display_stm32_ltdc.c` my GUI dont get draw properly when the display gets turned off and on again by PM. I get a blank screen with a random color instead of my GUI.

I have traced it back to `stm32_ltdc_suspend()` (see picure). When i replace `reset_line_toggle_dt(&config->reset)` with the old method of using `__HAL_RCC_LTDC_FORCE_RESET()` and `__HAL_RCC_LTDC_RELEASE_RESET()` the GUI gets drawn as it should again.

[image]

I thougth it maybe was a wrong DTS configuration but since `stm32_ltdc_init()` also uses `reset_line_toggle_dt(&config->reset)` and that works fine (although together with `__HAL_RCC_LTDC_FORCE_RESET()` and `__HAL_RCC_LTDC_RELEASE_RESET()` in `stm32_ltdc_suspend()`) that cant be the issue.

Im using STM32H743 processor and running LVGL.

**To Reproduce**
Build with following kconfig symbols:
```
#define CONFIG_STM32_LTDC_FB_NUM 1
#define CONFIG_USE_STM32_HAL_LTDC 1
#define CONFIG_DT_HAS_ST_STM32_LTDC_ENABLED 1
#define CONFIG_STM32_LTDC 1
#define CONFIG_STM32_LTDC_RGB565 1
#define CONFIG_STM32_LTDC_DISABLE_FMC_BANK1 1
#define CONFIG_PM_DEVICE 1
```

**Expected behavior**

The GUI should get draw again when display gets turned on.

erwango, 2025-02-04:
Thanks @hejpadigminvan for reporting this.
Would you mind trying changing the following line:
https://github.com/zephyrproject-rtos/zephyr/blob/52b8e3d3706f7503d6e4e36413c2f78a2ac95480/dts/arm/st/h7/stm32h743.dtsi#L52
to:
```
 resets = <&rctl STM32_RESET(APB3, 3U)>; 
```
 ?

hejpadigminvan, 2025-02-04:
Thank you @erwango this solves the issue! I noticed that there was wrong number in the dtsi. I saw in Rm0433 Reference manual that it should be 3. I didn't change it because i believed that dts counted bits from 1 (hence 4U) instead of 0 for some reason so i thought all was good.

I'm thinking of changing career now 🤣 Time to pull down the curtains....

I will fix a PR on this tomorrow.

erwango, 2025-02-06:
Re-opening as reported on v4.0

FRASTM, 2025-03-17:
done