zephyrproject-rtos/zephyr #85142: Documentation on tracing inaccurate

toonst, 2025-02-04:
The zephyr documentation on tracing states the following:

> 

> [Named Trace Events](https://docs.zephyrproject.org/latest/services/tracing/index.html#named-trace-events)
> Although the user can extend any of the supported serialization formats to enable additional tracing functions (or provide their own backend), zephyr also provides one generic named tracing function for convenience purposes, as well as to demonstrate how tracing frameworks could be extended.
> Users can generate a custom trace event by calling sys_tracing_named_event(), which takes an event name as well as two arbitrary 4 byte arguments. Tracing backends may truncate the provided event name if it is too long for the serialization format they support.

There is however no `sys_tracing_named_event()` function available in Zephyr. How can I raise a named tracing event from application?

kartben, 2025-02-04:
https://docs.zephyrproject.org/latest/doxygen/html/group__subsys__tracing__apis__named.html#ga205bc648bdf0152f8652f04f15cf48af

It would be lovely if you could send a PR to fix this unfortunate typo (see #74037 for when it was introduced)! Thanks! :)

toonst, 2025-02-04:
Thanks for the reference, implemented fix here: https://github.com/zephyrproject-rtos/zephyr/pull/85177

closed 2025-02-05 by kartben