zephyrproject-rtos/zephyr #85155: samples/net/sockets/http_server fails to build for several nrf5340/cpuapp/ns targets

aescolar, 2025-02-04:
**Describe the bug**
samples/net/sockets/http_server fails to build for several nrf5340/cpuapp/ns targets

**To Reproduce**
Steps to reproduce the behavior:
1. twister -T samples/net/sockets/http_server -s sample.net.sockets.http.server --all
2. See error

**Expected behavior**
No build failures

**Impact**
Multiple failures in weekly CI

**Logs and console output**
https://github.com/zephyrproject-rtos/zephyr/actions/runs/13095023661/job/36536314669
```
[377/451] Building C object modules/mbedtls/CMakeFiles/mbedTLSCrypto.dir/scratch/zephyrproject/modules/crypto/mbedtls/library/pk.c.obj
modules/crypto/mbedtls/library/pk.c: In function 'mbedtls_pk_get_psa_attributes':
modules/crypto/mbedtls/library/pk.c:475:17: warning: implicit declaration of function 'mbedtls_ecc_group_to_psa'; did you mean 'mbedtls_ecp_group_copy'? [-Wimplicit-function-declaration]
  475 |                 mbedtls_ecc_group_to_psa(ec->grp.id, &bits);
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~
      |                 mbedtls_ecp_group_copy
modules/crypto/mbedtls/library/pk.c: In function 'copy_from_psa':
modules/crypto/mbedtls/library/pk.c:943:18: warning: implicit declaration of function 'mbedtls_ecc_group_from_psa'; did you mean 'mbedtls_pk_copy_from_psa'? [-Wimplicit-function-declaration]
  943 |         grp_id = mbedtls_ecc_group_from_psa(PSA_KEY_TYPE_ECC_GET_FAMILY(key_type), key_bits);
      |                  ^~~~~~~~~~~~~~~~~~~~~~~~~~
      |                  mbedtls_pk_copy_from_psa
```
```
ERROR   - raytac_mdbt53v_db_40/nrf5340/cpuapp/ns samples/net/sockets/http_server/sample.net.sockets.http.server ERROR: Build failure
..
ERROR   - raytac_mdbt53_db_40/nrf5340/cpuapp/ns samples/net/sockets/http_server/sample.net.sockets.http.server ERROR: Build failure
..
ERROR   - thingy53/nrf5340/cpuapp/ns samples/net/sockets/http_server/sample.net.sockets.http.server ERROR: Build failure
..
INFO    - Total complete:   84/  84  100%  built (not run):   75, filtered:  846, failed:    0, error:    3
```

**Environment (please complete the following information):**

 - OS: CI Ubuntu Linux 22.04
 - Toolchain (e.g Zephyr SDK, ...): 0.17.0
 - Commit SHA or Version used

**Additional context**
Introduced by https://github.com/zephyrproject-rtos/zephyr/pull/82143

aescolar, 2025-02-04:
CC @mrodgers-witekio @kartben @jukkar @rlubos @valeriosetti

kartben, 2025-02-04:
Thanks for taking the time to file the issue @aescolar! It is indeed making several tests fail in weekly CI so would be great to get this fixed :)

mrodgers-witekio, 2025-02-04:
I think this is the only networking sample using ECDSA certificates.

So I wonder if there's some wider issue with using ECDSA and TF-M, or possibly something wrong with the Crypto-related Kconfig options chosen in the sample? I can take a quick look on the sample app side and see if there's anything obvious

mrodgers-witekio, 2025-02-04:
At a guess I suspect this is coming from the fact that PSA crypto is not used, and we should use PSA crypto and the various `CONFIG_PSA_WANT_ALG*` options? But I'm not that familiar with the PSA crypto side of things so would be good if someone's able to confirm.

`CONFIG_MBEDTLS_USE_PSA_CRYPTO=n` in particular doesn't seem to appear in any other samples.

```
# TLS configuration
CONFIG_MBEDTLS=y
CONFIG_MBEDTLS_BUILTIN=y
CONFIG_MBEDTLS_ENABLE_HEAP=y
CONFIG_MBEDTLS_HEAP_SIZE=60000
CONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN=2048
CONFIG_MBEDTLS_USE_PSA_CRYPTO=n
CONFIG_NET_SOCKETS_SOCKOPT_TLS=y
CONFIG_NET_SOCKETS_TLS_MAX_CONTEXTS=6
CONFIG_TLS_CREDENTIALS=y
CONFIG_TLS_MAX_CREDENTIALS_NUMBER=5
CONFIG_MBEDTLS_ECDH_C=y
CONFIG_MBEDTLS_ECDSA_C=y
CONFIG_MBEDTLS_ECP_C=y
CONFIG_MBEDTLS_ECP_DP_SECP256R1_ENABLED=y
CONFIG_MBEDTLS_KEY_EXCHANGE_RSA_ENABLED=n
CONFIG_MBEDTLS_KEY_EXCHANGE_ECDHE_ECDSA_ENABLED=y
CONFIG_MBEDTLS_CIPHER_CCM_ENABLED=y
CONFIG_MBEDTLS_CIPHER_GCM_ENABLED=y
```

valeriosetti, 2025-02-04:
Adding `CONFIG_PSA_WANT_KEY_TYPE_ECC_PUBLIC_KEY=y` solves the failing build, i.e.
`twister -c -T samples/net/sockets/http_server -s sample.net.sockets.http.server -p thingy53/nrf5340/cpuapp/ns`

The reason is that `PSA_WANT_KEY_TYPE_ECC_PUBLIC_KEY` is the guard for `mbedtls_ecc_group_to_psa()`, of course. However I don't like this solution that much because in this test case we're using legacy Mbed TLS as Crypto support because `CONFIG_MBEDTLS_USE_PSA_CRYPTO=n` so the Mbed TLS code should not check for PSA_WANT symbols. I need to think a bit more on this to see if there's a more elegant way to fix it.

PR #85201 (2025-02-10): [fix] samples/net/sockets/http_server fails to build for several nrf5340/cpuapp/ns targets
Change the sample configuration in order to replace legacy Mbed TLS crypto support with PSA one. The test case is also then split in 2 to differentiate boards that have TF-M from those that do not. This is done to avoid duplication of the PSA Crypto core.

Resolves #85155

This PR started from a [suggestion](https://github.com/zephyrproject-rtos/zephyr/pull/85168#issuecomment-2634602994) from @mrodgers-witekio
closed 2025-02-10 by kartben