zephyrproject-rtos/zephyr #85166: samples: Unnecessary use of POSIX API in MQTT samples

erikmansson, 2025-02-04:
**Is your enhancement proposal related to a problem? Please describe.**

I might be missing something here, but it seems to me that all MQTT samples are unnecessarily using the POSIX sockets API (`CONFIG_POSIX_API=y`) instead of the BSD API prefixed with `zsock_`. This leads the reader to think that the POSIX API might be necessary to use MQTT when it's actually not. Enabling the POSIX API might lead to namespace conflicts ([docs](https://docs.zephyrproject.org/latest/connectivity/networking/api/sockets.html#overview)), which I noticed caused problems with mDNS.

For example, see `zephyr/samples/net/mqtt_publisher/prj.conf:12` and `zephyr/samples/net/mqtt_publisher/src/main.c:134`.

**Describe the solution you'd like**

If the above is true, I would like for the examples to remove `CONFIG_POSIX_API=y` and use the BSD API instead. Basically just using the `zsock_` prefix for the affected calls.

**Describe alternatives you've considered**

Investigate API conflicts in the mDNS implementation.

**Additional context**

If someone can verify that this solution is good, I can submit a PR.

cfriedt, 2025-02-04:
@erikmansson - good catch.

The main culprit here is the poll.h API, which was only separated from the zsock API less than a year ago. 

But you are correct - POSIX does not need to be enabled for this sample application.

I'll put together a PR.

erikmansson, 2025-02-04:
@cfriedt Thanks! I would gladly put together a PR on it to contribute, but it's up to you.

cfriedt, 2025-02-04:
> @cfriedt Thanks! I would gladly put together a PR on it to contribute, but it's up to you. 

Go for it! I got pulled into another meeting 🫠

erikmansson, 2025-02-04:
@cfriedt Done!

cfriedt, 2025-02-05:
Preemptive +1 from me

erikmansson, 2025-02-10:
Closing this since using the POSIX API is preferred in examples. Another issue should be opened for the mDNS issue.

closed 2025-02-10 by erikmansson