zephyrproject-rtos/zephyr #85172: ETH NXP MCXN: eth_nxp_enet_qos buffer problem

theadib, 2025-02-04:
**Describe the bug**

After a sudden event no more network communication is possible.
No reaction on ping. No TCP send, no UDP send. No TCP receive. No TCP connect.
The application remains responsive to i.e.button.

**To Reproduce**

Hard to reproduce.
My application runs on custom MCXN947 board, based on FRDM_MCXN947.
My application uses BSD sockets, TCP, UDP and poll(). IPV4 only.
I have a setup to tortoure my application.
TCP connect, transmit data, receive response, disconnect.
UDP sending data.
Sometimes the tests runs 18h with no problem. Sometimes the problem comes after 20min or 4hours.
in the serial log I see: `[01:13:22.386,000] <err> eth_nxp_enet_qos_mac: Could not alloc RX pkt`

My collegue found once a endless loop in eth_nxp_enet_qos_tx() see screenshot attached.

**Expected behavior**

No such buffer leak.

**Impact**

No workaround, reset needed.
Currently I have idea how to inspect if the problem exists and then possibly try to reinit the network stack.

**Logs and console output**

`[01:13:22.386,000] <err> eth_nxp_enet_qos_mac: Could not alloc RX pkt`
see also screenshot of debugger session with endless loop in eth_nxp_enet_qos_tx()

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows) Windows or Linux
 - Toolchain (e.g Zephyr SDK, ...) SDK17
 - Commit SHA or Version used Zephyr main

**Additional context**

theadib, 2025-02-04:
https://paste.pics/SRQG9
I had problems attaching the screenshot.
this pastebin will be deleted in 14days :-(

theadib, 2025-02-05:
Image uploaded from other PC. See endless loop in eth_nxp_enet_qos_tx(). Note the fragments counter ...
[image]

mmahadevan108, 2025-02-28:
@theadib , your PR has merged. Can we close this issue?

theadib, 2025-03-01:
Hello @mmahadevan108  please close this one. Thanks.

theadib, 2025-05-04:
I filed an other bug report #89435
That one is more precise in the reproducible of the problem and the possible reason.
I will close this one

closed 2025-05-04 by theadib