zephyrproject-rtos/zephyr #85207: Flash stm32 succeed but it still the old programs that runs

kirianguiller, 2025-02-05:
**Describe the bug**
On a stm32n6570-dk, I program the hello world application for the board with west, get the logs that it worked but when resetting the board, the old program (demonstration program for the board already present when purchasing it) is the program that runs.

**To Reproduce**
1. Install latest version of zephyr
2. run, at the root of your zephyr project : `west build -b stm32n6570_dk samples/hello_world`
3. run : `west flash`
4. run serial log monitor : `minicom -D /dev/ttyACM0`

**Expected behavior**
After resetting the board, the hello world program should start.

**Impact**
It is not working and thus troublesome

**Logs and console output**
Logs for `west flash` : 

```
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner stm32cubeprogrammer
      -------------------------------------------------------------------
                        STM32CubeProgrammer v2.18.0
      -------------------------------------------------------------------

ST-LINK SN  : 003B00363333511731363730
ST-LINK FW  : V3J15M7
Board       : STM32N6570-DK
Voltage     : 3.28V
SWD freq    : 8000 KHz
Connect mode: Hot Plug
Reset mode  : Software reset
Device ID   : 0x486
Revision ID : Rev A
Device name : STM32N6xx
Device type : MCU
Device CPU  : Cortex-M55
BL Version  : --

Warning: Write address is ignored for hex, Srec, s19 and elf files
Opening and parsing file: zephyr.hex

Memory Programming ...
  File          : zephyr.hex
  Size          : 18.53 KB
  Address       : 0x34180400

Erasing memory corresponding to segment 0:
Download in Progress:
[==================================================] 100%

File download complete
Time elapsed during download operation: 00:00:00.024

RUNNING Program ...
  Address:      : 0x8000000
Application is running, Please Hold on...
Start operation achieved successfully
```

**Environment (please complete the following information):**
Ubuntu 24.04
Python 3.12.3
west==1.3.0

Thanks !

kirianguiller, 2025-02-05:
For the people that have successful setup, what are the logs you are getting when `west flash` is executed ?

kirianguiller, 2025-02-05:
And to add some context, I have no problem when flashing other boards like Nucleo-F401RE

gautierg-st, 2025-02-05:
You appear to be flashing the `zephyr.hex` file which probably means that you didn't get the `zephyr.signed.bin` file during your compilation.

Do you have the STM32 Signing Tool installed (bundled with STM32CubeProgrammer) and in your path? It should execute at the end of the compilation and generated a signed `.bin`. It is this file that needs to be flashed in order to be retained.

Here is the (end of the) output of my compilation:
```
[139/139] Linking C executable zephyr\zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:          0 GB         0 GB
             RAM:       24616 B       511 KB      4.70%
        IDT_LIST:          0 GB        32 KB      0.00%
Generating files from C:/zephyrproject/zephyr/build/zephyr/zephyr.elf for board: stm32n6570_dk
      -------------------------------------------------------------------
                          STM32 Signing Tool v2.20.0
      -------------------------------------------------------------------

Adding head padding bytes for payload with header v2.3
Header version 2.3 preparation...
Extracting Entry point value from the input file...
Entry point value : 0x341810f9
The headred image file generated successfully: C:/zephyrproject/zephyr/build/zephyr/zephyr.signed.bin

...
...

Post header 2.3 information :
    Binary type: 0x10
    Non authenticated payload length: 0x0
    Non authenticated payload hash: 0x0
```

kirianguiller, 2025-02-05:
I was the problem. `STM32_SigningTool_CLI` was not in my path, which led to the .signed.bin not being compiled.

After I added to the path, I had to delete the build and then the flash worked correctly. Thank you very much

kirianguiller, 2025-02-05:
I don't know it official maintainers of Zephyr would prefer to do it or not. I reopen the issue

erwango, 2025-02-06:
Sure, we'll need to improve indeed.
Though, note that there is already a warning at build if `STM32_SigningTool_CLI` is not found:
https://github.com/zephyrproject-rtos/zephyr/commit/3d040683938052a033708821450771f607684d22
We cannot break here since it would always fail in Zephyr CI.
I'll investigate how to error out on `west flash` but this is slightly more complex.

Note also that `STM32_SigningTool_CLI` sits at the same place than CubeProgrammer executable, so if one is installed, the other should be there.

erwango, 2025-02-06:
@kirianguiller Would you have a check to https://github.com/zephyrproject-rtos/zephyr/pull/85332 and see if it would have been clearer ?

kirianguiller, 2025-02-08:
Yes, I find this clearer. Thank you!

closed 2025-02-26 by kartben