zephyrproject-rtos/zephyr #85215: MQTT client over secure Websockets cant connect to Azure Event-grid Namespaces MQTT broker

LonesomeUnixCowboy, 2025-02-05:
**Describe the bug**

Im trying to connect to our MQTT broker hosted at Azure over secure websockets. Im using Client Certificate and Client Key together with ISRGROOTX1 CA that Azure wants me to use.

Im setting up the connection like this with the address to our host at port `443`:

[image]

And setting up MQTT client like this:

[image]

I have seen that the connection is stopping after `s_headers_done` event in http_client `parser_execute()`. In `s_headers_done` event http_client figures out that the message is chunked and sets next event to `s_chunked_size_start`, after that ` http_wait_data()` timeouts with code `-116`.

[image]

[image]

I can connect to the broker with the same certificates on my PC using MQTTX, so seemingly it looks like the fault is in the mqtt/http_client. I'm in discussion with Azure support about this and they cant find out if something is wrong on their side but are gladly to help out with the problem through this issue.

rlubos, 2025-02-05:
Any chance for an unencrypted wireshark pcap of what Azure sends in a response for the websocket connection upgrade request?

I'm pretty sure it has nothing to do with certificates, as at the TLS level the client seems to connect just fine. What bugs me however is that the server replies with `transfer-encoding: chunked` on the switching protocols response - that makes the HTTP parser wait for "chunks", while in practice websocket have its own encoding. It'd probably makes sense to ask Azure support why are they sending that header back, I didn't see such behavior with any other MQTT broker when connecting over websockets.

But anyway, maybe you could try with the following patch to see if the chunked encoding is really the case here? That will force the HTTP parser to ignore the message body, whatever it is:
```patch
diff --git a/subsys/net/lib/http/http_client.c b/subsys/net/lib/http/http_client.c
index e61aa37a3ab..79047529d39 100644
--- a/subsys/net/lib/http/http_client.c
+++ b/subsys/net/lib/http/http_client.c
@@ -327,6 +327,10 @@ static int on_headers_complete(struct http_parser *parser)
                req->internal.response.http_cb->on_headers_complete(parser);
        }
 
+       if (parser->status_code == 101) {
+               return 1;
+       }
+
        if (parser->status_code >= 500 && parser->status_code < 600) {
                NET_DBG("Status %d, skipping body", parser->status_code);
                return 1;
```

LonesomeUnixCowboy, 2025-02-06:
I have made a half-assed atempt to get wireshark to decrypt the traffic but I could not get it to work, i will make a new atempt.

I have tried your patch and this work, I can connect to the eventgrid. I have also tried to connect to another test broker `mqtt.eclipseprojects.io` (also secure) and that worked fine (see pic)

[image]

In their header there is no `transfer-encoding: chunked`. I'm not to good with http responses or what is best practice for how a server/client should behave, but i will poke Azure on why they send this header the way they do. Hopefully the will chime in here.

rlubos, 2025-02-06:
I've opened https://github.com/zephyrproject-rtos/zephyr/pull/85321 for review just in case.

jukkar, 2025-02-06:
> The "fix" from [#85215 (comment)](https://github.com/zephyrproject-rtos/zephyr/issues/85215#issuecomment-2636962616) seems more of a workaround for buggy server to me, but at the same time it should be rather harmless in general, perhaps we could just commit this?

Yes, I agree, the change is harmless, +1 from me.

closed 2025-02-11 by kartben