zephyrproject-rtos/zephyr #85217: hello_world fails to build for rpi_pico2

samblenny, 2025-02-05:
**Describe the bug**

`west build -p always -b rpi_pico2/rp2350a/m33 samples/hello_world` fails during linking.

Error messages include:
- "FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /home/sam/code/zephyr-workspace/zephyr/build/zephyr/zephyr_pre0.map"
- "...conflicting CPU architectures 17/2"
- "...failed to merge target specific data of file..."

(see console log section below for additional details)

I get similar errors if I attempt to build other samples including `samples/basic/blinky` and `samples/basic/minimal`.

Target platform: Raspberry Pi Pico 2

Diagnosis: Searching for similar error messages comes up with discussions of an ABI mismatch between code compiled with ARM soft floating point and hard floating point.

I do not know if this is a regression or not.

**To Reproduce**

1. Prepare Zephyr workspace with zephyr commit d3d7e18 and zephyr-sdk-0.17.0
2. `cd zephyr`
3. `west build -p always -b rpi_pico2/rp2350a/m33 samples/hello_world`

**Expected behavior**

The `west build ...` command should finish without errors.

**Impact**

It may be the case that building anything for rpi_pico2 is currently broken?

**Logs and console output**

```
(.venv) $ west build -p always -b rpi_pico2/rp2350a/m33 samples/hello_world
-- west build: making build dir /home/sam/code/zephyr-workspace/zephyr/build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/sam/code/zephyr-workspace/zephyr/samples/hello_world
-- CMake version: 3.25.1
-- Found Python3: /home/sam/code/zephyr-workspace/.venv/bin/python3 (found suitable version "3.11.2", minimum required is "3.10") found components: Interpreter 
-- Cache files will be written to: /home/sam/.cache/zephyr
-- Zephyr version: 4.0.99 (/home/sam/code/zephyr-workspace/zephyr)
-- Found west (found suitable version "1.3.0", minimum required is "0.14.0")
-- Board: rpi_pico2, qualifiers: rp2350a/m33
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.17.0 (/home/sam/zephyr-sdk-0.17.0)
-- Found toolchain: zephyr 0.17.0 (/home/sam/zephyr-sdk-0.17.0)
-- Found Dtc: /home/sam/zephyr-sdk-0.17.0/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/sam/code/zephyr-workspace/zephyr/boards/raspberrypi/rpi_pico2/rpi_pico2_rp2350a_m33.dts
-- Generated zephyr.dts: /home/sam/code/zephyr-workspace/zephyr/build/zephyr/zephyr.dts
-- Generated pickled edt: /home/sam/code/zephyr-workspace/zephyr/build/zephyr/edt.pickle
-- Generated devicetree_generated.h: /home/sam/code/zephyr-workspace/zephyr/build/zephyr/include/generated/zephyr/devicetree_generated.h
-- Including generated dts.cmake file: /home/sam/code/zephyr-workspace/zephyr/build/zephyr/dts.cmake
Parsing /home/sam/code/zephyr-workspace/zephyr/Kconfig
Loaded configuration '/home/sam/code/zephyr-workspace/zephyr/boards/raspberrypi/rpi_pico2/rpi_pico2_rp2350a_m33_defconfig'
Merged configuration '/home/sam/code/zephyr-workspace/zephyr/samples/hello_world/prj.conf'
Configuration saved to '/home/sam/code/zephyr-workspace/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/sam/code/zephyr-workspace/zephyr/build/zephyr/include/generated/zephyr/autoconf.h'
-- Found GnuLd: /home/sam/zephyr-sdk-0.17.0/arm-zephyr-eabi/arm-zephyr-eabi/bin/ld.bfd (found version "2.38") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/sam/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Using ccache: /usr/bin/ccache
-- Configuring done
-- Generating done
-- Build files have been written to: /home/sam/code/zephyr-workspace/zephyr/build
[... 29 lines cut ...]
/home/sam/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld: failed to merge target specific data of file /home/sam/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/../../picolibc/arm-zephyr-eabi/lib/./libc.a(libc_tinystdio_vprintf.c.o)
/home/sam/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld: error: /home/sam/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/../../picolibc/arm-zephyr-eabi/lib/./libc.a(libc_string_strchr.c.o): conflicting CPU architectures 17/2
/home/sam/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld: failed to merge target specific data of file /home/sam/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/../../picolibc/arm-zephyr-eabi/lib/./libc.a(libc_string_strchr.c.o)
/home/sam/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld: error: /home/sam/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/../../picolibc/arm-zephyr-eabi/lib/./libc.a(libc_tinystdio_fgetc.c.o): conflicting CPU architectures 17/2
/home/sam/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld: failed to merge target specific data of file /home/sam/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/../../picolibc/arm-zephyr-eabi/lib/./libc.a(libc_tinystdio_fgetc.c.o)
/home/sam/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld: error: /home/sam/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/../../picolibc/arm-zephyr-eabi/lib/./libc.a(libc_tinystdio_ungetc.c.o): conflicting CPU architectures 17/2
/home/sam/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld: failed to merge target specific data of file /home/sam/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/../../picolibc/arm-zephyr-eabi/lib/./libc.a(libc_tinystdio_ungetc.c.o)
/home/sam/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld: error: /home/sam/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/../../picolibc/arm-zephyr-eabi/lib/./libc.a(libc_tinystdio_compare_exchange.c.o): conflicting CPU architectures 17/2
/home/sam/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld: failed to merge target specific data of file /home/sam/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/../../picolibc/arm-zephyr-eabi/lib/./libc.a(libc_tinystdio_compare_exchange.c.o)
/home/sam/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld: error: /home/sam/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/../../picolibc/arm-zephyr-eabi/lib/./libc.a(libc_tinystdio_exchange.c.o): conflicting CPU architectures 17/2
/home/sam/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld: failed to merge target specific data of file /home/sam/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/../../picolibc/arm-zephyr-eabi/lib/./libc.a(libc_tinystdio_exchange.c.o)
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/sam/code/zephyr-workspace/zephyr/build
```

**Environment (please complete the following information):**

 - OS: Linux (Debian 12)
 - Toolchain: zephyr-sdk-0.17.0
 - Commit: d3d7e18

**Additional context**

n/a

samblenny, 2025-02-05:
I'm not an expert on tracking down this type of error, so I may be off the mark here. But, from what I can tell, the first error,

 `...picolibc/arm-zephyr-eabi/lib/./libc.a(strcmp.S.o): conflicting CPU architectures 17/2`

 is probably happening as a result of trying to link these two files:
- `zephyr/kernel/libkernel.a`
- `zephyr-sdk-0.17.0/arm-zephyr-eabi/picolibc/arm-zephyr-eabi/lib/libc.a`

I say that because, if I use grep to look for `strcmp` in `build/zephyr/zephyr_pre0.map`, I get this:
```
(.venv) $ grep strcmp build/zephyr/zephyr_pre0.map 
/home/sam/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/../../picolibc/arm-zephyr-eabi/lib/./libc.a(strcmp.S.o)
                              zephyr/kernel/libkernel.a(device.c.obj) (strcmp)
 .text          0x0000000000000000       0x1c /home/sam/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/../../picolibc/arm-zephyr-eabi/lib/./libc.a(strcmp.S.o)
 .data          0x0000000000000000        0x0 /home/sam/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/../../picolibc/arm-zephyr-eabi/lib/./libc.a(strcmp.S.o)
 .bss           0x0000000000000000        0x0 /home/sam/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/../../picolibc/arm-zephyr-eabi/lib/./libc.a(strcmp.S.o)
 .ARM.extab     0x0000000000000000        0x0 /home/sam/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/../../picolibc/arm-zephyr-eabi/lib/./libc.a(strcmp.S.o)
 .ARM.exidx     0x0000000000000000        0x8 /home/sam/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/../../picolibc/arm-zephyr-eabi/lib/./libc.a(strcmp.S.o)
 .debug_frame   0x0000000000000000       0x20 /home/sam/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/../../picolibc/arm-zephyr-eabi/lib/./libc.a(strcmp.S.o)
                0x000000000000109a       0x1a /home/sam/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/../../picolibc/arm-zephyr-eabi/lib/./libc.a(strcmp.S.o)
```

When I use `objdump` to look at the picolibc file from the zephyr-sdk-0.17.0/arm-zephyr-eabi toolchain, I see it appears to have been compiled for `armv4t`:

```
(.venv) $ ~/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-objdump -f \
    ~/zephyr-sdk-0.17.0/arm-zephyr-eabi/picolibc/arm-zephyr-eabi/lib/libc.a | grep -A 4 strcmp.S.o
strcmp.S.o:     file format elf32-littlearm
architecture: armv4t, flags 0x00000011:
HAS_RELOC, HAS_SYMS
start address 0x00000000
```

When I look at `libkernel.a`, it seems to have been built for `armv8-m`

```
(.venv) $ ~/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-objdump -f \
    build/zephyr/kernel/libkernel.a | head -n 6
In archive build/zephyr/kernel/libkernel.a:

main_weak.c.obj:     file format elf32-littlearm
architecture: armv8-m.main, flags 0x00000011:
HAS_RELOC, HAS_SYMS
start address 0x00000000
```

samblenny, 2025-02-05:
If I build for Pi Pico, which links successfully, the architecture in `build/zephyr/kernel/libkernel.a` is `armv6s-m`:

```
(.venv) $ rm -r build/
(.venv) $ west build -p always -b rpi_pico samples/hello_world
...
(.venv) $ ~/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-objdump -f build/zephyr/kernel/libkernel.a | head -n 6
In archive build/zephyr/kernel/libkernel.a:

main_weak.c.obj:     file format elf32-littlearm
architecture: armv6s-m, flags 0x00000011:
HAS_RELOC, HAS_SYMS
start address 0x00000000
```

samblenny, 2025-02-05:
Maybe the problem is that `west build` is trying to link against the picolibc library that comes with zephyr-sdk-0.17.0/arm-zephyr-eabi. Perhaps it should instead build picolibc from source using the picolibc module?

xudongzheng, 2025-02-05:
Did you upgrade your Zephyr SDK recently?

The error message seems quite similar to https://github.com/zephyrproject-rtos/sdk-ng/issues/854, which was resolved with `rm -rf ~/.cache`.

samblenny, 2025-02-05:
> Did you upgrade your Zephyr SDK recently?

No. For the development machine I've been using, 0.17.0 was the first and only Zephyr SDK version that I've installed.

> The error message seems quite similar to https://github.com/zephyrproject-rtos/sdk-ng/issues/854, which was resolved with rm -rf ~/.cache.

Hmm... deleting the cache actually seems to help. When I tried this, the build worked:
```
rm -r ~/.cache/zephyr
rm -r build
west build -p always -b rpi_pico2/rp2350a/m33 samples/hello_world
``` 

This time, when I grep the map file for strcmp, it's looking in `zephyr-sdk-0.17.0/arm-zephyr-eabi/picolibc/arm-zephyr-eabi/lib/thumb/v8-m.main/nofp/libc.a`:

```
(.venv) $ grep strcmp build/zephyr/zephyr_pre0.map
/home/sam/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/../../picolibc/arm-zephyr-eabi/lib/thumb/v8-m.main/nofp/libc.a(strcmp.S.o)
                              zephyr/kernel/libkernel.a(device.c.obj) (strcmp)
 .text          0x0000000000000000       0x14 /home/sam/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/../../picolibc/arm-zephyr-eabi/lib/thumb/v8-m.main/nofp/libc.a(strcmp.S.o)
 .data          0x0000000000000000        0x0 /home/sam/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/../../picolibc/arm-zephyr-eabi/lib/thumb/v8-m.main/nofp/libc.a(strcmp.S.o)
 .bss           0x0000000000000000        0x0 /home/sam/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/../../picolibc/arm-zephyr-eabi/lib/thumb/v8-m.main/nofp/libc.a(strcmp.S.o)
 .ARM.extab     0x0000000000000000        0x0 /home/sam/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/../../picolibc/arm-zephyr-eabi/lib/thumb/v8-m.main/nofp/libc.a(strcmp.S.o)
 .ARM.exidx     0x0000000000000000        0x8 /home/sam/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/../../picolibc/arm-zephyr-eabi/lib/thumb/v8-m.main/nofp/libc.a(strcmp.S.o)
 .debug_frame   0x0000000000000000       0x20 /home/sam/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/../../picolibc/arm-zephyr-eabi/lib/thumb/v8-m.main/nofp/libc.a(strcmp.S.o)
                0x000000000000109a       0x20 /home/sam/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/../../picolibc/arm-zephyr-eabi/lib/thumb/v8-m.main/nofp/libc.a(strcmp.S.o)

```

WilliamGFish, 2025-03-13:
Hello_world is not compiling for the RPi_Pico/rp2040/w...failure in 'sys_rand_get'

soburi, 2025-03-14:
@WilliamGFish 

> Hello_world is not compiling for the RPi_Pico/rp2040/w...failure in 'sys_rand_get'

I've checked, and there are no issues.
Please create a new issue if necessary.

pshannonsouthco, 2025-11-03:
above comment can be ignored. Apparently I placed my cmake <project()> function above the <find_package(Zephyr REQUIRED HINTS $ENV{ZEPHYR_BASE})> function and that corrupted my build system somehow