zephyrproject-rtos/zephyr #85246: Zephyr not booting with SPIRAM enabled on esp_wrover_kit

Mierunski, 2025-02-05:
**Describe the bug**

I'm using esp_wrover_kit/esp32/procpu on in custom project, enabling SPIRAM is halting in mcuboot phase. I can reproduce the same issue with hello_world sample.

**To Reproduce**

1. cd samples/hello_world
2. west build -b esp_wrover_kit/esp32/procpu -p --sysbuild
3. west build -t menuconfig
4. Enable `CONFIG_ESP_SPIRAM`
5. west flash
6. west espressif monitor
7. See that board halts during mcuboot

**Expected behavior**

App boots without problems

**Impact**

Showstopper, as internal RAM is not enough for application

**Logs and console output**

```
(.venv) ~/zephyrproject/zephyr/samples/hello_world (main) $ west flash
-- west flash: rebuilding
[0/6] Performing build step for 'hello_world'
[0/1] Re-running CMake...
Loading Zephyr default modules (Zephyr base (cached)).
-- Application: /home/sashir/zephyrproject/zephyr/samples/hello_world
-- CMake version: 3.31.5
-- Cache files will be written to: /home/sashir/.cache/zephyr
-- Zephyr version: 4.0.99 (/home/sashir/zephyrproject/zephyr)
-- Found west (found suitable version "1.3.0", minimum required is "0.14.0")
-- Board: esp_wrover_kit, qualifiers: esp32/procpu
-- Found host-tools: zephyr 0.17.0 (/home/sashir/zephyr-sdk-0.17.0)
-- Found toolchain: zephyr 0.17.0 (/home/sashir/zephyr-sdk-0.17.0)
-- Found BOARD.dts: /home/sashir/zephyrproject/zephyr/boards/espressif/esp_wrover_kit/esp_wrover_kit_procpu.dts
-- Generated zephyr.dts: /home/sashir/zephyrproject/zephyr/samples/hello_world/build/hello_world/zephyr/zephyr.dts
-- Generated pickled edt: /home/sashir/zephyrproject/zephyr/samples/hello_world/build/hello_world/zephyr/edt.pickle
-- Generated devicetree_generated.h: /home/sashir/zephyrproject/zephyr/samples/hello_world/build/hello_world/zephyr/include/generated/zephyr/devicetree_generated.h
-- Including generated dts.cmake file: /home/sashir/zephyrproject/zephyr/samples/hello_world/build/hello_world/zephyr/dts.cmake
Parsing /home/sashir/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/sashir/zephyrproject/zephyr/samples/hello_world/build/hello_world/zephyr/.config'
Merged configuration '/home/sashir/zephyrproject/zephyr/samples/hello_world/build/hello_world/zephyr/.config.sysbuild'
No change to configuration in '/home/sashir/zephyrproject/zephyr/samples/hello_world/build/hello_world/zephyr/.config'
Kconfig header saved to '/home/sashir/zephyrproject/zephyr/samples/hello_world/build/hello_world/zephyr/include/generated/zephyr/autoconf.h'
-- Espressif HAL path: /home/sashir/zephyrproject/modules/hal/espressif
-- Image partition /soc/flash-controller@3ff42000/flash@0/partitions/partition@20000
-- Using ccache: /usr/bin/ccache
-- Including signing script: /home/sashir/zephyrproject/zephyr/cmake/mcuboot.cmake
-- Configuring done (2.2s)
-- Generating done (0.1s)
-- Build files have been written to: /home/sashir/zephyrproject/zephyr/samples/hello_world/build/hello_world
[196/196] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
     mcuboot_hdr:          32 B         32 B    100.00%
        metadata:          80 B         96 B     83.33%
           FLASH:      145016 B    4194176 B      3.46%
     iram0_0_seg:       23808 B       224 KB     10.38%
     dram0_0_seg:       12720 B       192 KB      6.47%
     irom0_0_seg:       14072 B      4092 KB      0.34%
     drom0_0_seg:         64 KB      4092 KB      1.56%
    rtc_iram_seg:          0 GB         8 KB      0.00%
    rtc_slow_seg:          0 GB         8 KB      0.00%
    rtc_data_seg:          0 GB         8 KB      0.00%
     ext_ram_seg:      262136 B         2 MB     12.50%
        IDT_LIST:          0 GB         8 KB      0.00%
Generating files from /home/sashir/zephyrproject/zephyr/samples/hello_world/build/hello_world/zephyr/zephyr.elf for board: esp_wrover_kit
[1/6] Performing build step for 'mcuboot'
ninja: no work to do.
[6/6] Completed 'mcuboot'
-- west flash: using runner esp32
-- runners.esp32: reset after flashing requested
-- runners.esp32: Flashing esp32 chip on None (921600bps)
esptool.py v4.7.0
Found 34 serial ports
Serial port /dev/ttyUSB1
Connecting......
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 08:3a:8d:15:fd:d8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00001000 to 0x0000efff...
SHA digest in image updated
Wrote 65536 bytes at 0x00001000 in 1.1 seconds (461.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
-- west flash: using runner esp32
-- runners.esp32: reset after flashing requested
-- runners.esp32: Flashing esp32 chip on None (921600bps)
esptool.py v4.7.0
Found 34 serial ports
Serial port /dev/ttyUSB1
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting.....
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 08:3a:8d:15:fd:d8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00020000 to 0x00043fff...
Wrote 147456 bytes at 0x00020000 in 2.0 seconds (585.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
(.venv) ~/zephyrproject/zephyr/samples/hello_world (main) $ west espressif monitor
Serial port /dev/ttyUSB1
Connecting.....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
--- idf_monitor on /dev/ttyUSB1 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 12:21:46
rst:0x1 (POWERON_RESET),boot:0x1e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3ffe8000,len:10856
load:0x40078000,len:5948
ho 0 tail 12 room 4
load:0x400a0000,len:37356
entry 0x400a2c10
I (36) soc_init: MCUboot 2nd stage bootloader
I (37) soc_init: compile time Feb  5 2025 18:59:52
W (37) soc_init: Unicore bootloader
I (38) spi_flash: detected chip: generic
I (42) spi_flash: flash io: dio
I (45) soc_init: chip revision: v3.1
I (48) flash_init: SPI Speed      : 40MHz
I (52) flash_init: SPI Mode       : DIO
I (55) flash_init: SPI Flash Size : 4MB
I (90) spi_flash: flash io: dio
[esp32] [INF] Image index: 0, Swap type: none
[esp32] [INF] Loading image 0 - slot 0 from flash, area id: 1
[esp32] [INF] Application start=40083a4ch
[esp32] [INF] DRAM segment: paddr=00025d70h, vaddr=3ffb0000h, size=00738h (  1848) load
[esp32] [INF] IRAM segment: paddr=00020080h, vaddr=40080000h, size=05cf0h ( 23792) load
I (122) boot: DROM segment: paddr=00030000h, vaddr=3f400000h, size=01044h (  4164) map
I (122) boot: IROM segment: paddr=00040000h, vaddr=400d0000h, size=036F8h ( 14072) map
I (140) boot: libc heap size 179 kB.
I (140) quad_psram: This chip is ESP32-D0WD
I (140) esp_psram: Found 8MB PSRAM device
I (140) esp_psram: Speed: 40MHz
I (144) esp_psram: PSRAM initialized, cache is in normal (1-core) mode.

```

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit, current main - be91cfedfb54f9022dc0aa02c577b57b35f887a4

**Additional context**

rftafas, 2025-02-06:
We have pending updates to support ESP32 rev 3.1. This is more like a 'request' than a bug.

Mierunski, 2025-02-11:
This seems more like an issue with building with SPIRAM when sysbuild and mcuboot is enabled,
```
/home/sashir/zephyr-sdk-0.17.0/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32_zephyr-elf/bin/ld.bfd: zephyr/libzephyr.a(esp_psram_impl_quad.c.obj):(.iram1.10.literal+0xb0): undefined reference to `bootloader_common_vddsdio_configure'
/home/sashir/zephyr-sdk-0.17.0/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32_zephyr-elf/bin/ld.bfd: zephyr/libzephyr.a(esp_psram_impl_quad.c.obj): in function `esp_psram_impl_enable':
/home/sashir/zephyrproject/modules/hal/espressif/components/esp_psram/esp32/esp_psram_impl_quad.c:936: undefined reference to `bootloader_common_vddsdio_configure'
collect2: error: ld returned 1 exit status
```

sylvioalves, 2025-02-11:
This is already being handled. I guess PR will be avaiable in a couple of days.

PR #85625 (2025-02-14): soc: espressif: rework code startup to improve flash and spiram loader
This PR reworks all Espressif's SOC initialization code to fix booting issues regaring OCTAL FLASH/SPIRAM.
This also updates bootloader (either Simple Boot and MCUBoot) to use ROM functions during flash read/write operations, ensuring no dependencies of the flash driver initialization.
Runner handler is now common to all SoCs and fixes flash mode and frequency configuration, required for some flash memories.
Esptool is also updated to v4.8.1.

Fixes #64757
Fixes #85246
Fixes #83743
Fixes #82244