zephyrproject-rtos/zephyr #85253: Not able to show image from SD card on display with LVGL

denny61302, 2025-02-05:
**Describe the bug**

I tried to show images from the SD card with LVGL.

So far I can confirm I can show text label on the display (ST7789, RGB565), and I can mount and open the image file on the SD card.

However, I am not able to show any image on the display. I tried the PNG image (fail), so I converted it with LVGL image converter to BIN (fail). 

**To Reproduce**

Here are my config related to LVGL

CONFIG_LVGL=y
CONFIG_LV_Z_MEM_POOL_SIZE=16384
CONFIG_LV_Z_SHELL=y
CONFIG_LV_USE_LOG=y
CONFIG_LV_USE_LABEL=y
CONFIG_LV_USE_IMG=y
CONFIG_LV_FONT_MONTSERRAT_14=y
CONFIG_LV_THEME_DEFAULT_DARK=n
CONFIG_LV_USE_PNG=y
CONFIG_LV_Z_USE_FILESYSTEM=y
CONFIG_LV_MEM_CUSTOM=y

My snippet related to show image.

lv_obj_t * img;

img = lv_img_create(lv_scr_act());

lv_img_set_src(img, "/SD:/BLUE.PNG");
lv_obj_align(img, LV_ALIGN_CENTER, 10, 10); 

**Expected behavior**

**Impact**

**Logs and console output**

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

Windows 10
nRFConnect SDK v2.9.99

**Additional context**

kartben, 2025-02-06:
I don't think we integrate LVGL file system abstraction and therefore we don't support providing URIs to set_img. You'd need to read the contents of the file using Zephyr API and turn it into an LVGL image descriptor

kartben, 2025-02-06:
Nevermind, we do indeed support it, sorry (and neat!) so I'll let @faxe1008 comment  :)

faxe1008, 2025-02-06:
@denny61302 I took Zephyr rev `769df0b579b` and modified the gui test to actually show the SDL display. Here is the diff:

```
diff --git a/boards/native/native_sim/native_sim.dts b/boards/native/native_sim/native_sim.dts
index 162fcf07457..c8767da37ca 100644
--- a/boards/native/native_sim/native_sim.dts
+++ b/boards/native/native_sim/native_sim.dts
@@ -61,7 +61,7 @@
 
 	flashcontroller0: flash-controller@0 {
 		compatible = "zephyr,sim-flash";
-		reg = <0x00000000 DT_SIZE_K(2048)>;
+		reg = <0x00000000 DT_SIZE_K(8048)>;
 
 		#address-cells = <1>;
 		#size-cells = <1>;
@@ -72,7 +72,7 @@
 			compatible = "soc-nv-flash";
 			erase-block-size = <4096>;
 			write-block-size = <1>;
-			reg = <0x00000000 DT_SIZE_K(2048)>;
+			reg = <0x00000000 DT_SIZE_K(8048)>;
 
 			partitions {
 				compatible = "fixed-partitions";
@@ -97,7 +97,7 @@
 				};
 				storage_partition: partition@fc000 {
 					label = "storage";
-					reg = <0x000fc000 0x00004000>;
+					reg = <0x000fc000 0x00080000>;
 				};
 			};
 		};
diff --git a/tests/lib/gui/lvgl/app.overlay b/tests/lib/gui/lvgl/app.overlay
index 799664d1a19..c8978e2952a 100644
--- a/tests/lib/gui/lvgl/app.overlay
+++ b/tests/lib/gui/lvgl/app.overlay
@@ -5,13 +5,5 @@
  */
 
 / {
-	chosen {
-		zephyr,display = &dummy_dc;
-	};
 
-	dummy_dc: dummy_dc {
-		compatible = "zephyr,dummy-dc";
-		height = <240>;
-		width = <320>;
-	};
 };
diff --git a/tests/lib/gui/lvgl/prj.conf b/tests/lib/gui/lvgl/prj.conf
index 30b8e9b273b..287c2a7a627 100644
--- a/tests/lib/gui/lvgl/prj.conf
+++ b/tests/lib/gui/lvgl/prj.conf
@@ -1,7 +1,6 @@
-CONFIG_ZTEST=y
 
 CONFIG_DISPLAY=y
-CONFIG_SDL_DISPLAY=n
+CONFIG_SDL_DISPLAY=y
 
 CONFIG_LOG=y
 
@@ -9,7 +8,6 @@ CONFIG_FLASH=y
 CONFIG_FLASH_MAP=y
 CONFIG_FILE_SYSTEM=y
 CONFIG_FILE_SYSTEM_LITTLEFS=y
-CONFIG_LV_Z_AUTO_INIT=n
[... 338 lines cut ...]
+		return 0;
+	}
 
-ZTEST_SUITE(lvgl_screen, NULL, setup_lvgl, NULL, NULL, teardown_lvgl);
-ZTEST_SUITE(lvgl_fs, NULL, setup_fs, NULL, NULL, teardown_lvgl);
+	setup_fs();
+
+	lv_timer_handler();
+	display_blanking_off(display_dev);
+
+	lv_obj_t *img = lv_img_create(lv_scr_act());
+
+	lv_img_set_src(img, IMG_FILE_PATH);
+	lv_obj_align(img, LV_ALIGN_CENTER, 10, 10);
+
+	while (1) {
+		lv_timer_handler();
+		k_sleep(K_MSEC(10));
+	}
+}
```

It works as expected:

[image]

faxe1008, 2025-02-06:
Also as @JarmouniA pointed out you are using NCS it seems - not sure to which Zephyr version and LVGL this corresponds to.

denny61302, 2025-02-07:
@faxe1008 

Thanks for the reply.

I ran my code again with zephyr 4.0, and the same problem occurred.

Not like the test you showed. My file system is based on FatFs on SD card.

My SD card on SPI node (SDMMC) is working fine. 

I can run lv_fs_open and got the file open.

```
lv_fs_file_t f;
lv_fs_res_t res;
res = lv_fs_open(&f, "/SD:/BLUE.BIN", LV_FS_MODE_RD);
if(res != LV_FS_RES_OK) {
	LOG_ERR("Failed to open file");
} else {
	LOG_INF("File opened");
}

lv_fs_close(&f);
```

But still not able to use lv_img_set_src(img, "/SD:/BLUE.BIN") to show the image on the display.

```
lv_obj_t *img = lv_img_create(lv_scr_act());
lv_img_set_src(img, "/SD:/BLUE.BIN");
lv_obj_align(img, LV_ALIGN_CENTER, 0, 0);

lv_timer_handler();
display_blanking_off(display_dev);
```

> [@denny61302](https://github.com/denny61302) I took Zephyr rev `769df0b579b` and modified the gui test to actually show the SDL display. Here is the diff:
>
> ```
> diff --git [...]

faxe1008, 2025-02-07:
@denny61302 I did not post the entire image uint8_t array because Github does not play nicely with code blocks >3000 locs. You can replace the image with your own ones using lvgl image converter tool online.

closed 2025-04-28 by github-actions[bot]