zephyrproject-rtos/zephyr #85266: usb: mimxrt1180_evk_cm33: most usb cases run failure and console has no output

hakehuang, 2025-02-06:
**Describe the bug**

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - mimxrt1180_evk_cm33
 - What have you tried to diagnose or workaround this issue?
 - git bisect
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
 - the usb_device dose not work well since merged. 
 #85bdab00decc665a608c49153065557434f0a82b

**To Reproduce**

```
west twister -p mimxrt1180_evk/mimxrt1189/cm33  -T tests/subsys/usb/device
```

**Expected behavior**

Test PASS

**Impact**

USB device functions, and CI

**Logs and console output**

no log output

after failure linkserver can not attached anymore.

```
Nc: Probe Firmware: MCU-LINK on-board (r0E2) CMSIS-DAP V3.148 (NXP Semiconductors)
Nc: Serial Number:  D02LB5RKBSD1Y
Nc: VID:PID:  1FC9:0143
Nc: USB Path: 0001:003b:00
Nc: Using memory from core 3 after searching for a good core
Ns: redlink server client - failed to receive response (to op 62) from server after 2000ms - Connection timed out (rc 110)
Nc: connection failed - EE(47): Debug Mailbox Request Failed... Retrying
Ns: redlink server client - failed to receive response (to op 12) from server after 2000ms - Connection timed out (rc 110)
Nc: probe 3 TAP 3 failed to read TAP ID - Ee(FF). Redlink interface error 255.
Ed:02: Failed on connect: Ee(FF). Redlink interface error 255.
Et: Probe(0): Connected&Reset. Was: NotConnected. DpID: 5BA02477. CpuID: 00000000. Info: <None>
Nc:   Last stub error 0: OK
Nc:   Last sticky error: 0x0 AIndex: 3
Nc:   Debug bus selected: MemAp 3
Nc:   DAP Speed test unexecuted or failed
Ns: redlink server client - failed to receive response (to op 45) from server after 2000ms - Connection timed out (rc 110)
Nc: Debug protocol: SWD. RTCK: Disabled. Vector catch: Disabled.
Pc: (100) Target Operation Failed
Nc: invalid server read transfer - expect: 1, actual: 0
Nc: invalid server read transfer - expect: 2, actual: 0
Nc: error closing down debug session - Ee(47). Invalid Transfer
CRITICAL: Critical error
ERRMSG: Exception: Flash operation exited with code 1

```

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**

hakehuang, 2025-02-06:
if tesing the `samples/subsys/usb/dfu` it recongaized as `Bus 001 Device 094: ID 1fc9:014c NXP Semiconductors OO Blank RT Family`, but there is no mcuboot boards for this platform, @mmahadevan108 where can I build bootload to test dfu? do you have any guider? Thanks in advance.

mmahadevan108, 2025-03-04:
@hakehuang, we do not support the legacy Zephyr USB stack on RT1180. Hence  
1. Please test with `tests/subsys/usb/device_next` instead of `tests/subsys/usb/device`
2. Please use `-T sample.usb_device_next.cdc-acm` when testing with `samples/subsys/usb/cdc_acm`
3. `samples/subsys/usb/audio/headset`and `samples/subsys/usb/audio/headphones_microphone` are currently not supported with Zephyr USB Next
4. Please test with `samples/subsys/usb/console-next` instead of `samples/subsys/usb/console`
5. Please use `-T sample.usb_device_next.hid-mouse` when testing with `samples/subsys/usb/hid-mouse`
6. `samples/subsys/usb/testusb` is currently not supported with Zephyr USB Next

mmahadevan108, 2025-03-04:
> samples/subsys/usb/dfu

@hakehuang Please try with `samples/subsys/usb/dfu-next/` on this board.

hakehuang, 2025-03-05:
@mmahadevan108 thanks for updating, if we do not support `usb_device`, please help to update the   
`- usb_device -> - usbd`
in the <board>.yaml, so that case can auto matched.

mmahadevan108, 2025-03-05:
@hakehuang, we do support usb_device. However Zephyr is in the process of transitioning to a new USB Stack. Current we have the legacy `USB_DEVICE_STACK` and the new `USB_DEVICE_STACK_NEXT`.

We have stopped adding support for the legacy USB Device stack pn the new SoC's.

closed 2025-03-06 by fabiobaltieri